gbionics/human-dynamics-estimation #1: YARP architecture for the human dynamics estimation

claudia-lat, 2017-02-10:
I've jotted down a preliminary overview of the YARP modules architecture for the estimation of the dynamics in humans.

[image]

traversaro, 2017-02-10:
cc @diegoferigo @martalorenzini To understand where we can include the visualization.

claudia-lat, 2017-02-13:
For external wrenches we would like to have:
- the value
- the frame with respect to it's expressed
- the link on which it's applied

claudia-lat, 2017-02-13:
For the module related to the `human kinematics`, the inputs are:
- the raw data coming from Xsens acquisition
- a generic template for the human model

As the output of the module we would like to have also the specific model for the subject who performed the experiment.

MartaLorenzini, 2017-03-09:
I've added the RFmodule I've written (xsens-tf-bridge) to the architecture. It allows to publish Xsens segments frames on RViz, the ROS 3D visualizer. To publish the force plates and the iCub frames we are planning to employ already existing yarp modules. 

[image]

francesco-romano, 2017-04-03:
Done by ddfb721c59efacf942b0556e1cfcf42d7f44b3ed

closed 2017-04-03 by francesco-romano

traversaro, 2017-04-03:
Top! 👍