gbionics/human-dynamics-estimation #4: Visualizer for XSens driver

traversaro, 2017-02-16:
As agreed f2f, the first piece of code that @martalorenzini will contribute to this repo is the `xsens-tf-bridge` , that will connect to the xsens driver available at https://github.com/francesco-romano/xsens-mvn and will publish this information in a way such that it can be visualized on RViz .  

Open questions for which I will be happy to hear any feedback: 
* Location: do you agree with keeping the visualizer code here? I think avoiding having too many repository makes sense.
* Structure: we need to store somewhere the `xsens-tf-bridge`. I think it make sense to add a `visualization` folder in the root of the repo, and then put a folder containing `xsens-tf-bridge` inside the visualization folder.

The `xsens-tf-bridge` will be implemented as a classical (single threaded) `yarp::os::RFModule` .

traversaro, 2017-02-16:
@MartaLorenzini can you accept the invitation to https://github.com/orgs/robotology-playground/teams/codyco-developers so I can assigne you this issue?

traversaro, 2017-02-17:
@MartaLorenzini already has a nice prototype, that is missing the part in which data is read from the XSens driver. The data is contained in https://github.com/francesco-romano/xsens-mvn/blob/master/xsenswrapper/thrift/XsensFrame.thrift . 

For the time being, @MartaLorenzini can just copy the thrift file in her project and generate the code for the XSens frame type. Then, she can assume that the port will be read by a yarp::os::BufferedPort of the appropriate type, that will be then converted to a tf message sent on the TF topic.

traversaro, 2017-02-17:
One thing that I do not get of the `XsensFrame.thrift` is how the frames are identified in the list of data. Probably this information is found in the `segments` method of the RPC service:  https://github.com/francesco-romano/xsens-mvn/blob/master/xsenswrapper/thrift/XsensDriverService.thrift#L9  . For the time being, I think it safe to load that list from a configuration file, so that `xsens-tf-bridge` can also work on dumped data.

traversaro, 2017-02-17:
To enable to @MartaLorenzini to integrate her work on the repository it would be convenient if we could merge `repo-structure` in master. @claudia-lat @francesco-romano

francesco-romano, 2017-02-17:
@traversaro regarding https://github.com/robotology-playground/human-dynamics-estimation/issues/4#issuecomment-280592070 yes, that is the idea. 
The order of the segments are exposed through the RPC method which should be called in the configuration. Frames then come in that order

traversaro, 2017-02-17:
Yes, like we do for the controlboard. ROS Types such as JointState typically embed some metadata such as names in the streaming message (see for example http://docs.ros.org/api/sensor_msgs/html/msg/JointState.html ) but that is clearly a trade-off w.r.t. to the size of the message sent.

francesco-romano, 2017-02-17:
@traversaro @MartaLorenzini @claudia-lat branch merged.
Please move to master to use this repo, or branch off to create new features

francesco-romano, 2017-02-17:
Xsens drivers have been pushed in the new repository: https://github.com/robotology-playground/xsens-mvn

Right now they are in a branch as testing with the suit is required

francesco-romano, 2017-03-25:
Done

closed 2017-03-25 by francesco-romano