gbionics/human-dynamics-estimation #5: Installation of URDF files such that they can be found by both RViz and YARP modules

traversaro, 2017-02-16:
We need to understand how to handle the human subject URDF files.

Eventually we will need to handle a high number of subject specific URDF files, but for the time being I think we can agree that for the CoDyCo Y4 demo we will just use 2 or 3 URDF models that we will commit to the repo. 

However we need to make sure that the URDF that we install can be found by RViz for visualization purposes and by the YARP modules for estimation purposes . For ROS use, the URDF files need to be contained in a so-called [ROS Package](http://wiki.ros.org/Packages), while for YARP use we are quite free because we write the code that finds the URDF file.

In the end, we should be able to install the models in a known location, that can be found by ROS as a package and by our software, without using software that is not available on Windows. The classical way of installing ROS Packages is to use catkin, but this is not viable for us because we need to support Windows.

traversaro, 2017-02-16:
I just talked f2f with @MartaLorenzini and apparently we do not need models to be installed in a ROS package to be used, as long as they do not include meshes. :D

This is because RViz can also load a URDF model from a relative path. 
 
The ROS Package problem still exists for models that use meshes, but for now it is not problem for us.

traversaro, 2017-02-17:
@francesco-romano raised the good point that the XSens driver needs the information about the human, then that is needed to generate all the other models used. For the time being, this information will be stored separately such that the generator can be run offline, but eventually we should aim to have a single source of truth (https://en.wikipedia.org/wiki/Single_source_of_truth) for this parameters.

traversaro, 2017-05-09:
See https://github.com/robotology-playground/icub-models .

Yeshasvitvs, 2019-01-30:
The new human models are stored [internally](https://github.com/dic-iit/component_andy/tree/master/human-models). These models should be moved to [human-gazebo](https://github.com/robotology-playground/human-gazebo) public repository. In HDEv1 we stored all the human models internally with in HDE repository. But in HDEv2 we would like to make human models from `human-gazebo` repo installed a destination that can be discovered by the devices of HDEv2.

Yeshasvitvs, 2019-02-06:
The human-gazebo model contains the urdf files and the [CMakeLists.txt](https://github.com/robotology-playground/human-gazebo/blob/master/CMakeLists.txt) installs the urdf files according to the install prefix. This path needs to be set to `YARP_DATA_DIRS` to discover the human urdf files by HDEv2 devices through [xml config files](https://github.com/Yeshasvitvs/human-dynamics-estimation/blob/8ef2d6ce6eaae5ec2ce7eb7b93739c9ce94079fc/conf/xml/Human.xml#L21). More importanatly, the [human-gazebo](https://github.com/robotology-playground/human-gazebo) should be added as a dependency for HDEv2 in the release documentation. 

@lrapetti @claudia-lat @kouroshD @DanielePucci

closed 2019-02-11 by DanielePucci