gbionics/human-dynamics-estimation #6: Write and mantain a nice README for the repo

traversaro, 2017-02-16:
I know that everyone is thinking "let's do that when everything is working", but the state "everything is working" is never actually reached, so it is better to write the README as soon as possible. 

A nice example of repo README is for example : https://github.com/robotology/superquadric-model . 

Things that need to be in the README: 
 * Dependencies,
 * Links to papers,
 * Installation procedure (perhaps mentioning the `codyco-superbuild` or whetever method we will recommend to install the software).

lrapetti, 2019-06-28:
WIth https://github.com/robotology/human-dynamics-estimation/commit/31f0d2d64f7742817ee95dc63047285a4413b226 I started to clean the README of the repo.

What I would like to add is a further explanation on how to run the different demo in the section [Applications](https://github.com/robotology/human-dynamics-estimation#applications) maybe by adding links to wiki pages in which I can write all the details.

Together with this documentation, we should release also the documentation for `Wearables` (https://github.com/robotology/wearables/issues/14) since it is required for running the HDE devices.

DanielePucci, 2019-07-09:
Can we close this issue @lrapetti ?

lrapetti, 2019-07-10:
> Can we close this issue @lrapetti ?

Information on how to run the code are still missing, I would close the issue when everything is in the README. Anyway I have removed (in https://github.com/robotology/human-dynamics-estimation/commit/36685b5f6a1913cb1eea7935fd3e513467d26354) the message stating that the README was a work in progress.

DanielePucci, 2019-07-29:
We can close this issue in the meanwhile because is too generic.  Please @lrapetti open an issue related specifically to the HDE.

closed 2019-07-29 by DanielePucci

traversaro, 2019-07-29:
> We can close this issue in the meanwhile because is too generic. Please @lrapetti open an issue related specifically to the HDE.

This is issue is specific to HDE (it is an issue in the `human-dynamics-estimation` repo). In which sense a new "specific" issue should be opened?

lrapetti, 2019-07-30:
> This is issue is specific to HDE (it is an issue in the `human-dynamics-estimation` repo). In which sense a new "specific" issue should be opened?

Specific for the parts that are still missing in the `README`