gbionics/human-dynamics-estimation #15: Offline data provider

francesco-romano, 2017-02-27:
We should create substitutes of the YARP device drivers to be used in offline processing.

After a discussion with @diegoferigo @traversaro we came out with the following ideas:
- AMTI: ports dumped by the `yarpdatadumper` are enough.
- robot: we'd like to create a new read-only device (i.e. it provides only data and does not accept any command, e.g. implements only `IEncoders`, etc, and does not implement `IPositionControl`) which does not automatically connect at startup and just opens the needed ports.
- Xsens: we create a copy of `xsens_mvn_remote` driver which implements only `IFrameProvider` to provide streaming data. It also opens a partial `.mvnx` file with only the information about the model structure (segments, sensors, etc) to provide this kind of information.

Of course the streaming data will be provided by the pair yarpdatadumper/yarpdataplayer.
All the modules (in this repo at least) will need to be able to know which driver open depending on the online/offline mode.

Yeshasvitvs, 2019-01-30:
@diegoferigo I am not sure if this issue is relevant anymore under the new architecture of HDEv2. Please update it and close it if it not necessary anymore.

diegoferigo, 2019-01-31:
HDE v2 architecture changed completely and this issue is outdated. As discussed with @lrapetti, we can achieve something similar today by dumping a `WearableData` message and replaying it when needed.

There's also the discussion on the RPC port of the `IWearRemapper`, but I think it was hosted in a private repo. In any case, reading metadata through RPC should be optional now.

closed 2019-01-31 by diegoferigo