gbionics/human-dynamics-estimation #35: Eigen Sparse LLT solver supports only Column Major ordering

francesco-romano, 2017-03-24:
This code 
```cpp
Eigen::SparseMatrix<double, Eigen::RowMajor> A(5,5);
Eigen::SparseVector<double> res(5);
Eigen::SparseVector<double> b(5);
Eigen::SimplicialLLT<Eigen::SparseMatrix<double, Eigen::RowMajor> > dec;
dec.compute(A);
res = dec.solve(b);
```
does not compile with the following error:
```

Showing All Issues
/usr/local/include/eigen3/Eigen/src/SparseCholesky/SimplicialCholesky.h:283:60: No matching conversion for functional-style cast from 'const AdjointReturnType' (aka 'const Eigen::Transpose<const Eigen::SparseMatrix<double, 1, int> >') to 'MatrixU' (aka 'TriangularView<const Eigen::Transpose<const Eigen::SparseMatrix<double, 0, int> >, Eigen::Upper>')
```

while this code successfully compiles:
```cpp
Eigen::SparseMatrix<double, Eigen::ColMajor> A(5,5);
Eigen::SparseVector<double> res(5);
Eigen::SparseVector<double> b(5);
Eigen::SimplicialLLT<Eigen::SparseMatrix<double, Eigen::ColMajor> > dec;
dec.compute(A);
res = dec.solve(b);
```

While waiting for column major ordering implementation in iDynTree or more information from eigen (upstream) we switch to LU.

francesco-romano, 2017-03-24:
Given this message in the LU documentation:

> The input matrix A should be in a compressed and column-major form. Otherwise an expensive copy will be made. You can call the inexpensive makeCompressed() to get a compressed matrix.

we should start supporting column ordering as soon as possible.

Yeshasvitvs, 2019-01-30:
@francesco-romano @traversaro is this issue related to HDEv2? if not we can close it.

francesco-romano, 2019-01-31:
Of course I have no idea what HDEv2 is.

@traversaro can check better, but I think I implemented Column major ordering for iDynTree sparse matrices. I think you should check in the iDynTree Berdy-related classes how the solve is performed. 
In case it is not using a correct solver, you should probably move this issue there as AFAIK you migrated to use iDynTree classes.

Yeshasvitvs, 2019-02-06:
`HumanStateProvider` in HDEv2 uses iDynTree classes to do the ik. So, this issue can be closed.

closed 2019-02-06 by Yeshasvitvs

traversaro, 2019-02-06:
> @traversaro can check better, but I think I implemented Column major ordering for iDynTree sparse matrices.

I can confirm.