gbionics/human-dynamics-estimation #36: IK fails sometimes with 48DoF model

francesco-romano, 2017-03-25:
While the inverse kinematics successfully manages to obtain the joint configuration for the full 66 DoF model (i.e. 22 joints of 3 DoF each), when we use the simplified model the IK sometimes fails to obtain a reasonable solution, not only quantitative speaking, but also qualitative speaking, see attached gif

[image]

francesco-romano, 2017-03-25:
Before checking why the IK obtains this solution, we should check the URDF model (e.g. wrong joint limits, wrong rotation axes)

cc @MartaLorenzini

francesco-romano, 2017-07-07:
Before losing the dumped data, I attach some frames in which it is possible to experience the problem.
It actually seems an IK problem

[wrongFrames.tar.gz](https://github.com/robotology/human-dynamics-estimation/files/1131706/wrongFrames.tar.gz)

francesco-romano, 2017-09-11:
To be checked again after #71

Yeshasvitvs, 2019-01-30:
I guess this issue is not relevant anymore as we are using only 66DoF human models with HDEv2. So, I think we can close this issue.

@claudia-lat @DanielePucci

claudia-lat, 2019-01-31:
Closed #36

closed 2019-01-31 by claudia-lat