gbionics/human-dynamics-estimation #45: Post-demo activities

francesco-romano, 2017-03-30:
This issues summaries all the activities which should take place after the CoDyCo Y4 demo takes place.
I will put here also issues related to other repositories as they have been uncovered during the development of this project.

**Human-Dynamics-Estimation**
- [ ] make the configuration of priors more scalable. At the current state only the use cases needed by the demo have been implemented.

**YARP**
- [ ] various IDL issues (see [YARP#1132](https://github.com/robotology/yarp/issues/1132) [YARP#1143](https://github.com/robotology/yarp/issues/1143))
- [ ] yarpdataplayer enhancements (for offline processing) see [YARP#1144](https://github.com/robotology/yarp/issues/1144)
- [ ] New `yarp::dev` interfaces: `IMultipleAnalogSensors` (for `wholebodydynamics` and https://github.com/robotology-playground/forcetorque-yarp-devices) , `IFrameProvider` (for https://github.com/robotology-playground/xsens-mvn) and `IMultipleIMUs` (for https://github.com/robotology-playground/xsens-mvn)

claudia-lat, 2017-04-03:
Added two issues in Human-Dynamics-Estimation section.

claudia-lat, 2018-10-29:
Some of these activities have been done.  Others are no longer useful because of the refactoring (see [here](https://github.com/robotology-playground/wearables)).
I'm going to close this issue for now.
Closed #45.

closed 2018-10-29 by claudia-lat