gbionics/human-dynamics-estimation #48: YARP driver for offline replay of mvnx data

diegoferigo, 2017-03-31:
I report below the steps I'm working on in order to accomplish the third bullet point of Issue #15.

- [ ] Find an XML library that could potentially be included in YARP with `SAX` and `Schema Validation` support, and a proper license
- [ ] Write a SAX parser. Mandatory due to the big size of the `mvnx` file, `DOM` is not efficient in this case
- [ ] Validate the `mvnx` file with the `schema.xsd` provided by Xsens

diegoferigo, 2017-03-31:
For what concern XML libraries, the table below reports some choices with their features and drawbacks:

| Library name | Language | Features | License |
| --- | --- | --- | --- |
| Xerces | C++  | DOM, SAX, XML Schema              | Apache 2.0                     |
| POCO | C++   | DOM, SAX, XML Schema              | Boost *                            |
| libxml2 | C      | DOM, SAX                                      | GNU LGPL                     |
| expat   | C      | ~SAX                                               | MIT                                 |
| Qt        | C++  | DOM, SAX, Stream, XML Schema | GPL, LGPL, commercial |

[*] Schema Validation is in the POCO Pro Framework

Considering that Schema Validation is an important for us, even if Qt are bulky they are the only complete choice with a license that could fit our needs.

diegoferigo, 2018-03-30:
cc @lucaTagliapietra

drdanz, 2018-03-30:
Just for reference, IIRC when I checked some time ago `libxml2` was supporting "XML Schema Part 2" but "XML Schema Part 1" was still a work in progress, but I have no idea what "Part 1 and 2" mean and if there were progresses recently.

diegoferigo, 2018-03-30:
The reason why I discarded libxml2 was the C interface. I found way easier using the Qt, the APIs are in the same moment simple and provide a good flexibility while parsing a big xml in Stream mode.

The initial version I developed last summer is hosted at https://github.com/robotology-playground/xsens-mvn/tree/master/mvnxparser, we're in the process of polishing it and, maybe, detach the core (`XMLStreamReader`) as separated project.

traversaro, 2018-11-22:
Somebody remembers why in the end we did not go for Xerces? There was a specific reason or just a preference  for QtXml ?

diegoferigo, 2018-11-22:
I remember I explored solutions also for a possible inclusion into YARP, and I might have discarded it due to the license. Since we already use Qt libraries they looked a better solution.

Yeshasvitvs, 2019-01-30:
@diegoferigo @traversaro is this issue still relevant for HDEv2 ? If not we can close it.

traversaro, 2019-01-31:
If https://github.com/robotology/human-dynamics-estimation/issues/15 has been solved in the new architecture, then this can be closed.

diegoferigo, 2019-01-31:
This parser was used to read efficiently big `mvnx` files. Since now we decided to not renew the license of Xsens Studio, we don't need anymore the mvnx parser.

However, the `mvnx` parser is only an implementation of a generic stream-based xml parser, and the generic implementation might be useful for future projects. We can discuss f2f where to store it (right now is still in the deprecated `xsens-mvn` repo).

DanielePucci, 2019-02-04:
@diegoferigo 
Please write a short recap and, if this is the case as spoken F2F, close the issue

DanielePucci, 2019-02-11:
CC
@diegoferigo

diegoferigo, 2019-02-12:
Not much to add to my previous comment. @claudia-lat used the parser to avoid loading GB-sized log files in matlab to extract some information. Now, since we don't have anymore the license to generate `mvnx` files, we can move the base class [`XMLStreamReader`](https://github.com/robotology-playground/xsens-mvn/tree/master/mvnxparser/XMLStreamReader) somewhere else, since it is a generic wrapper of the stream-based parsing implemented in the Qt libraries. This is in preparation of deprecating `xsens-mvn` repo.

closed 2019-02-12 by diegoferigo