gbionics/human-dynamics-estimation #52: Automatically format code (clang-format)

francesco-romano, 2017-04-05:
It would be nice to define a clang-format.

I started with this commit 2748cd36629f57d9d29e476fe95025fe31ebd56a, but some things needs to be fine tuned (e.g. multiline pure virtual functions have `= 0` in a newline, enforce `{}` on every `if` etc)

Also, as it seems not possible to synchronize git hooks across remotes, (and with my initial tests they are creating some issues during commits) we should enforce it at PR level.

To run clang-format on this project one can run the following bash command in the root of the project
` find . -path "./build*" -prune -o -iname *.h -o -iname *.cpp  | xargs clang-format -i`

diegoferigo, 2017-06-15:
I'd like to bump again this discussion. For the time being, we can start a chat on a `clang-format` that everyone like(ish), keeping in mind that we will never find a solution that makes all happy. Let's approach this as a discussion with your partner, and find a compromise :)

In my opinion we can reach a draft here on repositories that we directly handle, and then move this discussion to all other developers. This would allow to start at least with a small group on the same line.

This is my `clang-format`:
```yaml
---
BasedOnStyle: WebKit
AlignAfterOpenBracket: Align
AlignConsecutiveAssignments: 'true'
AlignConsecutiveDeclarations: 'false'
AlignOperands: 'true'
AlignTrailingComments: 'true'
AllowAllParametersOfDeclarationOnNextLine: 'true'
AllowShortBlocksOnASingleLine: 'true'
AllowShortCaseLabelsOnASingleLine: 'false'
AllowShortFunctionsOnASingleLine: Inline
AllowShortIfStatementsOnASingleLine: 'false'
AllowShortLoopsOnASingleLine: 'false'
AlwaysBreakAfterDefinitionReturnType: None
AlwaysBreakAfterReturnType: None
AlwaysBreakBeforeMultilineStrings: 'false'
AlwaysBreakTemplateDeclarations: 'true'
BinPackArguments: 'false'
BinPackParameters: 'false'
BraceWrapping:
  AfterClass: false
  AfterControlStatement: false
  AfterEnum: true
  AfterFunction: true
  AfterNamespace: false
  AfterObjCDeclaration: false
  AfterStruct: false
  AfterUnion: false
  BeforeCatch: false
  BeforeElse: true
  IndentBraces: false
BreakBeforeBraces: Custom
BreakConstructorInitializersBeforeComma: 'true'
ColumnLimit: '100'
ConstructorInitializerAllOnOneLineOrOnePerLine: 'true'
ContinuationIndentWidth: '10'
Cpp11BracedListStyle: 'true'
DerivePointerAlignment: 'false'
DisableFormat: 'false'
ExperimentalAutoDetectBinPacking: 'false'
FixNamespaceComments: 'true'
IndentCaseLabels: 'true'
IndentWidth: '4'
KeepEmptyLinesAtTheStartOfBlocks: 'false'
Language: Cpp
NamespaceIndentation: None
PointerAlignment: Left
ReflowComments: 'true'
SortIncludes: 'true'
SpaceBeforeAssignmentOperators: 'true'
SpaceBeforeParens: ControlStatements
SpaceInEmptyParentheses: 'false'
SpacesInAngles: 'false'
SpacesInParentheses: 'false'
SpacesInSquareBrackets: 'false'
Standard: Cpp11
TabWidth: '4'
UseTab: Never
...
```

The diff wrt the file from @francesco-romano is the following:
```diff
--- /scratch/clang-format-romanz        2017-06-15 10:12:44.882058574 +0200
[... 127 lines cut ...]
+SpaceBeforeAssignmentOperators: 'true'
 SpaceBeforeParens: ControlStatements
-SpaceInEmptyParentheses: false
-SpacesBeforeTrailingComments: 1
-SpacesInAngles:  false
-SpacesInContainerLiterals: true
-SpacesInCStyleCastParentheses: false
-SpacesInParentheses: false
-SpacesInSquareBrackets: false
-Standard:        Cpp03
-TabWidth:        4
-UseTab:          Never
+SpaceInEmptyParentheses: 'false'
+SpacesInAngles: 'false'
+SpacesInParentheses: 'false'
+SpacesInSquareBrackets: 'false'
+Standard: Cpp11
+TabWidth: '4'
+UseTab: Never
 ...
```

Despite my `clang-format` forces 4 spaces indentation, I'd prefer using tabs for indentations and spaces for alignment. Without an automatic way like this, using this hybrid is a pain and in such a big codebase it would be difficult to maintain, however `clang-format` makes things easy. Tabs and spaces have their own reason to exists, each with pros and cons, and using them wisely exploiting the best of both is possible. Despite this, I'm gonna start with compromises from my side, I could agree on 4 spaces.

@traversaro @francesco-romano @DanielePucci @claudia-lat @gabrielenava @S-Dafarra @nunoguedelha @lucatagliapietra @Yeshasvitvs

traversaro, 2017-06-15:
Agnostic to everything, as long as there is consensus. 

The only thing I am **strongly** against is mixing tabs and spaces in any way. 

In a countless use cases (for example when visualizing source files in GitHub web interface, but also when you open a file in a random machine with a random editor with a random operating system) you can't (easily) specify the width of the tabs (on GitHub is always 8 spaces) and this screws up the visualization of any file in which tabs and spaces are mixed assuming that a tabs is 2 or 4 spaces.

drdanz, 2017-06-15:
-1000 for tabs

drdanz, 2017-06-15:
Why instead of starting with a huge file full of settings, don't we start with a "blank" one and we discuss and add one by one each setting? reviewing such a file is impossible

diegoferigo, 2017-06-15:
I propose this: let's take a portion of demo code that contains all the cases (even the complicated one) in the OP,

diegoferigo, 2017-06-15:
I created [this gist](https://gist.github.com/diegoferigo/716259a64adb1035e383db3864805ac5). Having a demo repository only for this probably too much, let me know if you have better ideas.

pattacini, 2017-06-15:
I'm not aware of the background behind this praiseworthy initiative, so I could miss important details, but let me raise one point 'cause I can see there might be some concerns, at least with me 😉 

I have nothing against defining official code styles for e.g. `yarp` or specific project repositories, but I'm a bit skeptical - to say the less - to sort of blindly extend these guidelines to all repositories in our ecosystem.

I'm a bit "conservative" with my own code style - to soften my religious truth of "_nothing of this world can convince me to use `{` inline with `if` or to wrap one single operation_" -, but at the same time I also hate seeing modules written with different styles, of course, so that I obviously adapt my contributions to the maintainer's choice when opening a PR on a repo that I don't maintain.

The role of the **maintainer** is the key in my view. Therefore, for most of our repositories (thus excluding `yarp` and a few others), the maintainer(s) should check the style of the contributed code and push developers to react accordingly, even by means of automatic checkers.

I don't think it's feasible to force people adhering to one single style everywhere, while it would be beneficial ending up with multiple coherent code styles, depending on the diverse maintainers, who are committed to enforcing them.

pattacini, 2017-06-15:
@diegoferigo 👍 for the global (but not restrictive) style + local style.

alecive, 2017-06-15:
@diegoferigo just to give you another reason to always use spaces instead of tabs, please be aware that [according to statistics, programming with spaces instead of tabs makes you richer](https://arstechnica.com/information-technology/2017/06/according-to-statistics-programming-with-spaces-instead-of-tabs-makes-you-rich/?comments=1)!!!

diegoferigo, 2017-06-16:
@alecive Let's close the never ending tab vs spaces [in this way](https://www.youtube.com/watch?v=SsoOG6ZeyUI) with a smile :smile: We can't change our minds about, it's always a compromise, I'm already using spaces here xD

So, **TL;DR** so far:

1. In projects maintained by one developer, like often happens, having a specific `.clang-format` for that project looks a good compromise
2. In bigger projects (like `yarp`) a global `.clang-format` could be potentially enforced, and the discussion can happen in [this gist](https://gist.github.com/diegoferigo/716259a64adb1035e383db3864805ac5)
3. For reducing style mismatches, the project-specific `.clang-format` should use as starting point the global one used in the bigger projects

drdanz, 2017-06-19:
Just found out this... https://clangformat.com/

drdanz, 2017-06-19:
Also this http://clang-format.me/

diegoferigo, 2017-06-20:
In the past I used [this tool](https://zed0.co.uk/clang-format-configurator/) and it is already linked in the gist, but that is another valid alternative. I'd suggest to discuss further steps f2f at the next IFI day.

traversaro, 2017-06-20:
cc @diegoferigo https://gist.github.com/diegoferigo/716259a64adb1035e383db3864805ac5#gistcomment-2125058

diegoferigo, 2017-06-20:
I didn't received notification about your comment, sorry.

diegoferigo, 2017-06-20:
Let's switch to https://github.com/robotology-playground/clang-format/issues/1

closed 2017-06-20 by diegoferigo