gbionics/human-dynamics-estimation #60: Fix AppVeyor or remove webhooks

traversaro, 2017-04-19:
@diegoferigo & @francesco-romano ?

francesco-romano, 2017-04-19:
I opened the issues on YARP, but nobody took care of those issues

traversaro, 2017-04-19:
Sorry, I forgot. 
Issues necessary to fix AppVeyor : 
* https://github.com/robotology/yarp/issues/1132

Yeshasvitvs, 2019-01-30:
@traversaro please update this issue and close it if it is not needed anymore.
Thanks

traversaro, 2019-01-31:
The issue in YARP that was preventing AppVeyor to work correctly has been solved since a long time (see https://github.com/robotology/yarp/issues/1132). 

I think a working Windows-based CI would drastically simplify the developement workflow, especially when a software is developed on Ubuntu, and so you want to make sure that is compiling (and running, if [tests](https://github.com/robotology/how-to-export-cpp-library#-add-a-test) are available) correctly on Windows without the need to manually check. AppVeyor is one possible service to implement Windows-based CI (and is probably the easiest to set up, as it is already there and just needs to be fixed) but possible alternative include Travis CI on Windows (used extensivly by @diegoferigo) and Azure Pipelines (the best one w.r.t. to the time that a build can last before being considered failer automatically, but we are still waiting for the green light to enable it in `robotology`.

To be honest, given that the `master` branch of this repo is effectivly unmantained and the installation instructions are not working, I think it would make sense to tag it for reference purpouse, and then merge `release/HDEv2` (or whatever branch you actually use for the demos) in `master` (probably documenting in the README that is still a work in progress) and work on that one. So then you can work on it with the usual PR + ci + review workflow, that is quite important for a software infrastructure of the complexity of HDE, especially if it is now used by default in lab's demo.

diegoferigo, 2019-01-31:
I always struggled with AppVeyor, mainly because I don't feel comfortable working with neither `cmd` nor `powershell`. The big advantage of Windows for Travis is that it uses [git bash](https://gitforwindows.org/) which provides a bash-like terminal. There are some caveat to know, especially related to env variables, but my experience is totally positive. The VMs spin up quickly and there's no waiting (this might be also because I think I'm the only user that has windows jobs xD).

The only problem is that currently Windows works only in repos without deployment. If a repo has any secret variable stored in the Travis web interface, Windows fails. However, the combination of working with a PR + review process and enabling Travis on the fork, would make everything work. In this case windows tests will run only on the branch of the fork, but it is good enough in my opinion.

DanielePucci, 2019-02-18:
We decide to use Travis. Let's close this issue.

traversaro, 2019-02-18:
I would just keep this issue open, and rename it from "Fix AppVeyor" to "Fix Windows CI".

traversaro, 2019-06-09:
See https://github.com/robotology/human-dynamics-estimation/issues/124#issuecomment-500250851 that also applies to AppVeyor.

lrapetti, 2019-06-10:
As discussed with @diegoferigo we plan to remove the AppVeyor test, since in the future we should move towards using Trevis also for Windows.

traversaro, 2019-06-10:
This is an addition reason to disable AppVeyor avoiding to wast resources on it.

lrapetti, 2019-06-10:
Closing since AppVeyor has been disabled. A new issue for adding back CI for Windows has been opened https://github.com/robotology/human-dynamics-estimation/issues/126/

closed 2019-06-10 by lrapetti