gbionics/human-dynamics-estimation #64: human-state-estimator depends on xsens-mvn and such dependencies is not documented

traversaro, 2017-05-09:
See https://github.com/robotology-playground/human-dynamics-estimation/blob/master/human-state-provider/CMakeLists.txt#L8 . 

cc @lucaTagliapietra

francesco-romano, 2017-05-10:
This dependency should have been very temporary.
Those YARP interfaces should be integrated in YARP devel

lrapetti, 2018-11-20:
In [`HDEv2`](https://github.com/robotology/human-dynamics-estimation/tree/release/HDEv2) the dependency from [`xsens-mvn`](https://github.com/robotology-playground/xsens-mvn) is dropped, and it will be replaced by the dependency on [`wearables`](https://github.com/robotology-playground/wearables) (It will require to be properly documented).

Yeshasvitvs, 2019-01-30:
@traversaro @lrapetti I think we can close this issue as it is not related to HDEv2. We can open a new issue if needed.

Yeshasvitvs, 2019-02-01:
We are closing this issue.

closed 2019-02-01 by Yeshasvitvs