gbionics/human-dynamics-estimation #70: How to compute on-line joint acceleration for the vector of measurements y

claudia-lat, 2017-08-30:
The joint accelerations are needed for filling in the vector y of the measurements.
For the time being, y is a vector with the entries equal to 0 apart from the external forces readings (force plates + robot).  It means that the estimation of the vector d depends only from the forces and doesn't take into account any joint acceleration contribution!
In [ MAPest ](https://github.com/claudia-lat/MAPest/blob/master/Experiments/23links_human/src/IK.m) the joint acceleration was computed by using a Savitzky-Golay filter.  Clearly is not possible to adopt it in the on-line acquisition case.
Suggestions on how to proceed? 
@lucaTagliapietra @francesco-romano  @DanielePucci @traversaro

claudia-lat, 2017-08-30:
@lucaTagliapietra suggested the state space filter solution:
https://github.com/RealTimeBiomechanics/Filter

francesco-romano, 2017-08-31:
@claudia-lat this is a quite general discussion that is not totally related to the software implementation.
IMHO, discussions of this kind should be hold in `loc2` elements.

DanielePucci, 2017-08-31:
@francesco-romano 👍

claudia-lat, 2017-09-01:
The issue was moved into `loc2/element_berdy`.

closed 2017-09-01 by claudia-lat