gbionics/human-dynamics-estimation #74: Document how to run the human+robot demonstration on a real iCub

traversaro, 2017-11-30:
Missing documentation: 
* How to publish the right JointState topic from the robot, see the last 4 commits in https://github.com/francesco-romano/robots-configuration/commits/icub_ros .

traversaro, 2017-11-30:
* For publishing the external forces, we still need the fork of wbd in https://github.com/robotology/codyco-modules/tree/nomerge/wbd_ros .

traversaro, 2017-11-30:
Some of this issues are related to things to do mentioned in https://github.com/robotology/human-dynamics-estimation/issues/45 .

traversaro, 2017-11-30:
cc @claudia-lat @lucaTagliapietra @diegoferigo .

Yeshasvitvs, 2019-01-30:
This information will be documented in the wiki page Running HDE under https://github.com/robotology/human-dynamics-estimation/wiki/Human-Dynamics-Estimation-Setup-and-Usage

Yeshasvitvs, 2019-02-01:
As someone the issues pointed out above are with respect to HDEv1 I would like to close this issue and open a new one with respect to HDEv2. Currently, we have a much cleaner way to run HDE demo for the human and for both human and icub in simulation. We haven't tried to run HDE with the real robot yet.

@traversaro @lrapetti @claudia-lat @DanielePucci

closed 2019-02-01 by Yeshasvitvs