gbionics/human-dynamics-estimation #75: Human-effort-bridge (often) crashes when launching Ruiz

claudia-lat, 2017-11-30:
If you launch Rviz when all the HDE modules on the yarpmanager are running, the module human-effort-bridge module cashes.  This implies that you are NOT able to visualize the effort on the model.

lucaTagliapietra, 2017-11-30:
Reproduced also on Linux - setup icub24
~~~*** Error in `human-effort-bridge': double free or corruption (fasttop): 0x00000000015b7730 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f86616ea7e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f86616f337a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f86616f753c]
/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.1(+0x6a7f9)[0x7f866203f7f9]
/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.1(_ZN4yarp2os4impl10BottleImpl5clearEv+0x37)[0x7f8662038ea7]
/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.1(_ZN4yarp2os8Property3putERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS0_5ValueE+0x15c)[0x7f86620e75ec]
/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.1(+0x12adad)[0x7f86620ffdad]
/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.1(+0x12c5db)[0x7f86621015db]
/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.1(+0x1301be)[0x7f86621051be]
/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.1(_ZN4yarp2os14ResourceFinder9findPathsERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x6a)[0x7f86620fc65a]
/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.1(_ZN4yarp2os18YarpPluginSelector4scanEv+0x2b7)[0x7f8662148d67]
/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.1(_ZN4yarp2os8Carriers7Private14scanForCarrierEPKNS0_5BytesE+0x168)[0x7f8662043408]
/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.1(_ZN4yarp2os8Carriers7Private13chooseCarrierEPKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEPKNS0_5BytesEbb+0x770)[0x7f8662043d20]
/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.1(_ZN4yarp2os4impl8Protocol23expectProtocolSpecifierEv+0xc0)[0x7f86620f3ed0]
/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.1(_ZN4yarp2os4impl8Protocol12expectHeaderEv+0x27)[0x7f86620f4407]
/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.1(_ZN4yarp2os4impl8Protocol4openERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0xa7)[0x7f86620f45d7]
/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.1(+0xfda27)[0x7f86620d2a27]
/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.1(+0x1658dd)[0x7f866213a8dd]
/usr/lib/libACE-6.3.3.so(_ZN21ACE_OS_Thread_Adapter6invokeEv+0xc6)[0x7f8660c39f56]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba)[0x7f866071e6ba]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7f866177a3dd]
======= Memory map: ========
00400000-00430000 r-xp 00000000 00:2b 1585748                            /usr/local/src/robot/codyco-superbuild/build/install/bin/human-effort-bridge
0062f000-00630000 r--p 0002f000 00:2b 1585748                            /usr/local/src/robot/codyco-superbuild/build/install/bin/human-effort-bridge
00630000-00631000 rw-p 00030000 00:2b 1585748                            /usr/local/src/robot/codyco-superbuild/build/install/bin/human-effort-bridge
01581000-01748000 rw-p 00000000 00:00 0                                  [heap]
7f8600000000-7f8600022000 rw-p 00000000 00:00 0 
7f8600022000-7f8604000000 ---p 00000000 00:00 0 
7f8608000000-7f8608022000 rw-p 00000000 00:00 0 
7f8608022000-7f860c000000 ---p 00000000 00:00 0 
7f860c000000-7f860c022000 rw-p 00000000 00:00 0 
7f860c022000-7f8610000000 ---p 00000000 00:00 0 
7f8612ffe000-7f8612fff000 ---p 00000000 00:00 0 
7f8612fff000-7f86137ff000 rw-p 00000000 00:00 0 
7f86137ff000-7f8613800000 ---p 00000000 00:00 0 
7f8613800000-7f8614000000 rw-p 00000000 00:00 0 
7f8614000000-7f8614022000 rw-p 00000000 00:00 0 
7f8614022000-7f8618000000 ---p 00000000 00:00 0 
7f8618000000-7f861807c000 rw-p 00000000 00:00 0 
7f861807c000-7f861c000000 ---p 00000000 00:00 0 
7f861c000000-7f861c022000 rw-p 00000000 00:00 0 
7f861c022000-7f8620000000 ---p 00000000 00:00 0 
7f8620000000-7f8620022000 rw-p 00000000 00:00 0 
7f8620022000-7f8624000000 ---p 00000000 00:00 0 
7f8624000000-7f8624022000 rw-p 00000000 00:00 0 
7f8624022000-7f8628000000 ---p 00000000 00:00 0 
7f86287f9000-7f86287fa000 ---p 00000000 00:00 0 
7f86287fa000-7f8628ffa000 rw-p 00000000 00:00 0 
7f8628ffa000-7f8628ffb000 ---p 00000000 00:00 0 
7f8628ffb000-7f86297fb000 rw-p 00000000 00:00 0 
7f86297fb000-7f86297fc000 ---p 00000000 00:00 0 
7f86297fc000-7f8629ffc000 rw-p 00000000 00:00 0 
7f8629ffc000-7f8629ffd000 ---p 00000000 00:00 0 
7f8629ffd000-7f862a7fd000 rw-p 00000000 00:00 0 
7f862a7fd000-7f862a7fe000 ---p 00000000 00:00 0 
7f862a7fe000-7f862affe000 rw-p 00000000 00:00 0 
7f862affe000-7f862afff000 ---p 00000000 00:00 0 
7f862afff000-7f862b7ff000 rw-p 00000000 00:00 0 
7f862b7ff000-7f862b800000 ---p 00000000 00:00 0 
7f862b800000-7f862c000000 rw-p 00000000 00:00 0 
7f862c000000-7f862c022000 rw-p 00000000 00:00 0 
7f862c022000-7f8630000000 ---p 00000000 00:00 0 
7f8630000000-7f8630022000 rw-p 00000000 00:00 0 
7f8630022000-7f8634000000 ---p 00000000 00:00 0 
7f8634000000-7f8634022000 rw-p 00000000 00:00 0 
7f8634022000-7f8638000000 ---p 00000000 00:00 0 
7f8638000000-7f8638022000 rw-p 00000000 00:00 0 
[... 77 lines cut ...]
7f865fa1d000-7f865fa52000 r-xp 00000000 08:06 1189703                    /usr/lib/x86_64-linux-gnu/libdc1394.so.22.1.11
7f865fa52000-7f865fc51000 ---p 00035000 08:06 1189703                    /usr/lib/x86_64-linux-gnu/libdc1394.so.22.1.11
7f865fc51000-7f865fc52000 r--p 00034000 08:06 1189703                    /usr/lib/x86_64-linux-gnu/libdc1394.so.22.1.11
7f865fc52000-7f865fc53000 rw-p 00035000 08:06 1189703                    /usr/lib/x86_64-linux-gnu/libdc1394.so.22.1.11
7f865fc53000-7f865fc93000 rw-p 00000000 00:00 0 
[... cut ...]
7f86602de000-7f86604dd000 ---p 00003000 08:06 7474867                    /lib/x86_64-linux-gnu/libdl-2.23.so
7f86604dd000-7f86604de000 r--p 00002000 08:06 7474867                    /lib/x86_64-linux-gnu/libdl-2.23.so
7f86604de000-7f86604df000 rw-p 00003000 08:06 7474867                    /lib/x86_64-linux-gnu/libdl-2.23.so
7f86604df000-7f8660510000 r-xp 00000000 08:06 1188284                    /usr/lib/x86_64-linux-gnu/libedit.so.2.0.53
7f8660510000-7f8660710000 ---p 00031000 08:06 1188284                    /usr/lib/x86_64-linux-gnu/libedit.so.2.0.53Aborted (core dumped)
~~~

lucaTagliapietra, 2017-11-30:
Catkin Package - to be used meshes are required

[human_viz_bridge.zip](https://github.com/robotology/human-dynamics-estimation/files/1519378/human_viz_bridge.zip)

claudia-lat, 2017-12-01:
Part of these files are in 
https://github.com/robotology/human-dynamics-estimation/tree/master/human-viz-bridge/misc

lucaTagliapietra, 2018-01-31:
A possible cause might be some mess left in YARP server by the previous executions.
Remember to run a clean before run everything.
`yarp clean --timeout 0.1`

However, this need to be confirmed by a more extensive set of trials.

Yeshasvitvs, 2019-01-30:
This issue is not relevant any more in the new architecture of HDEv2. So closing this issue.

@claudia-lat @diegoferigo @DanielePucci

closed 2019-01-30 by Yeshasvitvs