gbionics/human-dynamics-estimation #78: Human Dynamics Estimation interface module to send data to simulink through a device driver

Yeshasvitvs, 2018-02-08:


Yeshasvitvs, 2018-02-08:
[This module](https://github.com/Yeshasvitvs/human-dynamics-estimation/tree/hde_interface/hde-interface) implements a custom control board device driver wrapped with a controlboardwrapper2 to send human data as a control board device.

Yeshasvitvs, 2018-02-09:
Tested the driver with simulink blocks of get measurements for joint positions, velocity, acceleration and torques. The values read are correct.

Thank you very much for the discussions and support guys @diegoferigo @traversaro :grinning:

diegoferigo, 2018-02-12:
Well done @Yeshasvitvs!! When you're ready, open a PR so we can review it. We should discuss also the naming you used, let's see if we can make even it more clear.

Yeshasvitvs, 2018-02-12:
@diegoferigo I noticed that in matlab the get measurement block for joint acceleration is actually reading joint velocities :confused::question:

Yeshasvitvs, 2018-02-13:
[image]

@diegoferigo This data is collected using this [test](https://github.com/Yeshasvitvs/human-dynamics-estimation/tree/hde_interface/hde-interface/test/remote_controlboard) remote control board. The data in the first file is the actual values from HDE ports and the data in the bottom file are read from the remote control board. Here, the data is read correctly. So, I think we need to have a look at the matlab end of the remote controlboardremapper

diegoferigo, 2018-02-13:
Let's have a chat f2f for details. In the meantime, here is the code that reads accelerations

https://github.com/robotology/WB-Toolbox/blob/b75d36d856e03aba7f0fa2ad37b22b082078faa4/toolbox/src/GetMeasurement.cpp#L158-L168

DanielePucci, 2018-03-09:
@claudia-lat @lucaTagliapietra

Yeshasvitvs, 2018-03-20:
The `get acceleration` block of WB3.0 reads `joint velocities` instead of `joint accelerations` . It is tracked in [this issue](https://github.com/loc2/element_human-dynamics-estimation/issues/13)

Yeshasvitvs, 2018-03-20:
Additional testing and tuning of HDE control board driver is followed in [this issue](https://github.com/loc2/element_human-dynamics-estimation/issues/12)

closed 2018-03-20 by Yeshasvitvs