gbionics/human-dynamics-estimation #81: Considering base of the human in the controller for reactive pHRI

Yeshasvitvs, 2018-02-20:
I am implementing the reactive pHRI control laws of AnDy deliverable D4.1 by extending the current standup experiment. In this case, how should the base pose of the human to be considered? I have two questions in this regard

* If I recall correctly, with the current standup experiment setup the transformation from the robot to the human is a fixed transformation. What are the links considered on the robot and the human (e.g., `root_link` of the robot to `pelvis` of human??)for this fixed transformation? @claudia-lat @lucaTagliapietra @diegoferigo @DanielePucci 

* The human-state-provider module gives the values of `baseOriginWRTGlobal` , `baseOrientationWRTGlobal`, `baseVelocityWRTGlobal` as considered in this [thrift](https://github.com/robotology/human-dynamics-estimation/blob/8da06176f555d6d696664d769bcd05a7155460b9/human-state-provider/thrifts/HumanState.thrift#L37) file. Which human link is considered as the base link in general and in the current standup experiment particularly? @francesco-romano

Yeshasvitvs, 2018-02-22:
Ping! @DanielePucci @lucaTagliapietra @claudia-lat @francesco-romano

Yeshasvitvs, 2018-02-22:
Moved to [this](https://github.com/loc2/element_human-dynamics-estimation/issues/9) issue

closed 2018-02-22 by Yeshasvitvs

francesco-romano, 2018-02-24:
@Yeshasvitvs I can be wrong, but this might be the answer you are looking for: https://github.com/robotology/human-dynamics-estimation/blob/master/human-state-provider/conf/human-state-provider.ini#L4

Yeshasvitvs, 2018-02-24:
@francesco-romano please check the discussion on the other issue and let me know if you find something wrong

francesco-romano, 2018-02-26:
@Yeshasvitvs I am sorry, but I do not have access to private repositories.

Yeshasvitvs, 2018-02-26:
@francesco-romano I am afraid I do not have user privileges to grant access to you. So, I took a snapshot of the conversation. Please have a look and let me know your comments
[image]

francesco-romano, 2018-02-26:
Do you need help in some particular part?
Regarding the thrift, an alternative can be to use a port monitor attached on the input port to translate from thrift to simple yarn vector (@traversaro for more info on the subject).
For the world frame, just be careful, that the Xsens frame changes for every calibration of the suit

Yeshasvitvs, 2018-02-26:
For now I am not using the human base pose and velocity in the controller. It is good to know about the usage of port monitor to get yarp vectors from thrift message. Thank you @francesco-romano I will update on the controller status after testing