gbionics/human-dynamics-estimation #83: yarpdataplayer not streaming all the ports from HDE dump

Yeshasvitvs, 2018-04-10:
The recent data dump from the online HDE with FT shoes online has ports from HDE and FT shoes in different directories. If all of this data dump is played using yarpdataplayer only the HDE ports are streamed but not the FT ports. But if the directories are separated and played in different yarpdataplayers, all the ports are streamed correctly. This problem is observed once and has to be tested more to be sure. After that the issue can be opened in yarp

lpenco, 2018-04-23:
I have the same problem!
Cannot stream multiple data from a single yarpdataplayer

Yeshasvitvs, 2018-06-07:
The hde_dumper data we collected last time has the following directories

[image]

The HDE directory contains the following subdirectories with each containing `data.log` and `info.log` files for `yarpdataplayer` playback
[image]

The forces directory has the following subdirectories
[image]
[image]

`xsensDriver` directory also has `data.log` and `info.log` but when this directory is opened with yarpdataplayer no ports are opened. **I am not sure what data is recorded here**

Now, when I load the `hde_dumper` whole directory with `yarpdataplayer` only the `HDE` data ports are played back normally

[image]

When `HDE` directory is loaded with yarpdataplayer all the ports are played back normally

[image]

When `forces` directory is loaded with yarpdataplayer all the ports are opened correctly but one port, 'totalForce' of right shoe is not played back while the `totalForce` of the left shoe is playedback normally.

[image]

This behavior is observed with only one data dump and I haven't tested if further with more data. This has to be looked further with future data collection during pHRI @claudia-lat @lucaTagliapietra @diegoferigo @DanielePucci

Yeshasvitvs, 2018-06-07:
@lpenco Do you still have the problem. If possible please post the data dump details here.

Yeshasvitvs, 2019-02-01:
I believe this is related to the time synchronization as indicated [here](https://github.com/robotology/yarp/issues/1687#issuecomment-394781017). As this issue is related to HDEv1 data this issue is not relevant anymore. So we can close it.

@claudia-lat @diegoferigo @DanielePucci

closed 2019-02-01 by Yeshasvitvs