gbionics/human-dynamics-estimation #85: HDE Wiki Page Images Usage

Yeshasvitvs, 2018-12-05:
We decided to use the [wiki pages](https://github.com/robotology/human-dynamics-estimation/wiki/Human-Dynamics-Estimation-Setup-and-Usage) for documenting the HDE usage, which can help a new user to get started quickly. Unfortunately the current possible way to use images, as [suggested by github](https://help.github.com/articles/adding-images-to-wikis/), is to upload the images to the repo and use the url. As @traversaro pointed out this is not a clean solution for a public repository like [human-dynamics-estimation](https://github.com/robotology/human-dynamics-estimation), particularly when the amount of images and their sizes are large. An alternative way is considered [here](http://mikehadlow.blogspot.com/2014/03/how-to-add-images-to-github-wiki.html), which uses the idea that a wiki of a git repository is itself a git repository. So, in case of future changes to be done for the wiki pages, an user has to clone the wiki repo and add the new images to it and then they can be used in the wiki pages. 

@claudia-lat @diegoferigo @lrapetti

Yeshasvitvs, 2018-12-05:
@DanielePucci I am not sure if this information is known to the other members of the lab. In any case, it is better to put it somewhere which can be accessed by all the loc2 members.

traversaro, 2018-12-06:
> Unfortunately the current possible way to use images, as suggested by github, is to upload the images to the repo and use the url.

In that page it does not seem that GitHub suggests adding the image to the repo, if I am not wrong.

lrapetti, 2018-12-06:
> @DanielePucci I am not sure if this information is known to the other members of the lab. In any case, it is better to put it somewhere which can be accessed by all the loc2 members.

For example @S-Dafarra is using closed issue for uploading the wiki images (https://github.com/loc2/lab-organisation/issues/103, https://github.com/loc2/lab-organisation/issues/104)

S-Dafarra, 2018-12-06:
It's a dirty trick to avoid increasing the weight of the repo :grin:

Yeshasvitvs, 2018-12-06:
> > Unfortunately the current possible way to use images, as suggested by github, is to upload the images to the repo and use the url.
> 
> In that page it does not seem that GitHub suggests adding the image to the repo, if I am not wrong.

My bad, It is my interpretation of that!

claudia-lat, 2018-12-06:
> It's a dirty trick to avoid increasing the weight of the repo 😁

I do it in this dirty way as well, especially if I know that the image won't be editable in the future.  In this case I upload the file (e.g., the `.svg`) in order to give the possibility to the other users to modify it.  I often attach many screenshots in my descriptions (e.g., like documenting a procedure) and I don't upload them as images in the repo.

Yeshasvitvs, 2018-12-06:
In the past I uploaded the [wiki-images](https://github.com/loc2/lab-organisation/tree/master/wiki-images) to the lab-organisation repo and added the images for [How to use two iCub robots in gazebo simulation](https://github.com/loc2/lab-organisation/wiki/How-to-use-two-iCub-robots-in-gazebo-simulation). As this repo is private and it is mainly used for organizational purposes I believe it is not a problem. @traversaro please correct if I am wrong and if the wiki-images have to be removed from the repo and stored in the wiki repo of it.

traversaro, 2018-12-06:
Yes, I think images in private repos not used to store code is not really a big problem (but obviously depends on the case).

Yeshasvitvs, 2018-12-06:
So I will leave the wiki-images of lab-organization there.

Yeshasvitvs, 2018-12-11:
Closing this issue as we decided to store the images for HDE wiki inside the wiki repo itself.

closed 2018-12-11 by Yeshasvitvs