gbionics/human-dynamics-estimation #88: Update hde-readonly-controlboard module to be compatible with HDEv2

Yeshasvitvs, 2019-01-31:
Currently [hde-readonly-controlboard](https://github.com/Yeshasvitvs/human-dynamics-estimation/tree/hde_interface/hde-readonly-controlboard) is not compatible with new HDEv2. In this issue we track the details related to updating it to be compatible with taking joint position, velocity and acceleration data coming from `HumanStateProvider` and joint torque data coming from `HumanDynamicsEstimator` to pass the data to matlab simulink

@lrapetti

Yeshasvitvs, 2019-01-31:
Not necessarily. The first one was designed as a module to work with modules but subscribing to the output ports streaming joint data in thrift format. Now I want to change it to a device called ‘HumanControlBorad’ that will read joint position, velocity through ‘IHumanState’ interface and the subscribe to joint torques output port from ‘HumanDynamicsWrapper’. Later when we implement ‘HumanDynamicsWrapper’ to expose data through ‘IHumanDynamics’ we can update from the logic of reading joint torques port to reading ‘IHumanDynamics’ interface.

diegoferigo, 2019-01-31:
If I understood what @Yeshasvitvs meant, the new `HumanControlBoard` will have the support of attaching to `IHumanState` and `IHumanDynamics` interfaces. This means that it will run within the same `yarprobotinterface` process (that means same xml file).

This conforms to the architectural choice of HDE v2. As we discussed for all the HDE v2 devices, as soon as we need network transparency we can develop the pairs wrapper-client for each interface. This would allow to run `HumanControlBoard` on a different process / different machine.

Yeshasvitvs, 2019-02-01:
[HumanControlBoard](https://github.com/Yeshasvitvs/human-dynamics-estimation/tree/feature/humanstateprovider_multi_thread_ik/devices/HumanControlBoard) is added to devices. I implemented only the core methods to access data in matlab simulink through controlboardremapper. Accordingly, we get joint positions and joint velocities from `IHumanState` interface and joint torques from `IHumanDynamics` interace. Ideally, joint accelerations should also be available through `IHumanDynamics` interface but currently they are not set in `HumanDynamicsEstimator`. So, I am setting them to zero in `HumanControlBoard` device.

On reading `<HumanControlBoard>/stateExt:o` data we can read correctly the joint positions, velocities, accelerations and torques indicated by `[ok]`  at the end of the data. But the other methods display `[fail]` as they are not implemented in `HumanControlBoard` and are not necessary to get human joint data in simulink.

[image]

This test is done with offline wearable data. I will test this correctly in simulink controller next with both offline and online data next.

Yeshasvitvs, 2019-02-03:
I added the tests for `HumanControlBoard` device [here](https://github.com/Yeshasvitvs/human-dynamics-estimation/tree/feature/humanstateprovider_multi_thread_ik/devices/HumanControlBoard/tests) and ran the test  [`HumanSimulinkController`](https://github.com/Yeshasvitvs/human-dynamics-estimation/tree/feature/humanstateprovider_multi_thread_ik/devices/HumanControlBoard/tests/HumanSimulinkController) with [offline wearable data](https://github.com/dic-iit/component_andy/tree/master/Wearable_Data/XSensSuit) from xsens suit device.

Accordingly, with the data set of **left knee** movement of 90 degrees, I see the correct joint angles in simulink scope

[image]

For the **right knee** movement of 90 degrees, I see the correct joint angles in simulink scope

[image]

Also, the joint torques values are read correctly on simulink side from the `HumanControlBoard` device

[image]

[image]

**NOTE:** The joint torque values may not be reasonable with respect to the knee joint angles. This is because the time synchronization differences in the playback of xsens werable data and the ftshoes wearable data.

A final test with online data from suit and shoes with a human subject has to be done before closing this issue. @lrapetti and I will do this test soon and update the issue.

Yeshasvitvs, 2019-02-06:
During this testing https://github.com/dic-iit/component_andy/issues/193#issuecomment-461117811 @lrapetti and I tested the new `HumanControlBoard` device with the human movements as shown below

- A set of arms to T pose 5 times
- A set of two feet squat 5 times

[image]

The corresponding **joint angle data in degree** seen in matlab scope is shown below

[image]

[image]

The corresponding **joint torques** seen in matlab scope is shown below

[image]

[image]

So, the simulink controllers gets the human data faithfully from `HumanControlBoard` device.

One key details a new user has to be aware is that in case of feet or knee values and direction being wrong in the matlab scope. The problem could be feet or knee distortion as observed in this case 
https://github.com/dic-iit/component_andy/issues/188#issuecomment-461118542. 

We can close this issue now and when we perform another test of HDEv2 with another [human subject and the corresponding urdf model](https://github.com/robotology-playground/human-gazebo), we can update this issue with the matlab plots from the test.

@DanielePucci @lrapetti

closed 2019-02-06 by Yeshasvitvs