gbionics/human-dynamics-estimation #89: Resolving problems for installing HDE on Ubuntu 16.04

kouroshD, 2019-02-01:
When I wanted to install the HDE from [this branch](https://github.com/Yeshasvitvs/human-dynamics-estimation/tree/feature/humanstateprovider_multi_thread_ik) (which soon will be merged to the master branch) I faced the following problems:

- I had errors regarding the unorder_map:  the compiler was gcc-5 for Ubuntu 16.04, and we upgrade the compiler to gcc-7, and it compiles fine.

- There were problems with eigen3 in humanStateProvide.cpp, we add `SYSTEM PRIVATE ${EIGEN3_INCLUDE_DIR}` the following lines of the CMakeLists `target_include_directories(HumanStateProvider PUBLIC
    $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}>) ` and the problem is resolved.

- In HumanDynamicsEstimator.cpp, if we substitute `#include <Eigen/SparseCholesky>` with `#include <eigen3/Eigen/SparseCholesky> `, the compiler will not throw warnings anymore.

cc @Yeshasvitvs @lrapetti

Yeshasvitvs, 2019-02-01:
@kouroshD please document also the details for setting up `ma27` solver and other issues related to human model urdfs encountered today.

traversaro, 2019-02-01:
Just to understand, why are you using Ubuntu 16.04 ?  The next releases of YARP will not support Ubuntu 16.04 (see https://github.com/robotology/yarp/blob/master/.github/CONTRIBUTING.md#supported-systems), and so except in cases where a given projects explicitly requires Ubuntu 16.04, we should aim to migrate to Ubuntu 18.04 .

kouroshD, 2019-02-02:
It was an old installation, I didn't know about this fact. I will migrate to Ubuntu 18.04 soon for the ubuntu machine I am using.

traversaro, 2019-02-02:
Great!

kouroshD, 2019-02-05:
iDynTree branch should be another branch, not master:
```
cd <idyntree>
git checkout -t feature/improvedRobotStatePublisher
```
build again the superbuild, but before change the configuartion to:
```
YCM_EP_DEVEL_MODE_iDynTree : ON
```

lrapetti, 2019-02-05:
> * [](https://github.com/Yeshasvitvs/wearables/tree/feature/ICub-device-impl)
> * change  `/devices/IAnalogSesorToIWear/src/IAnalogSensorToIWear.cpp`, becasue the current implementation is depended on yarp dev branch, by changing this lline it will be compatible with master branch:
>   updated lines:
> 
> ```
>     bool getData(std::vector<double>& vector, const size_t offset = 0)
>     {
> -        std::copy(buffer.data() + offset, buffer.end() + offset, vector.data());
> +        std::copy(buffer.data() + offset, buffer.data() + buffer.size() + offset, vector.data());
>         return true;
>     }
> ```

As discussed also with @diegoferigo, it may take long to get the release of the new `YARP` and for the moment it would be better to ensure the compatibility with the current branch of `YARP`.

> commet the line `#add_subdirectory(HumanControlBoard)` in file `HDE/devices/CMakeLists.txt` : it does not compile in macos currently.

We get the following error when trying to compile it in `MacOS`:
```
[ 42%] Linking CXX shared module ../../lib/yarp/HumanControlBoard.so
Undefined symbols for architecture x86_64:
  "iDynTree::ModelLoader::loadModelFromFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      hde::devices::HumanControlBoard::open(yarp::os::Searchable&) in HumanControlBoard.cpp.o
  "iDynTree::ModelLoader::model()", referenced from:
      hde::devices::HumanControlBoard::open(yarp::os::Searchable&) in HumanControlBoard.cpp.o
  "iDynTree::ModelLoader::isValid()", referenced from:
      hde::devices::HumanControlBoard::open(yarp::os::Searchable&) in HumanControlBoard.cpp.o
  "iDynTree::ModelLoader::ModelLoader()", referenced from:
      hde::devices::HumanControlBoard::open(yarp::os::Searchable&) in HumanControlBoard.cpp.o
  "iDynTree::ModelLoader::~ModelLoader()", referenced from:
      hde::devices::HumanControlBoard::open(yarp::os::Searchable&) in HumanControlBoard.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/yarp/HumanControlBoard.so] Error 1
make[1]: *** [devices/HumanControlBoard/CMakeFiles/HumanControlBoard.dir/all] Error 2
make: *** [all] Error 2
```

lrapetti, 2019-02-05:
> > * change  `/devices/IAnalogSesorToIWear/src/IAnalogSensorToIWear.cpp`, becasue the current implementation is depended on yarp dev branch, by changing this lline it will be compatible with master branch:
> >   updated lines:
> > 
> > ```
> >     bool getData(std::vector<double>& vector, const size_t offset = 0)
> >     {
> > -        std::copy(buffer.data() + offset, buffer.end() + offset, vector.data());
> > +        std::copy(buffer.data() + offset, buffer.data() + buffer.size() + offset, vector.data());
> >         return true;
> >     }
> > ```
> 
> As discussed also with @diegoferigo, it may take long to get the release of the new `YARP` and for the moment it would be better to ensure the compatibility with the current `master` branch of `YARP`.

His back-compatibility is introduced with https://github.com/Yeshasvitvs/wearables/pull/5/commits/8aa47b8b9d5b581ca1d64af7e6bd414864906fef. Soon to be merged

lrapetti, 2019-02-06:
> We get the following error when trying to compile it in `MacOS`:
>
> ```
> [ 42%] Linking CXX shared module ../../lib/yarp/HumanControlBoard.so
> Undefined symbols for architecture x86_64: [...]

Apparently, the problem here seems to be solved by adding `iDynTree::idyntree-estimation` to the `target_link_libraries` in addition to `iDynTree::idyntree-model` in order to use the `ModelLoader`.
I don't think this is meant to be working in this way.
Am I correct @traversaro? In case we can open an issue in `iDynTree`

traversaro, 2019-02-07:
The `iDynTree::ModelLoader` class is contained in the `iDynTree::idyntree-modelio-urdf` exported target (see https://github.com/robotology/idyntree/blob/114671fb1a012f6214b35b8462d463fea99781fd/src/model_io/urdf/CMakeLists.txt#L57). 

To be honest, I think it is wise to reccomended and document to just link against the list of exported targets `${iDynTree_LIBRARIES}`, that contains all the targets exported by iDynTree, so you are sure to always get everything. The mapping between the specific iDynTree symbols and their target is not properly documented anywhere, and unfortunatly at the moment is not easy to `deprecate` a CMake imported target, see https://gitlab.kitware.com/cmake/cmake/issues/17824, so having the symbols <--> target mapping as part of the public API of the library complicates the process of refactoring the symbols between targets (i.e. moving a class for a target to another).

kouroshD, 2019-02-18:
Since I have not faced more problems, we can close this issue.

closed 2019-02-18 by kouroshD