gbionics/human-dynamics-estimation #90: Investigate how to launch HDEv2 with real icub

Yeshasvitvs, 2019-02-01:
Currently, we tested HDEv2 with human and with human plus icub in simulation but we have never tried to HDEv2 with human and real robot yet. At the moment we are not clear on how to achieve this because xsens is launched on the windows machine that is connected to xsens router which is isolated from IIT network. So, in case of running HDEv2 with human and real icub, should we change the icub server settings so that it will be connected to xsense router or is there any other cleaner way to do it. 

In this issue we track the details related to setting up and running HDEv2 with human and real icub.

@S-Dafarra @traversaro @diegoferigo @lrapetti @kouroshD @claudia-lat

S-Dafarra, 2019-02-01:
What's the problem of connecting the XSens to the IIT network? I was actually thinking to connect the XSens router to the robot local network. We may ask Matteo for suggestions anyway.

Yeshasvitvs, 2019-02-01:
@S-Dafarra  From @claudia-lat previous experience connecting to the xsens router to IIT network disrupts the IIT network.

diegoferigo, 2019-02-01:
Yes we should ask @mbrunettini what it the best network interfaces configuration in Windows that allows:

- Having a network interface connected to the Xsens router
- Having a network interface connected (even through wifi) to the robot head

Since the `icub-srv` has www access, the second interface might provide capability of exiting to the www from the Windows pc.

---

As a side comment, @mbrunettini what is the constraint on windows of being connected to the Xsens router with the ethernet cable, and to the IIT network with the WiFi of the laptop? I never got it right.

S-Dafarra, 2019-02-01:
I would avoid modifying the robot wifi connection. If possible I would connect the XSens router to the i iCub switch via LAN cable. If it is a problem connecting Xsens to any IIT network, we may temporarily switch in external mode, disconnecting completely the robot from the IIT network and using only the server switch.

diegoferigo, 2019-02-01:
Just to be clear, I didn't propose to modify the robot wifi connection. What I meant was a configuration all related to the AnDy XPS laptop.

Yeshasvitvs, 2019-02-25:
@S-Dafarra helped me in understanding how the icub server is set in `home` and `travel` configuration. More details can be found here https://github.com/robotology/robotology.github.io/wiki/How-to-set-up-the-iCub-workstation-from-scratch

Assuming that the icub server is in `home` configuration, for the purpose of working together with the xsens system, we need to remove the lan cable connecting the IIT local network to the port named `INT` (internal). Then we need to connect a lan cable between the port `INT` and one of the many ports present on the top rack (black colored lan ports). 

Then we can connect the Xsens router to one of these ports. To make the xsens router discover the icub server we need to set xsens routers default `Default Gateway` value. So, we need to login to the xsens router and change the default gateway value to `10.0.0.1` which is the ip of the iCub server. For more information related to this check [here](https://support.opendns.com/hc/en-us/articles/228007727-Configuration-for-ASUS-RT-AC68U). In this way we can have a closed network between the icub server and xsens router.

@kouroshD we can try to check it this evening if this will work to our needs

S-Dafarra, 2019-02-25:
> To make the xsens router discover the icub server we need to set xsens routers default `Default Gateway` value.

If possible I would use the xsens router as a simple ``hub``. Having two routers in a network (the other is the iCub server) messes up the routing of packets.

Yeshasvitvs, 2019-02-25:
The process to change the mode of the xsens router is as indicated [here](https://www.asus.com/support/FAQ/1015009/)

Yeshasvitvs, 2019-02-26:
We got an answer from @mbrunettini  as:

```

Hi,
The Xsens Wifi Access Point is not configured as a router (that’s the default configurations), but as far as I know only as point-to-point connection between a pc (connected on a LAN  port) and the suit. There is no routing, gateway and other networking resources needed by the iCub subnet.
The only option I see to integrate that hardware on out setup is using a pc connected to both networks, such as a laptop connected via wired ETH to xsens and to iCub using wifi, using two cabled eth, one for xsens and one for iCub. On that pc we can run yarp and xsens software and send xsens data to other machines in the iCub environment.
Bye
 
Matteo

```

Also @mbrunettini advised that the solution in the previous comment https://github.com/robotology/human-dynamics-estimation/issues/90#issuecomment-467089360 may not work

```

I definitely don’t think it can work:
1)     the xsens subnet address pool is different from iCub’s
2)     there are services on the iCub server that can be harmed by this connection
3)     As far as I know we cannot change the xsens router operative mode
 
I strongly suggest to use my below option.
 
```

So, I removed the `Default Gateway` value `10.0.0.1` and left it blank as before.

Yeshasvitvs, 2019-02-26:
I tried to check how to get xsens and icub together on the same network but I cannot figure out what it means to have 

> The only option I see to integrate that hardware on out setup is using a pc connected to both networks, such as a laptop connected via wired ETH to xsens and to iCub using wifi, using two cabled [...]

I think it is best to request @mbrunettini to configure it out correctly tomorrow morning as soon as possible as it is a time sensitive matter to do experiments for iros 2019 submission.

@DanielePucci

mbrunettini, 2019-02-27:
Not on the same network, but 

> using a pc connected to both networks

there is no configuration to changes. Just connect a PC to both iCub and Xsens router. You'll get two different IP address on that PC, one each  network connnection.

mbrunettini, 2019-02-27:
> So, I removed the `Default Gateway` value `10.0.0.1` and left it blank as before.

PLEASE DO NOT CHANGE XSENS ROUTER CONFIGURATION: IT WAS CUSTOM MADE BY XSENS AND I DON'T KNOW ITS DETAILS. If you mess up sometheing, we need to ask xsens support.

Yeshasvitvs, 2019-02-27:
> Just connect a PC to both iCub and Xsens router. You'll get two different IP address on that PC, one each network connection.

Thanks @mbrunettini everything is worked correctly as you advised!

Yeshasvitvs, 2019-02-27:
The correct configuration steps we used are updated in [this issue](https://github.com/dic-iit/component_andy/issues/216). We can close this issue now.

closed 2019-02-27 by Yeshasvitvs