gbionics/human-dynamics-estimation #91: Cleaning HDEv2 for release

Yeshasvitvs, 2019-02-06:
The [human-dynamics-estimation](https://github.com/robotology/human-dynamics-estimation) repo originally contains HDEv1 modules. The [recent development](https://github.com/Yeshasvitvs/human-dynamics-estimation/tree/feature/humanstateprovider_multi_thread_ik) of HDEv2 is almost done and in the light of this, it is better to clean up HDEv1 related code. As suggested by @traversaro we can delete the files of HDEv1 as there will not be any support for HDEv1. In this issue, we track the details related to the cleaning of HDEv2 by removing HDEv1 related modules, configuration files, urdf files etc.

@diegoferigo @claudia-lat @lrapetti @DanielePucci

diegoferigo, 2019-02-07:
Yes this would be great!

Yeshasvitvs, 2019-02-07:
[This commit](https://github.com/Yeshasvitvs/human-dynamics-estimation/commit/8ef2d6ce6eaae5ec2ce7eb7b93739c9ce94079fc) deletes all the HDEv2 related modules and updates the HDEv2 cmake file accordingly.

diegoferigo, 2019-02-07:
@Yeshasvitvs Since we have tons of staging changes, and there are no current users of hde outside our group, what about:

1. Merging the current `release/HDEv2` to `master` (after minor commit renaming and squash, I can do it since they are mine)
2. "Wiping" the current leftovers of HDEv1
3. Start opening clear PRs on top of this situation

Yeshasvitvs, 2019-02-08:
@diegoferigo Sure, it will give a cleaner history. Although, I suggest to have a `debug` branch and merge  `release/HDEv2` to `debug`. It is due to the fact that we still have some development left to do on HDEv2. I can open a PR to `debug` branch to which we can push the current code base. Once we have all the basic infrastructure to run HDEv2 to conduct experiments we can push those changes from `debug` to `master`

 What do you think ?

diegoferigo, 2019-02-08:
I would merge directly to `master`. If we really want, we can create a tag with the latest working version of HDEv1 (`xsens-mvn` based).

Yeshasvitvs, 2019-02-08:
> I would merge directly to `master`. If we really want, we can create a tag with the latest working version of HDEv1 (`xsens-mvn` based).
> 
> In any case current `master` is not usable.

My bad! I was using `debug` instead of stating `devel` branch. So, my point is to have a `devel` branch to which we can do the PR from the development till now for HDEv2. You can merge `release/HDEv2` branch (after name changes) to the master but the upstream PR can be opened on `devel` branch. On approval and merging we can also push the changes from `devel` to `master`

Yeshasvitvs, 2019-02-08:
Also, the upstream has a lot of branches which needs to be cleaned.

[image]

diegoferigo, 2019-02-08:
> My bad! I was using `debug` instead of stating `devel` branch. So, my point is to have a `devel` branch to which we can do the PR from the development till now for HDEv2. You can merge [...]

I would avoid for the time being a `devel` branch. Let's start using it if and when we'll need features of dependencies only available in their `devel` branch.

DanielePucci, 2019-04-11:
Please @Yeshasvitvs merge [your fork ](https://github.com/Yeshasvitvs/human-dynamics-estimation/tree/feature/humanstateprovider_multi_thread_ik) with master of this repo, i.e. [human-dynamics-estimation](https://github.com/robotology/human-dynamics-estimation). As specified in https://github.com/robotology/human-dynamics-estimation/issues/98 please address also the license issue while merging 

CC
@traversaro @lrapetti

Yeshasvitvs, 2019-04-11:
The stable branch of HDEv2 from my fork is pushed to master branch of the upstream https://github.com/robotology/human-dynamics-estimation/commit/8cfc88c8fd9fad41ebe837e1c91c59f91df376b9 

cc @diegoferigo

diegoferigo, 2019-04-13:
@Yeshasvitvs Please next time before merging code try to double check if there are commits with a `WIP` label. If you don't want to squash them with other ones, at least rename them through an interactive rebase removing the label, its very ugly having them in the repo history :/

traversaro, 2019-04-13:
Sorry @diegoferigo , @Yeshasvitvs is innocent, it is actually my fault. As far as I understood, the current `master` was broken (i.e. there could not be any non-IIT users for the repo), and all the users in the lab where already effectively using the `Yeshasvitvs:feature/humanstateprovider_multi_thread_ik` branch as an effective  `master` (it was the branch used for any demo, and it was the branch on which the branches with new features by @lrapetti and @kouroshD were based). For this reason, I saw no problem in just making clear the current situation, in which the v1 is not maintained, and the `Yeshasvitvs:feature/humanstateprovider_multi_thread_ik` is the branch used for stable demos. As we could force push the `Yeshasvitvs:feature/humanstateprovider_multi_thread_ik`  branch before (knowing that the branches based on it needed to take this into account), we can as well force push the current `master` branch.

diegoferigo, 2019-04-13:
No problem from my side. The merge of the HDEv2 branch to master was pending since some while now, and the only missing bit was cleaning commits. But with demos, bug fixes and so on this cleaning has always been postponed.

Now I'm fine to force-push on master, even if it would be something to avoid. Though, I just want to clarify that if there were users outside IIT (and I do not think there are, not on our current master at least) and we force push, we would break all their local clones and forks. I'm not sure what is worse wrt to an outdated master that soon would have been fixed (after the merge of the new branch).

At this point, I would keep all the current commits without major squashing / reordering (that would be too demanding) and just clean the messages. Examples:

- https://github.com/robotology/human-dynamics-estimation/commit/9d7797c5f57fb8b578979e7a597625aadb72404c
- https://github.com/robotology/human-dynamics-estimation/commit/b7392e14164ed5cb6782344aa62e21eb0f972978

And possibly, squash commits close in time such as the ones the edit config files with all the attempts (example https://github.com/robotology/human-dynamics-estimation/commit/9fdaaf6403e70c40793d2a0731ff45f855e6f5f9).

DanielePucci, 2019-04-13:
@diegoferigo 
> Though, I just want to clarify that if there were users outside IIT (and I do not think there are, not on our current master at least) and we force push, we would break all their local clones and [...]

You are completely right. We took this decision also in light of the fact that basically no external IIT user (to the best of our knowledge) was using the code. It was an attempt to bound and start the recover for the overall mess around the HDE related code https://github.com/robotology/human-dynamics-estimation/issues/102

traversaro, 2019-04-15:
> Though, I just want to clarify that if there were users outside IIT (and I do not think there are, not on our current master at least) and we force push, we would break all their local clones and [...]

Personally I was not really concern of the outdated master, but rather  by the plethora of different branches used for  demos on day by day basis.

lrapetti, 2019-12-13:
I think HDEv2 was merged into master long time ago with https://github.com/robotology/human-dynamics-estimation/pull/100.
Closing this issue.

closed 2019-12-13 by lrapetti