gbionics/human-dynamics-estimation #93: Resolve problems  to install of ROS in MacOS for the HDE visualization

kouroshD, 2019-02-06:
I Want to install ROS on Mac (Mojave, Version 10.14) in order to visualize the outcomes of HDE, and in my case human state provider, and I have face several issues.
So I post it here the future issues I will face, later, I will add the issues I faced before and how I solved it.

CC @traversaro @diegoferigo @S-Dafarra @lrapetti @claudia-lat

kouroshD, 2019-02-06:
I have this problem, which a part of it is :
```
In file included from /Users/kdarvish/Desktop/ros_ws/src/bond_core/bondcpp/src/timeout.cpp:30:
In file included from /Users/kdarvish/Desktop/ros_ws/src/bond_core/bondcpp/include/bondcpp/timeout.h:33:
In file included from /Users/kdarvish/Desktop/ros_ws/install_isolated/include/ros/ros.h:38:
In file included from /Users/kdarvish/Desktop/ros_ws/install_isolated/include/ros/time.h:53:
In file included from /Users/kdarvish/Desktop/ros_ws/install_isolated/include/ros/platform.h:55:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/string:477:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/string_view:176:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/__string:56:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/algorithm:641:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/cstring:79:9: error: no member named 'strcoll' in the global namespace; did you mean 'strtoll'?
using ::strcoll;
      ~~^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/stdlib.h:172:3: note: 'strtoll' declared here
         strtoll(const char *__str, char **__endptr, int __base);
```
I followed suggestions provided [here](https://www.bountysource.com/issues/39690326-installation-indigo-on-osx-failed-while-building-nodelet_tutorial_math), but still threre is the problem.
@traversaro

kouroshD, 2019-02-06:
https://github.com/ros/rosdistro/issues/13333#issuecomment-274421336

traversaro, 2019-02-07:
Sorry, I just read this now. @kouroshD did you solved this issue?

diegoferigo, 2019-02-08:
After discussing f2f, I pointed @kouroshD to this Travis build that install ROS lunar in macOS HS.

https://github.com/diegoferigo/blockfactory-demo-ros/blob/afb767a92297a1b543284d7ee17fe7bde74c8752/.travis.yml#L66-L87

I just restarted the build to check that everything still works even today, and the build succeeded.

traversaro, 2019-02-08:
I guess @lrapetti was checking out https://github.com/mikepurvis/ros-install-osx .

lrapetti, 2019-02-08:
> I guess @lrapetti was checking out https://github.com/mikepurvis/ros-install-osx .

I have not tried yet, but I will in the next days.

As reference, it seems that there are still some problems (and fixes) in `High Sierra` (which I am currently using)(see https://github.com/mikepurvis/ros-install-osx/issues/116).

kouroshD, 2019-02-11:
@diegoferigo What you have written is similar what is mentioned in [ros official website](http://wiki.ros.org/kinetic/Installation/OSX/Homebrew/Source) and https://github.com/mikepurvis/ros-install-osx; but I had a lot of problems to install it.
I  would say I have resolved the problem followining https://github.com/mikepurvis/ros-install-osx.
In the next comment I will put the issues I had and how I solved them while following different instructions, some of the responses I know It is not clean, but finally it worked.

At the end, I could not install 3 packages, so I exclude them from my ros installation: diagnostics, pcl_ros, stereo_image_proc.

kouroshD, 2019-02-11:
Error in   brew tap ros/deps:

Related to a deprecated function call:

I have forked the repo and made changes.
$ brew tap kouroshD/homebrew-deps

$ install --from-paths src --ignore-src --rosdistro lunar -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
actionlib: No definition of [python-wxtools] for OS [osx]
roslisp: No definition of [google-mock] for OS [osx]
catkin: No definition of [google-mock] for OS [osx]

$ sudo pip install -U wxpython

$ rosdep install --from-paths src --ignore-src --skip-keys google-mock --rosdistro lunar -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
actionlib: No definition of [python-wxtools] for OS [osx]

$ brew install wxpython

$ rosdep install --from-paths src --ignore-src --skip-keys google-mock python-wxtools --rosdistro lunar -y
given path 'python-wxtools' does not exist

$ rosdep install --from-paths src --ignore-src --skip-keys google-mock --skip-keys python-wxtools --rosdistro lunar -y

—> Error: 
==> libtool
In order to prevent conflicts with Apple's own libtool we have prepended a "g"
so, you have instead: glibtool and glibtoolize.
ERROR: the following rosdeps failed to install
  homebrew: Failed to detect successful installation of [qt5]

$ brew info pyqt5
pyqt: stable 5.10.1 (bottled)
Python bindings for v5 of Qt
https://www.riverbankcomputing.com/software/pyqt/download5
Not installed
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/pyqt.rb
==> Dependencies
Required: python ✔, python@2 ✔, qt ✔, sip ✘
==> Analytics
install: 4,966 (30 days), 13,552 (90 days), 78,894 (365 days)
install_on_request: 3,061 (30 days), 9,194 (90 days), 49,793 (365 days)
build_error: 0 (30 days)

$ brew install pyqt
$ brew install pyqt5

$ brew info pyqt5
pyqt: stable 5.10.1 (bottled)
Python bindings for v5 of Qt
https://www.riverbankcomputing.com/software/pyqt/download5
/usr/local/Cellar/pyqt/5.10.1_1 (985 files, 45.0MB) *
  Poured from bottle on 2019-02-05 at 21:21:46
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/pyqt.rb
==> Dependencies
Required: python ✔, python@2 ✔, qt ✔, sip ✔
==> Analytics
install: 4,966 (30 days), 13,552 (90 days), 78,894 (365 days)
install_on_request: 3,061 (30 days), 9,194 (90 days), 49,793 (365 days)
build_error: 0 (30 days)

$ brew install qt5
$ brew link --force qt

$  rosdep install --from-paths src --ignore-src --skip-keys google-mock --skip-keys python-wxtools --rosdistro lunar -y

[... 528 lines cut ...]
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/kdarvish/Documents/ros-install-osx/lunar_desktop_full_ws/devel/.private/pcl_ros/lib/libpcl_ros_features.dylib] Error 1
make[1]: *** [CMakeFiles/pcl_ros_features.dir/all] Error 2
make: *** [all] Error 2
cd /Users/kdarvish/Documents/ros-install-osx/lunar_desktop_full_ws/build/pcl_ros; catkin build --get-env pcl_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

Solution
https://stackoverflow.com/questions/53776238/building-point-cloud-library-pcl-on-macos-x-mojave-error-ld-library-not-foun

librviz_tutorial                                [ Unrelated job failed ]
Abandoned <<< rqt_rviz                                        [ Unrelated job failed ]
Abandoned <<< rviz_plugin_tutorials                           [ Unrelated job failed ]
Abandoned <<< rviz_python_tutorial                            [ Unrelated job failed ]
Abandoned <<< urdf_tutorial

kdarvish@/opt/ros/lunar/bin$ rviz
dyld: Symbol not found: __cg_jpeg_resync_to_restart
  Referenced from: /System/Library/Frameworks/ImageIO.framework/Versions/A/ImageIO
  Expected in: /usr/local/lib/libJPEG.dylib
 in /System/Library/Frameworks/ImageIO.framework/Versions/A/ImageIO
[1]    21295 abort      rviz

https://github.com/sgrif/pq-sys/issues/1

EDITOR='open -Wne' sudo -e /Users/kdarvish/Documents/ros-install-osx/lunar_desktop_full_ws/build/pcl_ros/CMakeFiles/pcl_ros_features.dir/link.txt

lrapetti, 2019-02-22:
I was able to install `ros` in `macOS` (Version: `10.13.6` `High Sierra`) using https://github.com/mikepurvis/ros-install-osx, and solving some issues that came out (the issue where mostly documented in https://github.com/mikepurvis/ros-install-osx/issues/116).

I have installed the `lunar-Desktop` (setting [`ROS_DISTRO=lunar`](https://github.com/mikepurvis/ros-install-osx/blob/master/install#L25) and [`ROS_CONFIGURATION=desktop`](https://github.com/mikepurvis/ros-install-osx/blob/master/install#L26)). Moreover, I have removed some packages that were giving me problems, I have removed
 - `vision_opencv`
 - `rqt`

those packages were giving some problems and should not be required for `human-dynamics-estimation`.

Then the other problem I have faced during the installation are the following:
- **XQuartz**
```
Already up-to-date.
==> brew cask install caskroom/cask/brew-cask 
Error: Cask 'brew-cask' is unavailable: '/usr/local/Homebrew/Library/Taps/caskroom/homebrew-cask/Casks/brew-cask.rb' does not exist.
Warning: Cask 'xquartz' is already installed.

To re-install xquartz, run:
  brew cask reinstall xquartz
Log out and in to finalize XQuartz setup.
```
Solved by commenting out `XQuartz` in [`install`](https://github.com/mikepurvis/ros-install-osx/blob/master/install):
```
# XQuartz
#  if ! hash xquartz 2>/dev/null; then
#    brew install caskroom/cask/brew-cask
#    brew cask install xquartz
#    echo "Log out and in to finalize XQuartz setup."
#    exit 0
#    fi
```

- **google-mock**
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
roslisp: No definition of [google-mock] for OS [osx]
catkin: No definition of [google-mock] for OS [osx]
```
Solved by adding `--skip-keys google-mock` to`rosdep install` in [`install`](https://github.com/mikepurvis/ros-install-osx/blob/master/install):
```
rosdep install --from-paths . --ignore-src --rosdistro ${ROS_DISTRO} -y --as-root pip:no --skip-keys google-mock
``` 

- **logError**, **logWarn**
```
Errors     << tf2:make /Users/lrapetti/Software/ros-install-osx/lunar_desktop_ws/logs/tf2/build.make.000.log                                                                                           
/Users/lrapetti/Software/ros-install-osx/lunar_desktop_ws/src/geometry2/tf2/src/buffer_core.cpp:126:5: error: use of undeclared identifier 'logWarn'
    logWarn("%s",ss.str().c_str());
    ^
/Users/lrapetti/Software/ros-install-osx/lunar_desktop_ws/src/geometry2/tf2/src/buffer_core.cpp:134:5: error: use of undeclared identifier 'logWarn'
    logWarn("%s",ss.str().c_str());
    ^
/Users/lrapetti/Software/ros-install-osx/lunar_desktop_ws/src/geometry2/tf2/src/buffer_core.cpp:221:5: error: use of undeclared identifier 'logError'
    logError("TF_SELF_TRANSFORM: Ignoring transform from authority \"%s\" with frame_id and child_frame_id  \"%s\" because they are the same",  authority.c_str(), stripped.child_frame_id.c_str());
    ^
/Users/lrapetti/Software/ros-install-osx/lunar_desktop_ws/src/geometry2/tf2/src/buffer_core.cpp:227:5: error: use of undeclared identifier 'logError'
    logError("TF_NO_CHILD_FRAME_ID: Ignoring transform from authority \"%s\" because child_frame_id not set ", authority.c_str());
    ^
/Users/lrapetti/Software/ros-install-osx/lunar_desktop_ws/src/geometry2/tf2/src/buffer_core.cpp:233:5: error: use of undeclared identifier 'logError'
    logError("TF_NO_FRAME_ID: Ignoring transform with child_frame_id \"%s\"  from authority \"%s\" because frame_id not set", stripped.child_frame_id.c_str(), authority.c_str());
    ^
/Users/lrapetti/Software/ros-install-osx/lunar_desktop_ws/src/geometry2/tf2/src/buffer_core.cpp:240:5: error: use of undeclared identifier 'logError'
    logError("TF_NAN_INPUT: Ignoring transform for child_frame_id \"%s\" from authority \"%s\" because of a nan value in the transform (%f %f %f) (%f %f %f %f)",
    ^
/Users/lrapetti/Software/ros-install-osx/lunar_desktop_ws/src/geometry2/tf2/src/buffer_core.cpp:255:5: error: use of undeclared identifier 'logError'
    logError("TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id \"%s\" from authority \"%s\" because of an invalid quaternion in the transform (%f %f %f %f)",
    ^
/Users/lrapetti/Software/ros-install-osx/lunar_desktop_ws/src/geometry2/tf2/src/buffer_core.cpp:277:7: error: use of undeclared identifier 'logWarn'
      logWarn("TF_OLD_DATA ignoring data from the past for frame %s at time %g according to authority %s\nPossible reasons are listed at http://wiki.ros.org/tf/Errors%%20explained", stripped.child_frame_id.c_str(), stripped.header.stamp.toSec(), authority.c_str());
      ^
[... 58 lines cut ...]
```
Same as before:
```
/ros-install-osx/lunar_desktop_full_ws$ catkin build --start-with rospy
```

- **rospkg**
```
Errors     << dynamic_reconfigure:make /Users/lrapetti/Software/ros-install-osx/lunar_desktop_full_ws/logs/dynamic_reconfigure/build.make.000.log                                                                                                 
Traceback (most recent call last):
  File "/Users/lrapetti/Software/ros-install-osx/lunar_desktop_full_ws/src/dynamic_reconfigure/cfg/Test.cfg", line 37, in <module>
    from dynamic_reconfigure.parameter_generator_catkin import *
  File "/Users/lrapetti/Software/ros-install-osx/lunar_desktop_full_ws/devel/.private/dynamic_reconfigure/lib/python2.7/site-packages/dynamic_reconfigure/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 38, in <module>
  File "/Users/lrapetti/Software/ros/lib/python2.7/site-packages/roslib/__init__.py", line 50, in <module>
    from roslib.launcher import load_manifest
  File "/Users/lrapetti/Software/ros/lib/python2.7/site-packages/roslib/launcher.py", line 42, in <module>
    import rospkg
ModuleNotFoundError: No module named 'rospkg'
```
It can be solved by `pip` installing `rospkg`:
```
sudo -H python2.7 -m pip install -U rospkg
```

lrapetti, 2019-02-22:
After the installation presented in the previous comment, I was able to run the tests for `ros`, and for the communication of `ros` with `YARP`.

However, I am still having some problem in visualizing the model when running `HDE` in `rviz`. I can see the `TF`  moving in `rviz`, and I am not getting any error, however I am not able to see the model:
[image]

lrapetti, 2019-02-22:
I checked again with @Yeshasvitvs, and I was able to visualize the model (the problem was probably due to the way I was launching it). So I would say the installation of `ROS` and `rviz` was successful in my case since I was able to run it in the `HDE` framework.

DanielePucci, 2019-02-25:
closing #93

closed 2019-02-25 by DanielePucci

lrapetti, 2019-02-26:
> I checked again with @Yeshasvitvs, and I was able to visualize the model (the problem was probably due to the way I was launching it). So I would say the installation of `ROS` and `rviz` was successful in my case since I was able to run it in the `HDE` framework.

I was experiencing again the model not being loaded. At the moment I don't know what it is due to, since I am not getting any error. However in case it is not visualized, it is sufficient to try to run again `rviz` (sometimes it has to be run multiple times before being able to visualize it).

lrapetti, 2019-09-20:
Currently the usage of for installing ROS on macOS is broken mainly due to https://github.com/mikepurvis/ros-install-osx/issues/140. I was able to install `ros_comm` + all the dependencies of `rviz` (including the messages) but not `rviz` itself, so that I am able to run everything (roscore, publisher) a part for the visualization. In order to be able to compile also `rviz` the solution is to manually compile `sip` and `pyqt` with `python2`.

lrapetti, 2019-10-08:
Unfortunately, in order to install ROS, it is required to tap an older version of `boost` (see comment https://github.com/mikepurvis/ros-install-osx/issues/116#issuecomment-386189047) that seems to be breaking `gazebo`