gbionics/human-dynamics-estimation #95: Avoiding key pitfalls in setting up HDEv2 demo

Yeshasvitvs, 2019-02-19:
This issue highlights some of the observations we made during the demo setups of HDE.
#### Problem: Yarp network cannot be detected on other machines connected to the xsens router network

**Solution:** Double check that that **yarp namespace** is correctly set and see if the **ip** address of the machine where yarpserver is running is correctly set into the `<namespace>.conf` file. This typically is of the form `ip 100000 yarp`. Once this is set correctly, `yarp detect` will be able to discover the yarp server.

#### Yarp ports cannot be found

**Solution:** Typically when the main machine on which yarp server is run is connected to the xsens router after switching the router off and on, the yarp ports of newly launched yarp processes (like wearable devices) cannot be discovered. The likely reason for this is due to the fact that a pre running yarp server on the machine may be used to open the ports and are hence not discovered by the newly setup network. A clean solution to avoid this problem is to restart all the machines to be connected to xsens network everytime the xsens router is turned off and on.

@lrapetti please add other missing details we observed today while setting up for tread mill video

@claudia-lat @DanielePucci

traversaro, 2019-02-19:
If you do not know where to store this info, I suggest to add a short FAQ document, that you can link from the README of the repo.

Yeshasvitvs, 2019-02-19:
This info can be put in readme directly. I opened this issues just to keep track of things happening now in setting up HDEv2 demo

lrapetti, 2019-02-20:
Also https://github.com/dic-iit/component_andy/issues/168 is related to this kind of network configuration problems

Yeshasvitvs, 2019-02-21:
Yesterday during the recording of videos to demonstrate floating base HDE with walking on treadmill we noticed that the the ''piedi storti'' problem (distorted feet) https://github.com/dic-iit/component_andy/issues/188#issuecomment-465644687. A key detail we believe that will help to avoid this problem is to make sure that xsens calibration is of `Good` quality. This can be enforced by setting the [minimum-calibration-quality-required](https://github.com/robotology-playground/wearables/blob/f6999a0f5f36161b4aa4bbdc32ec7bbe2f508092/conf/Xsens/xsens_windows_all_wrappers.xml#L21) to `Good`. Also, yesterday we had to do the calibration without wearing the FTShoes and with the [default-calibration-type](https://github.com/robotology-playground/wearables/blob/f6999a0f5f36161b4aa4bbdc32ec7bbe2f508092/conf/Xsens/xsens_windows_all_wrappers.xml#L18) set to `NposeWalk` option.

claudia-lat, 2020-11-16:
We can close this issue since we did no longer experienced this problem.

closed 2020-11-16 by claudia-lat