gbionics/human-dynamics-estimation #96: Resolve problem of pairing the (fake) links in human state provider

kouroshD, 2019-02-21:
Trying to make pairs between the fake links introduced [here](https://github.com/dic-iit/component_wholebody-teleoperation/issues/161#issuecomment-465918910) , we noticed that it it can't make correctly the pairs, for all links. Investigating more, on the [code of the human state provider](https://github.com/kouroshD/human-dynamics-estimation/commit/c3bd983a60cb03af51dca18c890009e8f0af4dc5#diff-af5a41cc9a872c1dd7da746d7ad6d0f7) in [function `static void createEndEffectorsPairs`](https://github.com/kouroshD/human-dynamics-estimation/commit/c3bd983a60cb03af51dca18c890009e8f0af4dc5#diff-af5a41cc9a872c1dd7da746d7ad6d0f7R635) we understood that it guarantees that it will find a pair, but it could not guarantee the minimum distance neighborhood (at least) in the case of adding the fake links. Therefore, we modified the code , such that now it can find minimum distance link pairs (but still I can not guarantee that it always find the minimum distance pair of the links). In any case, I suggest later, we can merge [this commit](https://github.com/kouroshD/human-dynamics-estimation/commit/c3bd983a60cb03af51dca18c890009e8f0af4dc5#diff-af5a41cc9a872c1dd7da746d7ad6d0f7), as it have better performance and more predictable output.

Yeshasvitvs, 2019-02-21:
Also, we need to update the function name from `createEndEffectorsPairs` to something like `createLinkPairs`

kouroshD, 2019-04-16:
Since we have found and resolved the problem as mentioned in the first comment, I close this issue.

closed 2019-04-16 by kouroshD