gbionics/human-dynamics-estimation #98: Add LICENSE to the code repo

traversaro, 2019-04-08:
The headers refer to LGPL2+ license, but the repo does not contain a LICENSE.* file. I think you can refer to the blockfactory repo for an example of a repo licensed under LGPL2+.

traversaro, 2019-04-08:
cc @DanielePucci can we find someone to address this? Thanks!

DanielePucci, 2019-04-08:
Let's talk about this in the next AnDy iteration meeting. @claudia-lat please take care of putting it into the agenda, thanks a lot

DanielePucci, 2019-04-08:
Thanks @traversaro for opening a task on this https://github.com/dic-iit/component_andy/issues/231

Yeshasvitvs, 2019-04-08:
@traversaro just to be clear, you are referring to the code related to HDEv2 right? If possible please tag the code you are referring to. Thanks

traversaro, 2019-04-08:
I am referring to the repo as an whole, in particular most of visitors will just visit the repo as it is, so the master branch.

Yeshasvitvs, 2019-04-08:
The upstream is not cleaned up yet and updated to HDEv2. The cleanup details will be tracked in #91

traversaro, 2019-04-08:
If having the master pointing to HDEv2 is planned to still take some time, I think we can just add a simple LICENSE for LGPL2 to the master branch in he meanwhile. By the way, what is the blocker of https://github.com/robotology/human-dynamics-estimation/issues/91 ?

Yeshasvitvs, 2019-04-08:
Nothing is blocking at the moment but still there is a bit of development to do such as #99. Also, the history of HDEv2 code base is not straight and clean as it is on different users forks, we will get to it soon.

traversaro, 2019-04-08:
Depending on your plans, if you think this will still take some time, I suggest to edit the README's on the  master branch to explain that the repo in the master branch in its current form cannot be used, perhaps linking to this issue.

Yeshasvitvs, 2019-04-08:
Ok 😉 I will add a note on top of the current readme indicating the development of HDEv2

Yeshasvitvs, 2019-04-08:
Master branch README is updated with a note https://github.com/robotology/human-dynamics-estimation/commit/7bae305e75003b68d2721fcb8e2f95b19f39a1a0

Yeshasvitvs, 2019-04-11:
LGPL-2.1 code LICENSE added https://github.com/robotology/human-dynamics-estimation/commit/29299b98064f87a6112fc5c344251790cafedfa5

@traversaro @DanielePucci @diegoferigo @lrapetti @claudia-lat

Yeshasvitvs, 2019-04-11:
The header note on master branch is updated https://github.com/robotology/human-dynamics-estimation/commit/3f6e58f2dde3a37013a274e408dab60714cb2f27

traversaro, 2019-04-11:
Fixed by https://github.com/robotology/human-dynamics-estimation/commit/29299b98064f87a6112fc5c344251790cafedfa5 .

closed 2019-04-11 by traversaro