gbionics/human-dynamics-estimation #99: Update HumanWrenchProvider with correct wrench transformation from robot hands

Yeshasvitvs, 2019-04-08:
After identifying that the wrench transformation is incorrect https://github.com/dic-iit/component_andy/issues/153#issuecomment-479478038, and after implementing new joint sensor `VirtualJointKinSensor` in `ICub` wearables device https://github.com/robotology-playground/wearables/issues/32, in this issue we track the details related to updating the `HumanWrenchProvider` with the correct information coming from the `HumanStateProvider` and `ICub` wearable device to compute the correct wrench tranformation from the hands of the robot to the hands of the human.

Yeshasvitvs, 2019-04-10:
Things to check:

- [ ] Check if the transformation is correct
- [ ] Check the joint torque changes  in upper body through Rviz visualization
- [ ] Check the joint torque changes in upper body through matlab plots

traversaro, 2019-04-10:
Recap of Telegram discussion with @Yeshasvitvs (HRH HumanRightHand,  LH: l_hand, LHDF  : l_hand_dh_frame, RL: root_link)
* The frame in which  the force/torque published on `/wholeBodyDynamics/left_arm/cartesianEndEffectorWrench:o` is origin of `l_hand_dh_frame`, orientaion of `root_link` , i.e. ${}^{LHDF[RL]}\mathrm{f}^x_{LH}$ (see https://github.com/robotology/robots-configuration/blob/devel_iCubGenova04/iCubGenova04/estimators/wholebodydynamics.xml and WBD_OUTPUT_EXTERNAL_WRENCH_PORTS docs in http://wiki.icub.org/codyco/dox/html/classyarp_1_1dev_1_1WholeBodyDynamicsDevice.html)

Yeshasvitvs, 2019-04-10:
The correct method to compute the relative transformation for the robot hands to robot feet is [getRelativeTransform](http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1KinDynComputations.html#af35ca9a0b82e0481a477ae8717c10ff2) with four signatures! Thanks @traversaro :grinning: 

@kouroshD Still a bit of work to do tomorrow! Thanks for the incredible support today!

Yeshasvitvs, 2019-04-11:
`HumanWrenchProvider` device of HDEv2 will receive human joint data from `HumanStateProvider` and robot joint data from `ICub` wearable device. Concerning the transformation from wrench at the hands of the robot to the hands of the human, we need to compute three different transformations:

- The first transformation from the robot hands to the robot feet i.e `RobotFeet_H_RobotHands`. This can be achieved using iDynTree KinDyn class by setting the model and updating the state using the joint information coming from the `ICub` wearable device. A tricky detail here is concerning the `RobotHand` frame. The wrenches at the hands are given by [WBD_OUTPUT_EXTERNAL_WRENCH_PORTS](https://github.com/robotology/robots-configuration/blob/5034a233bc726cf2965bbf949b531b38db336fa6/iCubGenova04/estimators/wholebodydynamics.xml#L63) of `WholeBodyDynamics` config file. The order of the frames defined for the ports is documented [here](http://wiki.icub.org/codyco/dox/html/classyarp_1_1dev_1_1WholeBodyDynamicsDevice.html)
    - The first element is the link of which the published external wrench is applied. 
    - This wrench is expressed around the origin of the frame named as the second parameter,
    -  With the orientation of the third parameter.

So, on applications like `HDE` where `HumanWrenchProvider` depends on the frames used for a wrench port, the order has to be correctly considered to obtain the wrench transformation. At the moment, there is no way to communicate this frame order information used for `WholeBodyDynamics` information. In the test with `iCubGenova04` (greeny), the `ICub` wearable device is launched by subscribing to the hands wrenches through [frames defined here](https://github.com/robotology/robots-configuration/blob/5034a233bc726cf2965bbf949b531b38db336fa6/iCubGenova04/estimators/wholebodydynamics.xml#L64). Accordingly, the `RobotFeet_H_RobotHand` transformation actually becomes `l_sole_H_l_hand_dh_frame[root_link]`, where the hand frame is mixed with the origin of `l_hand_dh_frame` with the orientation of `root_link`. This can be computed using [getRelativeTransformExplicit](http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1KinDynComputations.html#af35ca9a0b82e0481a477ae8717c10ff2) method as `getRelativeTransformExplicit(l_sole,l_sole,l_hand_dh_frame,root_link)`. @traversaro please correct if you notice any mistakes. 

- The second transformation is a known fixed transformation from the hands of feet of the robot to the feet of the human i.e. `HumanFeet_H_RobotFeet`. At the moment with the pHRI standup experiment, this is hardcoded in the configuration file.
- The third and last transformation is from human feet to the human hand i.e. `HumanHand_H_HumanFeet`. This can be achieved by using the human model and human joint information.

Yeshasvitvs, 2019-04-12:
`HumanWrenchProvider` is updated with the correct transformation https://github.com/robotology/human-dynamics-estimation/commit/83fe259f7a580b1f957d4e261e0e0fdae6a4f350 with the correct frames in the configuration file https://github.com/robotology/human-dynamics-estimation/commit/05d0656941f79fcc3b972e6db5b23ac46b140633. The hands wrench visualization is a

[image]

[image]

Yeshasvitvs, 2019-04-12:
The above test is done without `HumanDynamicsEstimator` device. The joint torque estimates are correctly changing according to the changes at the hands wrench. During this testing, I noticed that rviz freezes sometimes due to `tfs` unavailability. An issue is opened to investigate and fix that issue. #103

Yeshasvitvs, 2019-04-15:
Today @lrapetti and I tested `HDE` with `pHRI` configuration file. My machine is crashing a lot today, probably due to the changes made to fix the fancontrol (I am trying to rollback by removing the changes made). So, we tested `HDE` with complete `pHRI` configuration file and Rviz run on @lrapetti machine. The Ik was  a bit slow ~300ms and hence the visualization is a bit laggy but all the wrench transformations and joint torque estimates in the upper body are correct. So, we can close this issue.

[image]

Additional tuning related to improving rviz visualization will be tracking in #103

closed 2019-04-15 by Yeshasvitvs