gbionics/human-dynamics-estimation #101: Clean up branches

lrapetti, 2019-04-11:
GIven the last activities of [`Cleaning HDEv2`](https://github.com/robotology/human-dynamics-estimation/issues/91) we deprecated `Human-Dynamics-Estimation version 1`.

I think it would be good to take the opportunity to archive all the stale branches in `robotology/human-dynamics-estimation` that were implemented on top of HDEv1:
- [ ] `feature/floatingBase`
- [ ] `feature/HumanRobotPose`
- [ ] `demo/AnDyY1`
- [ ] `feature/floating_base`
- [ ] `imu_lin_acc_in_y`
- [ ] `human_joint_acc_in_y`
- [ ] `updateToUseIDynTree0_8`
- [ ] `all_accelerations_in_y`
- [ ] `ftShoes_only_human`
- [ ] `clang-format`
- [ ] `release/HDEv2` 

@Yeshasvitvs @diegoferigo @claudia-lat 
Is there any particular concern with any of those branches.

CC @traversaro @DanielePucci

DanielePucci, 2019-05-06:
As soon as we push a working version with integration IK, we'll clean all the branches here.

lrapetti, 2019-12-13:
HDEv2 is now merged (https://github.com/robotology/human-dynamics-estimation/issues/91), I think that the branches based on HDEv1 mentioned [above](https://github.com/robotology/human-dynamics-estimation/issues/101#issue-432230936) can be safely archived.
@Yeshasvitvs

Yeshasvitvs, 2020-01-27:
@lrapetti @kouroshD 

I would like to start the cleanup for this repo starting with the removal of stale branches as pointed out in https://github.com/robotology/human-dynamics-estimation/issues/101#issue-432230936. The following branches are at least one to two years old and are not longer relevant to the current code base. 

### Stale Branches 

- [ftShoes_only_human](https://github.com/robotology/human-dynamics-estimation/tree/ftShoes_only_human)
- [all_accelerations_in_y](https://github.com/robotology/human-dynamics-estimation/tree/all_accelerations_in_y)
- [human_joint_acc_in_y](https://github.com/robotology/human-dynamics-estimation/tree/human_joint_acc_in_y)
- [imu_lin_acc_in_y](https://github.com/robotology/human-dynamics-estimation/tree/imu_lin_acc_in_y)
- [feature/floating_base](https://github.com/robotology/human-dynamics-estimation/tree/feature/floating_base)
- [demo/AnDyY1](https://github.com/robotology/human-dynamics-estimation/tree/demo/AnDyY1)
- [feature/floatingBase](https://github.com/robotology/human-dynamics-estimation/tree/feature/floatingBase)
- [clang-format](https://github.com/robotology/human-dynamics-estimation/tree/clang-format)

As agree with @lrapetti and @kouroshD the above branches will be deleted.

Yeshasvitvs, 2020-01-27:
After deleting the stale branches, we need to align the other branches to stable branches that can be used for demo purposes. Currently, the code base of this repo is used along the following two research axes and for two demos. At the current state `devel` is the base code for all the other branches. 

### Teleoperation

The branch that is fully functional for showing teleoperation demos is [add-secondary-calibration](https://github.com/robotology/human-dynamics-estimation/tree/add-secondary-calibration). So, this  branch needs to be merged to the devel branch.

[devel](https://github.com/robotology/human-dynamics-estimation/tree/devel) <------------ [add-secondary-calibration](https://github.com/robotology/human-dynamics-estimation/tree/add-secondary-calibration)

The other branches that are recently opened for teleoperation research axes are the following:

- [feature/AddQP](https://github.com/robotology/human-dynamics-estimation/tree/feature/AddQP)
- [feature/tmpAddRetargetting](https://github.com/robotology/human-dynamics-estimation/tree/feature/tmpAddRetargetting)
- [integration_IK_singularity](https://github.com/robotology/human-dynamics-estimation/tree/feature/integration_IK_singularity)
- [feature/teleoperation](https://github.com/robotology/human-dynamics-estimation/tree/feature/teleoperation)

@kouroshD @lrapetti can you please go over these branches and check if they are still relevant to the current code base. If they are, please open appropriate PRs.

### Human Dynamics Visualization

The branch that implemented the feature of external wrench estimation through stack of tasks berdy implementation is  [feature/visualize-berdy-estimated-wrench](https://github.com/robotology/human-dynamics-estimation/tree/feature/visualize-berdy-estimated-wrench). A new branch that tracks the additional details of calibration done for AnDy suit video are available in [new-offset-removal-options](https://github.com/robotology/human-dynamics-estimation/tree/new-offset-removal-options). So, the code alignment for this work is  

[devel](https://github.com/robotology/human-dynamics-estimation/tree/devel) <------------ [feature/visualize-berdy-estimated-wrench](https://github.com/robotology/human-dynamics-estimation/tree/feature/visualize-berdy-estimated-wrench) <------------ [new-offset-removal-options](https://github.com/robotology/human-dynamics-estimation/tree/new-offset-removal-options)

@lrapetti can you please check if [test_demo](https://github.com/robotology/human-dynamics-estimation/tree/test_demo) branch is still relevant or can we delete it.

lrapetti, 2020-01-28:
- [integration_IK_singularity](https://github.com/robotology/human-dynamics-estimation/tree/feature/integration_IK_singularity)
- [test_demo](https://github.com/robotology/human-dynamics-estimation/tree/test_demo)

This two branch can be safely removed.

Yeshasvitvs, 2020-01-30:
Following PR https://github.com/robotology/human-dynamics-estimation/pull/156, `new-offset-removal-options` branch is deleted.

Yeshasvitvs, 2020-01-30:
The `master` branch is rebased on the `devel` branch and then merged to it https://github.com/robotology/human-dynamics-estimation/commit/fd3e9042b59d6c37730327bbe152c03822ef3708

Both the `master` and `devel` branches are aligned. I tested the `devel` branch and the standard HDE demo is working without any errors

[image]

CC @kouroshD it is better to do a quick test for retargeting with this branch when you have time.Thanks.

lrapetti, 2020-01-31:
[`add-secondary-calibration`](https://github.com/robotology/human-dynamics-estimation/tree/add-secondary-calibration] branch is now merged (https://github.com/robotology/human-dynamics-estimation/pull/164) into `master`, so that branch can be deleted (I would do it but it seems I don't have the rights.

I don't know if it should be merged also into `master`, @Yeshasvitvs how was your plan to proceede?

traversaro, 2020-02-01:
> The master branch is rebased on the devel branch and then merged to it fd3e904
ll "
Guys, please avoid rebasing (and in any case avoid to use `git push -f`, as in all force options you are supposed to use it only if you know what it means : ) ) for public used branches, as this: 
* rewrites history of the branch, breaking `git pull` for anyone that was using that branch 
* it make impossible to recover a previous working commit if there is a regression, as in the case of https://github.com/robotology/human-dynamics-estimation/issues/171 .

See on this https://dev.to/rkfg/comment/103c, but you can also just google "git push force on master".

Yeshasvitvs, 2020-02-03:
> [`add-secondary-calibration`]([https://github.com/robotology/human-dynamics-estimation/tree/add-secondary-calibration]](https://github.com/robotology/human-dynamics-estimation/tree/add-secondary-calibration%5D) branch is now merged (#164) into `devel`, so that branch can be deleted (I would do it but it seems I don't have the rights.
> 
> I don't know if it should be merged also into `master`, @Yeshasvitvs how was your plan to proceede?

There was a branch rule making the branches after PR [protected](https://help.github.com/en/github/administering-a-repository/about-protected-branches). This is why you were not able to delete the branch after the PR is merged.

Yeshasvitvs, 2020-02-03:
> I don't know if it should be merged also into master, @Yeshasvitvs how was your plan to proceede?

@lrapetti lets wait for the update on #171 to be stable. Following #174 we can get devel and master aligned again.

DanielePucci, 2020-02-03:
If I am not wrong, both #171 and #174 have been closed, so the alignment may take place

claudia-lat, 2020-03-09:
Please @kouroshD add your comment on this.

kouroshD, 2020-03-09:
I backed up the branches: `feature/AddQP`,  `feature/tmpAddRetargetting`, 
`feature/teleoperation`.
 In any case, they were merged already to devel, so we can remove these branches from `human-dynamics-estimation`.

Yeshasvitvs, 2020-03-09:
@kouroshD if available please add the references to the PR merges related to those branches

Yeshasvitvs, 2020-03-10:
Deleting the following three branches:
- `feature/AddQP`
- `feature/tmpAddRetargetting`
- `feature/teleoperation`

@kouroshD @lrapetti @claudia-lat @DanielePucci

Yeshasvitvs, 2020-03-10:
Now the human-dynamics-estimation repo is rid of all the unnecessary branches. Only the following branches are maintained:

- `master` and `devel` branches, which are aligned until a recent commit. The master branches is tagged with https://github.com/robotology/human-dynamics-estimation/releases/tag/v2.1.0
- `feature/visualize-berdy-estimated-wrench` is another branch under active development which has to be aligned with devel branch (#187 ).
- [gh-pages](https://github.com/robotology/human-dynamics-estimation/tree/gh-pages) is another branch which in the past was maintained for having an autogenerated website for the repo. However, it has not been updated in a while. We need to discuss the maintenance of this branch.

Closing this issue for now

@lrapetti @kouroshD @lrapetti @DanielePucci

closed 2020-03-10 by Yeshasvitvs