gbionics/human-dynamics-estimation #103: Improve rviz visualization for HDE with pHRI configuration

Yeshasvitvs, 2019-04-12:
While updating wrench transformation #99 one thing we noticed is that the `tfs` are often failing because of which rviz visualization freezes momentarily. At first, I thought this may be due to the heavy computation needed when launching the full [pHRI](https://github.com/robotology/human-dynamics-estimation/blob/master/conf/xml/pHRI.xml) config file. So, I added a new [hands-pHRI](https://github.com/robotology/human-dynamics-estimation/blob/master/conf/xml/hands-pHRI.xml) configuration file with only hand wrenches and with `HumanDynamicsEstimator` device commented out. Still, the `tfs` are sometimes failing for rviz visualization. This problem did not occur when we run HDE only for [Human](https://github.com/robotology/human-dynamics-estimation/blob/master/conf/xml/Human.xml). But in case of `pHRI` file, there is an additional computation of hands wrench transformation from robot to human hands using the joint quantities for both human and the robot with [KinDyn computations](https://github.com/robotology/human-dynamics-estimation/blob/05d0656941f79fcc3b972e6db5b23ac46b140633/devices/HumanWrenchProvider/HumanWrenchProvider.cpp#L593). Also, at the moment we are not setting the transform server period explicitly from the config file and the [default value is 0.01](https://github.com/robotology/yarp/blob/master/src/devices/transformServer/FrameTransformServer.cpp#L509) with [transforms_life_time](https://github.com/robotology/human-dynamics-estimation/blob/05d0656941f79fcc3b972e6db5b23ac46b140633/conf/xml/Human.xml#L5) parameter explicitly set to 0.2. Playing with these two values will hopefully improve the visualization without rviz freezing. In this issue, we will try to investigate this problem and improve the rviz visualization while running `HDE` with pHRI configuration file.

Yeshasvitvs, 2019-04-12:
I did some preliminary test and the played with the `period` and `transforms_life_time` parameters of `transformServer` device. The `tfs` seems to be stable for correctly tuned parameters but again this tuning is done with gazebo simulation and my machine is having serious over heating problems to run `HDE` gazebo and rviz together. It is better to test and tune the parameters for `pHRI` case with the real robot and with `HDE` and rviz running on different machines.

traversaro, 2019-04-12:
Can't we just keep the `transforms_life_time` to 2 or 3 seconds? What is the problem of having a long  `transforms_life_time` ?

Yeshasvitvs, 2019-04-12:
I am not sure how it changes the visualization for dynamic motions but I am working with offline data from frank without movements. with this data `transforms_life_time` set to 2 or more seconds and period set to default value of 0.01 the `tfs` seems table when running HDE only for human case. But with pHRI config file it is still not stable enough visualization. But again this is done with gazebo.

Yeshasvitvs, 2019-04-15:
During https://github.com/robotology/human-dynamics-estimation/issues/99#issuecomment-483313796 and https://github.com/robotology/human-dynamics-estimation/issues/104#issue-433365059 we notice that the `tfs` are sometimes not stable and during these tests the ik was running considerable slower ~300 ms. We need to investigate further on any machine where ik can run much faster (probably monster!) and see if the visualization will be smoother. 

@DanielePucci

Yeshasvitvs, 2019-04-16:
> @Yeshasvitvs you are not using the IB IK, right?

@DanielePucci no, this is with pair-wise IK

Yeshasvitvs, 2019-04-17:
### An overview of the system :

- First all the producer wearable devices i.e `xsens suit`, `FTShoes` and `ICub` are launched with `period` value set in their configuration files
- `HDE` is launched with either `Human.xml` or `pHRI.xml` configuration file. Both of these files contain `transformServer` device at the beginning.
- `HumanStatePublisher` [sets the ground to human base (/pelvis) tf](https://github.com/robotology/human-dynamics-estimation/blob/c69dc6cafdd394dcfd93eb709772eb8f838bab8b/publishers/HumanStatePublisher/HumanStatePublisher.cpp#L131) and [publishes the human base tf](https://github.com/robotology/human-dynamics-estimation/blob/master/publishers/HumanStatePublisher/HumanStatePublisher.cpp#L228) along [with the joint states of the human](https://github.com/robotology/human-dynamics-estimation/blob/c69dc6cafdd394dcfd93eb709772eb8f838bab8b/publishers/HumanStatePublisher/HumanStatePublisher.cpp#L201) 
- `yarpstatepublisher` takes the parameters `--namePrefix Human --tfPrefix /Human/ --model humanSubject03_66dof.urdf --base-frame Pelvis --jointstates-topic "/Human/joint_states` and using the joint states ros messages from `HumanStatePublisher` computes the `tfs` for all the human links and sends them to `transformServer`
- When ros joint states are enabled on the robot, the robot joint values are streamed as a ros messages on `/icubSim/joint_states` topic. Then `yarpstatepublisher` is stated with the parameters `--namePrefix Robot --tfPrefix /icubSim/ --model model.urdf --base-frame root_link --jointstates-topic "/icubSim/joint_states"`. It will compute the `tfs` for all the links of the robot and send them to the `transformServer`
- `HumanRobotPosePublisher` will check for the necessary transforms (in `transformServer`) to compute the ground to robot base transform.
- The `rviz` visualization displays human and robot models for which it needs the `tfs` of all the links including the base links. Also, it displays joint torques of human as temperature which depend on the spherical joint names in the human model. The wrench messages are displayed as arrows and they depend on the link names in the human model.

### Observed problems
- While running HDE with `Human.xml` file, we observed sometimes that the joint torques and wrench messages are sometimes frozen. This problem is probably related to the rate at which the data is coming in the topics related to them, which is based on the periods of respective publishers. Additionally, I think we can tune the `Decay Time` and `History Length` parameters in `rviz` to improve the visualization of joint torques and wrench values.

[image]

- While running HDE with `pHRI.xml` file and using `HumanRobotPosePublisher`, the `getTransform` queries for tfs like robot base to robot foot or ground to human base, are often returning errors. Upon checking the `tfs` dumped in a text file, I can see that these `tfs` are available. A possible reason is that the `tfs` are possibly streamed fast or not retained longer. As a result, we cannot compute the ground to robot base `tf` and hence the robot model is not displayed correctly. 

CC @traversaro @diegoferigo @lrapetti @DanielePucci

Yeshasvitvs, 2019-05-22:
We discovered that the main problem for the rviz visualization is due to https://github.com/robotology/human-dynamics-estimation/issues/104#issuecomment-491946567

After that issue is fixed ros visualization is stable. So, I will close this issue.

CC @lrapetti

closed 2019-05-22 by Yeshasvitvs