gbionics/human-dynamics-estimation #104: Update HumanRobotPosePublisher with rpc port control

Yeshasvitvs, 2019-04-15:
During https://github.com/robotology/human-dynamics-estimation/issues/99#issuecomment-483313796 we notice there is a slight lag in the `tfs` and hence the rviz visualization is a bit laggy. Also, we tested `HumanRobotPosePublisher` and the visualization is bad probably because the  robot base tf is continuously streamed with the current `HumanRobotPosePublisher`. As @diegoferigo suggested [before](https://github.com/dic-iit/component_andy/issues/102#issuecomment-446108987) it is better to set the robot pose through a rpc port. Also, this can be stabilitzed only if the tfs are streamed correctly without any interruption.

[image]

Hopefully this is the last piece of puzzle to fix towards fully functional HDEv2

CC @DanielePucci @diegoferigo @lrapetti @traversaro

Yeshasvitvs, 2019-04-17:
Added the rpc functionality to `HumanRobotPosePublisher` https://github.com/robotology/human-dynamics-estimation/commit/229d41aa67cdee3a4080d8da477da3f18fc8436a

It was hard to get the correct tf of the robot base with offline wearable data and icub in simulation. So, I will test it with online data soon. In any case, `transformServer` seems to be a bit of bottleneck.

Yeshasvitvs, 2019-04-17:
**NOTE:** In the above gif the icub legs are having a weird transformation. This is because the `model` I was using with `yarprobotstatepublisher` is that of `iCubGenova04` while the urdf model used for `rviz`is that of `iCubGazeboV2_5`

Yeshasvitvs, 2019-04-17:
Just as a test I set the ground to robot base `tf` to be **identity** in the `run()` of `HumanRobotPosePublisher` and the rviz visualization of the robot is stable as below. So, following https://github.com/robotology/human-dynamics-estimation/issues/103#issuecomment-484180697 solving the problem of errors in `getTransform()` queries to `transformServer` will enable us to compute a correct ground to robot base tf.

[image]

DanielePucci, 2019-04-29:
We are still investigating why the transforms of the human are not received correctly from transform server.

Yeshasvitvs, 2019-05-13:
Today @lrapetti and I tested `HumanRobotPosePublisher` and discovered that the bottle neck is when looking fro `ground` to `/Human/Pelvis` transform from the transform server. However, this tf was available when reading the `/tf` ros topic. The problem actually lies in `HumanStatePublisher` which gives the `ground` to `/Human/Pelvis` tf directly to the `/tf` ros topic https://github.com/robotology/human-dynamics-estimation/blob/22ea287f7e09acdf4bb2f367506035e88ef8b3a7/publishers/HumanStatePublisher/HumanStatePublisher.cpp#L134 

and 

https://github.com/robotology/human-dynamics-estimation/blob/22ea287f7e09acdf4bb2f367506035e88ef8b3a7/publishers/HumanStatePublisher/HumanStatePublisher.cpp#L259

But the correct way to handle this tf is to open a transform server client in `HumanStatePublisher` and then setting the transform for `ground` to `/Human/Pelvis`. This change is included in https://github.com/robotology/human-dynamics-estimation/commit/ee783b8b0e6c5367696c64b0a1b3058a823a7d2f https://github.com/robotology/human-dynamics-estimation/commit/1efc5ef9ba7be13ab9e964d31fdee0ea9fea1c3b. Now, the `HumanRobotPosePublisher` is running smoothly and the `ground` to `/icub/root_link` transform can be set once through the rpc functionality and stream it consistently. 

**NOTE:** The final transform from `ground` to `/icub/root_link` is still not correct. This will be corrected soon!

@diegoferigo @DanielePucci

Yeshasvitvs, 2019-05-22:
Ground to robot base transformation computation is fixed in https://github.com/robotology/human-dynamics-estimation/commit/a810ac6352da8987ee151eb88738f12a94230f40

[image]

Now, the `HumanRobotPosePublisher` rpc port takes `set` command to set the correct ground to robot base transformation and `reset` command to set the ground to robot base transformation to Identity.

I will do a final testing with the pHRI experiment and close this issue.

CC @lrapetti

DanielePucci, 2019-05-28:
Waiting for testing on real robot.

Yeshasvitvs, 2019-06-13:
Today @lrapetti and I tested it with the real robot and the robot base pose with respect to human is set correctly.

[image]

A PR https://github.com/robotology/human-dynamics-estimation/pull/130 is opened to master.
I will close this issue.

closed 2019-06-14 by Yeshasvitvs