gbionics/human-dynamics-estimation #108: Investigate and Fix segmentation fault in HumanDynamicsEstimator

Yeshasvitvs, 2019-05-03:
While launching `HDE` segmentation fault occurs on linux machines a times and almost always on mac @lrapetti . We identified that `HumanDynamicsEstimator` device is the cause of this problem. When is it commented out `HDE` runs smooth on both the machines without any errors. A while ago we did some debugging and noticed that the segmentation fault is cause during berdy initialization. We need to investigate this problem in detail and fix it asap!

@lrapetti @kouroshD @traversaro @DanielePucci

traversaro, 2019-05-03:
Do you plan to get a valgrind log?

Yeshasvitvs, 2019-05-03:
@traversaro at the moment I do not have.

Yeshasvitvs, 2019-05-03:
@traversaro valgrind log file https://gist.github.com/Yeshasvitvs/57d4de2a4f62396ed470ff61bffdb982

traversaro, 2019-05-03:
The first invalid read: 
~~~
==9209== Invalid read of size 8
==9209==    at 0x261F33B7: hde::devices::HumanDynamicsEstimator::run() (in /home/yeshi/software/robotology-superbuild/build/install/lib/yarp/HumanDynamicsEstimator.so)
==9209==    by 0x5702829: yarp::os::PeriodicThread::Private::run() (in /home/yeshi/software/robotology-superbuild/build/install/lib/libYARP_OS.so.3.1.0)
~~~
seems to be juicy. However, I suggest to build the superbuild/the project in `Debug`/`RelWithDebInfo`, so you can get the exact line where you have the memory failure. I also suggest to use Valkyrie (sudo apt install valkyrie) to load the output of Valgrind.

Yeshasvitvs, 2019-05-03:
The problem was due to starting the device periodic thread run method before attaching all the necessary devices! Thanks @diegoferigo for finding this bud :grinning: It is fixed in https://github.com/robotology/human-dynamics-estimation/commit/4148f180be462605d2675e2c038c11ab123ec22b @lrapetti please try to run it with offline data and see if you still have any problems.

DanielePucci, 2019-05-03:
Well done guys @Yeshasvitvs @lrapetti @diegoferigo

lrapetti, 2019-05-06:
> @lrapetti please try to run it with offline data and see if you still have any problems

Great, no problem also on `MacOS`.
@Yeshasvitvs  We can close the issue.

closed 2019-05-06 by Yeshasvitvs