gbionics/human-dynamics-estimation #109: Investigate and Fix segmentation fault on HDE termination

Yeshasvitvs, 2019-05-03:
Currently the termination of HDE is not clean and always results in segementation fault. This is probably due to bad way of handling the termination of some of the devices of HDE. I did some preliminary tests and ran yarprobotinterface with a config file that only contains `IWearRemapper` and `HumanStateProvider`. If I understood well, currently the device termination process is designed as a call to  `close()`--->`detachAll()`-->`detach()`. Instead, I notice that the `detachAll()` method is called first. @traversaro what are termination steps of `yarprobotinterface` and how the devices are handled during termination ??

traversaro, 2019-05-03:
I am not an expert on  when yarprobotinterface "actions" run, I guess it is related to the `level` attribute, follow the code to understand more:
* https://github.com/robotology/yarp/blob/24eb21146091e4cc3d5bba0856f1ab8fe9b202df/src/yarprobotinterface/Device.cpp#L433
* https://github.com/robotology/yarp/blob/24eb21146091e4cc3d5bba0856f1ab8fe9b202df/src/yarprobotinterface/Robot.cpp#L565

However, I am quite  sure that `close` is the last method that can be called on a device, so for  sure  it is called **after** `detachAll` (mirroring how `attachAll` is called after `open`).

Yeshasvitvs, 2019-05-06:
@traversaro thanks for the pointers

I update the termination procedure for devices and wrappers in https://github.com/robotology/human-dynamics-estimation/commit/cd763cd14f248a8a847ebe4708e8f3c934f03828

I will soon updated it for the publishers and the close this issue.

Yeshasvitvs, 2019-05-06:
Termination process of publishers is updated in https://github.com/robotology/human-dynamics-estimation/commit/d7c457615a77ea9433130749baeaa4be7b467785

I noticed that when the `HumanStateWrapper` is included in the configuration file `HumanStatePublisher` is causing segmentation fault when the detach action level is set to `5`. `HumanDynamicsPublisher` does not have this problem. I am not certain what is happening in the case of `HumanStatePublisher` but when the detach action level is set to `4` the device termination is correct without any segmentation faults. So, to be consistent I put the detach action levels of all the publishers in `HDE` to `4`. 

#### Quick Summary

The termination steps of devices, wrappers and publishers were not handled correctly and this was causing segmentation faults when termination the yarprobotinterface. So, the termination steps are updated as following:

- The first call on termination is to `detachAll()` method which called `detach()` method. Inside `detach()` the periodic thread is stopped by calling `stop()`. The stop method calls the `releaseThread()` method. After the thread termination we can clear the resources in detach.
- After the call to `detachAll()`, the next and final call in termination step is to `stop()` method.

@lrapetti @diegoferigo @traversaro

lrapetti, 2019-05-07:
I reopen the issue since the problem seems to be still happening on my machine.

The source of the problem may be the [`IWearRemapper`](https://github.com/robotology-playground/wearables/tree/master/devices/IWearRemapper)

lrapetti, 2019-05-07:
Sorry, I didn't noticed there was already an issue related in [wearables](https://github.com/robotology-playground/wearables/issues/37) for the `IWearRemapper`.

closed 2019-05-07 by lrapetti

Yeshasvitvs, 2020-09-15:
I was noticing seg fault again while doing some experiments, and the reason seem to the again related to the same detach level `5`  of `HumanDynamicEstimator` as that of `HumanStateProvider`. So, they both are terminated at the same time, but inside `HumamDynamicsEstimator` when calling `kindyn` `setrobotstate` if the size of joints is zero, it throws an error. I took another look at the detach levels for different devices, and I settled on the following levels

The following three are the main devices that are computational units which carry some part of the algorithms

- `HumanStateProvider`    --------> level 5
- `HumanWrenchProvider`    --------> level 4
- `HumanDynamicsEstimator`    --------> level 3

All the other publishers/wrappers access data from the one of the above devices. So, I set their detach level to `2`. 
Finally, the `HumanDataCollector` devices accesses data from all the above devices, and it is important for logging with data to `.mat` file. So, I set its detach level to `1`.

With the above detach levels, the devices termination is cleaner now.

[image]

@lrapetti @kouroshD @claudia-lat @S-Dafarra