gbionics/human-dynamics-estimation #111: Closing and Merging PR "Integration based IK and Global IK #106"

kouroshD, 2019-05-07:
As I talked with @lrapetti  @Yeshasvitvs, I open this issue in order to address the merging of the open pull Request #106. 
By closing this PR, we can make new branches to add features to the master or devel and progress better and also clean up the repo and delete the unnecessary branches.
Currently, for teleoperation purposes I use this branch https://github.com/lrapetti/human-dynamics-estimation/tree/feature/integration_based_ik_rebased, which has been already rebased on `feature/teleoperation`.

As I know, there are no bugs on it, but there are active enhancement which are ongoing.

Yeshasvitvs, 2019-05-07:
As discussed with @lrapetti @kouroshD we will document the code base realignment of integration based IK and teleoperation. @lrapetti please add the details related to that. 

Accordingly, we will push the rebased integration based IK to a new branch in the upstream. After the plan of taking AnDy related videos we will proceed to merge this branch to master.

lrapetti, 2019-05-07:
Done, the `integration based` + `teleoperation` can be found in https://github.com/robotology/human-dynamics-estimation/tree/integration_based_teleoperation.

**N.B.** the code has already been rebased with the master of `human-dynamics-estimation`, which means it contains the latest changes and it works with [`feature/ICub-device-impl`](https://github.com/robotology-playground/wearables/tree/feature/ICub-device-impl) branch of wearbles (so older datasets should no longer work).

lrapetti, 2019-06-18:
This now can be found in HDE [`devel`](https://github.com/robotology/human-dynamics-estimation/tree/devel) branch.

closing.

closed 2019-06-18 by lrapetti