gbionics/human-dynamics-estimation #112: Understand increase in computational time reated to yarprobotstatepublisher

lrapetti, 2019-05-11:
We observed on `MacOS` that while running the `HumanStateProvider` with `HumanStatePublisher`, the computational time increase (of about 50%) when we run the `yarprobotstatepublisher`. This happens even if the `yarprobotstatepublisher` runs on a different machine.

If I'm not wrong, up to now we have not experienced the same behaviour on `Linux`.

@Yeshasvitvs

lrapetti, 2019-05-11:
CC @traversaro @diegoferigo @DanielePucci

Yeshasvitvs, 2019-05-11:
This is the same behavior on Linux system too.

traversaro, 2019-05-11:
>  the computational time

Computational time of what? How do you measure it?

> increase of about 50% for each `yarprobotstatepublisher` running

50% of what? I mean, an increase of the 50% of the initial computational time (so without any `yarprobotstatepublisher` running) or 50% of the computational time w.r.t. to the previous test with one less `yarprobotstatepublisher`. I mean, the progression is something like `0: 100, 1: 150, 2: 200, 3: 250, etc etc` or `0: 100, 1: 150, 2: 225, 3: 337.5, etc etc.

If the increase is always of 50% of the initial computational time, then the most probably explanation is that you are seeing the overhead of publishing the joint state on the `/joint_states` topic (b.t.w. at which  frequency is data being published on the ROS topic `/joint_states` ?). Personally I would not find that surprising, as communication and related code (such as serialization/de-serialization of the messages)  can be quite resource-hungry (see for example https://github.com/robotology/yarp/issues/353), and the ROS1 protocol is TCP-based (see Section 6 of http://wiki.ros.org/ROS/Technical%20Overview) and not multicast, so it does scale approximately linearly with the number of subscribers. 

If this computational time is too high and **you want/need to reduce it**, first of all I strongly suggest to recompiled all the code that you use (for this is convenient to use the superbuild) in `RelWithDebInfo` mode and **profile the code** to actually validate that what is taking time is the code related to publishing on `/joint_states`. This can  be done with an appropriate tool such as `callgrind`  + `kcachegrind` (see https://baptiste-wicht.com/posts/2011/09/profile-c-application-with-callgrind-kcachegrind.html) or flamegraphs (see http://www.brendangregg.com/FlameGraphs/cpuflamegraphs.html https://bitbucket.org/osrf/gazebo/wiki/Profiling%20and%20Benchmarking%20Gazebo) or any profiling tool do you prefer. 
Once you verified  that the code related to publishing on  `/joint_states` is indeed what is causing this increase in computational time, you need to decide how to tackle the problem. Possible solutions or mitigation strategies include:
* Optimize the YARP code that is run whenever you publish a `JointState` message on the the `/joint_states` topic. This may be complex, but it has the huge advantage that would benefit any kind of ROS connection we do via  YARP. In this cases, typically the low-hanging fruit for optimizations are any kind of memory allocation that is performed at runtime instead of reusing buffers. 
* Restructure the interconnections of your distributed system to avoid any use of `/joint_states` topic at all. If the only thing that you need to do is to publish the transform of the robot model on RViz via the /tf topic, you can think of taking the logic contained inside the yarprobotstatepublisher (that, as you can see in https://github.com/robotology/idyntree/blob/c8bf721b771fa4b1e7c3a940632e121060719a19/src/tools/yarprobotstatepublisher/src/robotstatepublisher.cpp#L191, is extremely limited, as the forward kinematics is handled via the KinDynComputations class) and putting it in a new `HumanTFPublisher` YARP device, that would just read the joint positions via the `getJointPositions` method as already done by `HumanStatePublisher`,  compute the forward kinematics and publish the result on the `/tf`topic. This would achieve  the same end result of the chain `HumanStateProvider <-(getJointPositions C++ call)-> HumanStatePublisher <-- (/joint_states ROS topic --> yarprobotstatepublisher`, without the need of passing via the`/joint_states` topic. 

By the way, this is good example of why it may be useful to always carefully separate the concern of "Computation" from the concern of "Communication"  (see 5C) in robotics software.

lrapetti, 2019-05-11:
> > the computational time
> 
> Computational time of what? 

As computational time, I meant the time that the `HumanStateProvider` device to solve the Inverse Kinematics problem.

> How do you measure it?

It is measured by [std::chrono::high_resolution_clock](https://en.cppreference.com/w/cpp/chrono/high_resolution_clock): [from](https://github.com/robotology/human-dynamics-estimation/blob/integration_based_teleoperation/devices/HumanStateProvider/HumanStateProvider.cpp#L670) -> [to](https://github.com/robotology/human-dynamics-estimation/blob/integration_based_teleoperation/devices/HumanStateProvider/HumanStateProvider.cpp#L698)

> 50% of what? I mean, an increase of the 50% of the initial computational time (so without any `yarprobotstatepublisher` running) or 50% of the computational time w.r.t. to the previous test with one [...]

50% of the computational time without any `yarprobotstatepublisher`. (but the number is approximative and dependent on the type of model/configuration we are running)

> which frequency is data being published on the ROS topic `/joint_states` ?

Usually, we run the `HumanStatePublisher` with a period around 0.02 (20ms)

traversaro, 2019-05-11:
> It is measured by [std::chrono::high_resolution_clock](https://en.cppreference.com/w/cpp/chrono/high_resolution_clock): [...]

That is interesting, as it should exclude any direct influence from the rest of the code. However, it is still possible that there is an indirect influence. It would be interesting to also use the `std::clock` (https://en.cppreference.com/w/cpp/chrono/c/clock) to actually understand how much time has been spend in running the function vs the fime in which the thread was waiting to be scheduled.

traversaro, 2019-05-11:
It would be interesting to measure the time actually spent in execution by that thread (the so-called CPU time) vs the time actually spent (the one that you already measure). I thought we could that via the `std::clock` function, but that apparently measures the CPU time of all the threads of the process. To measure the CPU time of a single thread, it is necessary to use some platform (POSIX) specific code but that should not be a problem for a quick test: https://stackoverflow.com/questions/44916362/how-can-i-measure-cpu-time-of-a-specific-set-of-threads .

DanielePucci, 2019-05-28:
Please @Yeshasvitvs  and @Yeshasvitvs call for a specific meeting to address this issue.

claudia-lat, 2020-11-16:
We documented the problem here and the problem is no longer affecting our results.
Closing it.

closed 2020-11-16 by claudia-lat