gbionics/human-dynamics-estimation #114: Use time-dipendent gain in IB IK

lrapetti, 2019-05-16:
Implement time dependent gains in link velocity correction.

lrapetti, 2019-06-18:
This dependency is relevant in order to ensure convergence given that we deal with discrete time problem (see also https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=804). It would be good to extend the theory (in https://github.com/dic-iit/component_andy/issues/226 for continuous time) properly for discrete time.

From an implementation point of view, if we can ensure the IK runs faster then the choosen [`period`](https://github.com/robotology/human-dynamics-estimation/blob/devel/conf/xml/HumanStateProvider.xml#L20) of the `HumanStateProvider` device (the frequency at wich the `run()` function is called), we can directly use the `period` for computing the gain. If instead the speed of the ik solver is unstable (changing a lot), and can take longer then the chosen `period`, we should measure the time passed between integration steps. In general, I would say we should be able to guarantee a solution within the `period`, so the first case would be better.

claudia-lat, 2020-11-09:
The priority for this issue is low. No needed for the moment.

closed 2020-11-09 by claudia-lat