gbionics/human-dynamics-estimation #116: Test HDE yarpmanager application

Yeshasvitvs, 2019-05-22:
Following the creation of `HDE` yarpmanager application https://github.com/dic-iit/component_andy/issues/192 we have to test it and see if all the components are launched correctly through yarpmanager.

CC @lrapetti

Yeshasvitvs, 2019-07-25:
@lrapetti we should test this as early as possible and close this issue. It will help to launch the demos directly using yarpmanager and reduces bus factor.

Yeshasvitvs, 2020-01-27:
@lrapetti did we do anything for this ?? :sweat_smile: 
If not, we put this task right after setting up HDE ubuntu machine https://github.com/dic-iit/component_andy/issues/332

lrapetti, 2020-01-27:
> @lrapetti did we do anything for this ?? 😅

No, but @kouroshD did it for the retargeting application (https://github.com/robotology/human-dynamics-estimation/pull/155), indeed it will serve as base for other applications (basically the dynamic estimation part should just be add)

Yeshasvitvs, 2020-02-04:
@lrapetti Great. So, can we close this issue ?

lrapetti, 2020-02-04:
> @lrapetti Great. So, can we close this issue ?

No, I think the one for HDE is not done, but shouldn't take much starting from the application created by @kouroshD (he did just the one for retargeting)

Yeshasvitvs, 2020-08-24:
Dependencies between different modules of yarpmanager application are important to ensure Rviz model visualization, a related issue https://github.com/robotology/human-dynamics-estimation/issues/116 for the retargeting demo yarpmanager application

CC @lrapetti

lrapetti, 2020-12-01:
Closing via https://github.com/robotology/human-dynamics-estimation/pull/220

closed 2020-12-01 by lrapetti