gbionics/human-dynamics-estimation #119: Add manipulability term in integration-based IK

kouroshD, 2019-05-29:
In this issue we, we want to write down the theory for adding the manipulability term in integral-based IK, so that we have good performance while avoiding high velocity of joints close to singularity.
This issue follows and enhances the issue we addressed before #110 .

kouroshD, 2019-06-10:
Starting from the basic Formulas provided in [#226](https://github.com/dic-iit/component_andy/issues/226) 

$$
{x}(t) = h(q(t)) , 
{\mathrm{v}}(t) = J(q(t)) \nu(t)
$$

We want to use manipulability term to avoid the singularity [ref. Marani22202](https://ieeexplore.ieee.org/document/1014830).

Measure of manipulability is defined as:

$$
mom=\sqrt{\left |  J J^{T}\right |}
$$

which is equal to product of the singular values of $J$. When a model reaches its singularity, $mom$ value becomes close to zero. The idea is when it reaches a threshold $\bar{m}$, we restric the singularity by moving on a hyper-surface where $mom$ is constant, therefore:

$$
\delta mom(q)=\frac{\partial mom(q)}{\partial q} \delta q=\frac{\partial mom(q)}{\partial q} J^{\dagger} \delta x = 0
$$

From previous formula we can deduct to the following formula for the IK problem:
$$
 \delta q=J^{\dagger} (\delta {x} -\delta {x}_{corr})
$$

and $\delta {x}_{corr}$ is computed as:

$$
\delta {x}_{corr}=\frac{  1-sign(\delta x \cdot n_{m})  }{2}(\delta x \cdot n_{m})n_{m}k(mom, \bar{m})+k(mom, \bar{m}/2)n_{m}
$$

in which $n_{m}$ is the unitary vector orthogonal to the manipulability surface:

$$
n_{m}=\frac{\left (\frac{\partial mom(q)}{\partial q} J^{\dagger}  \right )^{T}}{\left | \frac{\partial mom(q)}{\partial q} J^{\dagger} \right |}
$$ 
and $k(mom, \bar{m})$ is a positive well-shaped function of $mom$, in which for values bigger than $\bar{m}$ is equal to zero, and for values less than $\bar{m}/2$ is equal to one.

kouroshD, 2019-06-10:
In the above formulas, the computation of $mom$ in (2) and equation (6) to compute the $n_{m}$ are computationally expensive. 
For computing $mom$, we can compute the lower triangular matrix $L$ and finally $mom=\sqrt{trace(L)}$.

However, for computing $\frac{\partial mom(q)}{\partial q} $, I am not sure how we can compute it and as I saw in iDynTree there is not a method for that. 
Therefore, we should compute it either symbolically or numerically.
@DanielePucci  @traversaro  @lrapetti @Yeshasvitvs

kouroshD, 2019-06-10:
However, instead of following this approach we may use the manipulability ellipsoid idea as a constraint in qp problem in #123 and we may reach our goal defined here.

traversaro, 2019-06-10:
@kouroshD What is $q$ in your notes? However, this computations seems to be something that can be related to the Hessian  of the forward kinematics functions, see https://github.com/robotology/idyntree/issues/464 for a related discussion.

kouroshD, 2019-06-10:
$q$ is the general configuration space vector of a system,  composed of floating base position and velocity, and the joint configuration. But for manipulability, in general, the joints' configuration are important.
 Thank you @traversaro, I will go through it and I will update the issue accordingly.

DanielePucci, 2019-06-10:
@kouroshD 
> $q$ is the general configuration space vector of a system,  composed of floating base position and velocity

Let's discuss this because it seems to me you are somehow parametrising the base orientation with minimal representations

kouroshD, 2019-06-10:
> @kouroshD
> 
> > $q$ is the general configuration space vector of a system,  composed of floating base position and velocity
> 
> Let's discuss this because it seems to me you are somehow parametrising the base orientation with minimal representations
Actually, I made a typo. It is composed of base position and orientation, and joint configurations $q=(^{I}p_B, ^{I}R_B, s)$. In a lot of cases, base pose frame can affect the whole system manipulability.

DanielePucci, 2019-06-11:
@kouroshD 
So in this case, it is not really clear to me what the symbol $\frac{\partial}{\partial q}$ means

kouroshD, 2019-06-19:
As I have talked f2f with @DanielePucci, I update this issue with this correction that, the manipulability term is only considered for internal structure of the robot, and we can avoid introducing $^{I}p_B, ^{I}R_B,$ in configuration vector matrix. Therefore, $q=(s)$.

kouroshD, 2019-11-04:
@DanielePucci  @lrapetti  Since, currently human-state-provider works fine even close to singularities, I do not see the necessity for implementing this issue. If you agree, we can close this issue.

DanielePucci, 2019-11-11:
Deal @kouroshD !

closed 2019-11-13 by kouroshD