gbionics/human-dynamics-estimation #123: Implement qp for solving integration-based IK problem

kouroshD, 2019-06-06:
As discussed already with @DanielePucci , we think it is better to solve the IK problem using qp instead of pseudo inverse. We believe the result can can lead to a faster computation wrt to the current [benchmarks](https://github.com/robotology/human-dynamics-estimation/issues/110#issuecomment-491546478).

Also, by using qp we can add custom constraints to the optimization problem such as joint velocity and value limits and [tendons length limit](https://github.com/dic-iit/component_wholebody-teleoperation/issues/182).

kouroshD, 2019-06-10:
Following the discussion on Integration-based IKin [#226](https://github.com/dic-iit/component_andy/issues/226), We define the optimization problem as following:
$$
arg min_{\nu} E^{T} P E
\\\\
s.t.: E=V-J \nu , \\\\
l \leq A \nu \leq  u
$$
where $V_{m \times 1}$ is the task velocity vector and $\nu_{n \times 1}$ is the configuration velocity vector. 
By defining a new variable $y_{(m+n) \times 1}$, we rewrite the optimization problem in qp form as following:
$$
y=\begin{bmatrix}
\nu\\\\
 E
\end{bmatrix} 
$$
and the qp form of optimization problem will be:
$$
arg min_{y} y^{T} P' y
\\\\
s.t.: l' \leq A' y \leq  u'
$$
and:
$$
P'=\begin{bmatrix}
0_{n \times n} & 0_{n \times m}\\\\
 0_{m \times n} & P_{m \times m} 
\end{bmatrix}_{(n+m)\times(n+m)}
$$
and:

$$
A' =\begin{bmatrix}
J_{m \times n} & I_{m \times m}\\\\
A_{c \times n} & 0_{c \times m} 
\end{bmatrix} _{(c+m) \times (n+m)}
$$

and

$$
u'=\begin{bmatrix}
V
\\\\ 
u
\end{bmatrix}_{(c+m) \times 1} 
$$

and 

$$
l'=\begin{bmatrix}
V
\\\\ 
l
\end{bmatrix}_{(c+m) \times 1}
$$
where $c$ is the number of constraints we have.

kouroshD, 2019-06-10:
cc @DanielePucci  @lrapetti

kouroshD, 2019-06-12:
This is the result of the qp implementation comitted in [8fc12144d](https://github.com/robotology/human-dynamics-estimation/commit/8fc12144d0d957833120e5392467ae5911d7f043)
```
[INFO]IB-IK qp:  
0.0542266 0.0430803 0.00267819 0.00346871 -0.00808338 -0.00761784 -0.0309469 -0.00488929 -0.903387 0.0284212 -0.0227808 -0.0353919 0.00959447 -0.00142491 -0.0738395 0.00457509 0.0165475 0.081924 -0.0190887 0.016655 0.035728 -0.00444537 1.89043 -0.367696 -0.381219 -0.0149356 0.0143597 -0.0312914 0.0696717 0.00203498 -0.0593211 0.00892665 -0.0215014 0.002141 -0.837141 0.00270986 0.0157401 -0.0059313 
```
```
[INFO]IB-IK, sparseRobustCholeskyDecomposition:  
0.0542299 0.0430829 0.00267835 0.00345082 -0.00807833 -0.00761415 -0.0309464 -0.00484515 -0.903482 0.0284617 -0.022779 -0.0353897 0.00961157 -0.00142373 -0.0738241 0.00457394 0.0165277 0.0819279 -0.0190536 0.0166558 0.0357346 -0.00441799 1.89054 -0.367753 -0.381246 -0.0149321 0.0143549 -0.0312916 0.0696879 0.00203418 -0.059324 0.00892277 -0.0215044 0.00214071 -0.837192 0.00270692 0.0157435 -0.00591423 
```
The run time for solving qp problem is `2-4 e-04s`.

kouroshD, 2019-06-19:
We have test previous week the qp implementation. The visualization of the outputs in rviz were fine. 
Also, the efficiency was higher.
@lrapetti  @Yeshasvitvs

DanielePucci, 2019-07-22:
Thq qp o=implementation is done but we are having some issue swith #132. As soon as #132 is closed we can open a PR related to this and #132.

DanielePucci, 2019-07-29:
Let's wait for the PR cc @kouroshD

kouroshD, 2019-07-31:
The PR: https://github.com/robotology/human-dynamics-estimation/pull/137

kouroshD, 2019-11-04:
The PR is merged, so I close this issue.

closed 2019-11-04 by kouroshD

prashanthr05, 2021-11-20:
Ciao guys! 
For debugging the human base estimation problem I am working on, I was reading the `InverseVelocityKinematics` code where the QP matrices are generated.
I came across these lines,
https://github.com/robotology/human-dynamics-estimation/blob/b40eda165e3749e6e4719c31daf2a27444a4336d/HumanDynamicsEstimationLibrary/algorithms/src/InverseVelocityKinematics.cpp#L836-L847

where to generate the `A_prime` matrix, the upper left corner is set to zero.
Whereas according to the comment, https://github.com/robotology/human-dynamics-estimation/issues/123#issuecomment-500450379, this block must be  $J$ (`fullJacobianBuffer` if I am not wrong?). 

Is there a reason why this is set to zero? or is it a bug? Thanks in advance!

cc @lrapetti @kouroshD

prashanthr05, 2021-11-21:
I answer myself for the above comment, let me know if it's correct.
I think the assignment of $J$ and $V$ in `A_prime` and (`u_prime`, `v_prime`) is done during the `solveInverseDifferentialKinematics` step when the robot configuration is updated. At the `prepareOptimizer` stage, we do not have access to those quantities, and so it is assigned to zero.

lrapetti, 2021-11-22:
Hi @prashanthr05, it seems to be like that, but @kouroshD probably can confirm it

kouroshD, 2021-11-25:
yes, that's true @prashanthr05