gbionics/human-dynamics-estimation #124: Travis is broken

traversaro, 2019-06-09:
On the master branch, the Travis compilation tests are broken. It has been like that for a long time, but I am opening an issue to track this.

traversaro, 2019-06-09:
If there is no plan to fix this in the short or medium term, I suggest to just go ahead and deactivate Travis. Wasting 3.5 minutes for 8 builds for each commit, on builds that are known to be broken, is removing useful resources to the rest of the robotology projects that use Travis (Travis resources are shared inside an organization).

lrapetti, 2019-06-10:
Addressed in https://github.com/robotology/human-dynamics-estimation/pull/125.

traversaro, 2019-06-10:
Note that the issue is still present for the master branch, cc @DanielePucci .

lrapetti, 2019-06-10:
I can cherry pick the same commit and apply also in `master`

lrapetti, 2019-06-10:
> I can cherry pick the same commit and apply also in `master`

https://github.com/robotology/human-dynamics-estimation/pull/127, I am waiting for the checks before merging.

lrapetti, 2019-06-11:
Travis is now fixed for Linux by removing the support for `Ubuntu 16.04`.

For MacOS, the failure described in https://github.com/robotology/human-dynamics-estimation/pull/128#issuecomment-500768108 has to be addressed.

lrapetti, 2019-06-12:
> Travis is now fixed for Linux by removing the support for `Ubuntu 16.04`.
> 
> For MacOS, the failure described in [#128 (comment)](https://github.com/robotology/human-dynamics-estimation/pull/128#issuecomment-500768108) has to be addressed.

I think the reason of this failure is the fact we are not installing `icub-main` in the CI environment, but it is used in the [`RobotPositionController.cpp`](https://github.com/robotology/human-dynamics-estimation/blob/master/devices/RobotPositionController/RobotPositionController.cpp#L22). I will try to add it and check if it fixies Travis.

traversaro, 2019-06-12:
Mhh, how does that work without `find_package(ICUB REQUIRED)` and properly linking the iCub  libraries?

lrapetti, 2019-06-12:
> Mhh, how does that work without `find_package(ICUB REQUIRED)` and properly linking the iCub libraries?

I was wondering the same, it is not present. But the [`ctrl`](https://github.com/robotology/human-dynamics-estimation/blob/devel/devices/RobotPositionController/CMakeLists.txt#L28) library should be in `icub-main`, so it should be somehow linked in our machine since everyone is able to compile it.

traversaro, 2019-06-12:
I am not really sure what you mean by "linked in our machine" (see https://github.com/green7ea/cpp-compilation#a-linkers-job and section 2.2 of https://www.dropbox.com/s/nsimhycn8bgherf/meson_sample_chapter.pdf for more info on "linking").

lrapetti, 2019-06-12:
> I am not really sure what you mean by "linked in our machine"

I meant that in our machine (HDE users), we are don't get any compilation problem so the ctrl library is linked.

traversaro, 2019-06-12:
> I meant that in our machine (HDE users), we are don't get any compilation problem so the ctrl library is linked.

Ack, I was confused because linking is something done at the executable/library level and not at the machine level, now I see what you mean.

lrapetti, 2019-06-12:
I have open a PR https://github.com/robotology/human-dynamics-estimation/pull/129 to check if adding `icub-main` solves the problem.

lrapetti, 2019-06-12:
> I have open a PR #129 to check if adding `icub-main` solves the problem.

Apparently, it solved the problem since at least one of the two MacOs environment finished correctly. However, I just add the build and install step for `icub-main`, @traversaro  do you think I should also add `find_package(ICUB REQUIRED)` in order to have a clear error when `icub-main` is not installed?

traversaro, 2019-06-13:
> However, I just add the build and install step for icub-main, @traversaro do you think I should also add find_package(ICUB REQUIRED) in order to have a clear error when icub-main is not installed?

Yes. You just need to add `find_package(ICUB REQUIRED)` as actually the linking is already provided by this line: https://github.com/robotology/human-dynamics-estimation/blob/8cfc88c8fd9fad41ebe837e1c91c59f91df376b9/devices/RobotPositionController/CMakeLists.txt#L28 that I did not noticed before. Note that we did not noticed this problem before because in icub-main we are not exporting the CMake targets with a namespace, that would permit to catch this problems (use of a namespaced target in `target_link_libraries` more easily).

lrapetti, 2019-06-18:
Travis is now fixed in `devel` (see https://github.com/robotology/human-dynamics-estimation/pull/129), waiting for merging it into master.

DanielePucci, 2019-07-29:
Travis is fixed, in the `devel` branch. Let's close it.

closed 2019-07-29 by DanielePucci