gbionics/human-dynamics-estimation #132: Add customized constraints to qp solver of Integration-based IK

kouroshD, 2019-06-19:
Referring to issue #123 , where we have implemented the qp for solving IK problem. Later, we have decided to add the possibility for the user to add a custom linear constraint based on configuration space vector and velocity vector and add it to the qp problem. What we would like to implement are as following:
- [ ] joints with different velocity limit
- [ ] velocity limit for the base frame
- [ ] linear inequality/equality constraints for joint values

An implementation of the current status can be found in [commit 99e8c9](https://github.com/robotology/human-dynamics-estimation/commit/99e8c906f887202c1826e1562ac7a9c67cab312a)

kouroshD, 2019-06-19:
Starting from issues #123 and [#51](https://github.com/dic-iit/element_qp-reactive-control/issues/51), the qp constraints is defined as following:
$$
l^{\dot{x}} \leq A' y, \leq  u^{\dot{x}}
$$
where $y$ is the configuration velocity vector and the estimation error, $A', l^{\dot{x}}, u^{\dot{x}}$ are the generalized constraint matrix, lower and upper boundaries defined in #123 .

In general, we can define the joint limits and linear constraint of the joint values (https://github.com/dic-iit/component_wholebody-teleoperation/issues/182) with the following formula:
$$
l^x \leq B q \leq u^x
$$
where $q$ is the configuration vector and $B , l^x , u^x$ . Using the theory provided [here](https://github.com/dic-iit/element_qp-reactive-control/issues/51), we write only one of the linear constraints, i.e., one row of the provided formula as following:
$$
l^x_i \leq \begin{bmatrix}
b_{i,1} & ...  & b_{i,n}
\end{bmatrix}  \begin{bmatrix}
q_1
\\\\
\vdots 
\\\\
q_n
\end{bmatrix} \leq u^x_i
$$
or as following 
$$
l^x_i \leq   b_{i,1} q_1 + ...  + b_{i,n} q_n \leq u^x_i
$$

Assuming, initially in the estimation problem we respect the constraints, We will have therefore three cases:
- $ l^x_i < b_{i,1} q_1 + ...  + b_{i,n} q_n <  u^x_i$ is inside the limit: therefore we will not have constraint on velocities, therefore: $-\infty \leq b_{i,1} \nu_1 + ...  + b_{i,n} \nu_n \leq \infty$. In reality there are the joint velocity limit, therefore we can rewrite the previous formula as following: $min_{i} \leq b_{i,1} \nu_1 + ...  + b_{i,n} \nu_n \leq max_{i}$; in which $ min_i <0, max_i >0 $.

- $ b_{i,1} q_1 + ...  + b_{i,n} q_n \approx  u^x_i$: in this case, we can impose following constraints: $min_{i} \leq b_{i,1} \nu_1 + ...  + b_{i,n} \nu_n \leq 0$.

- $ b_{i,1} q_1 + ...  + b_{i,n} q_n \approx  l^x_i$: in this case, we can impose following constraints: $0 \leq b_{i,1} \nu_1 + ...  + b_{i,n} \nu_n \leq max_i$.

Therefore, we can summarize and write the constraint imposed by joint configurations for the configuration velocity vector as following:
$$
min_{i} \times tanh(k_u( (b_{i,1} q_1 + ...  + b_{i,n} q_n) -l^x_i ))  \leq b_{i,1} \nu_1 + ...  + b_{i,n} \nu_n \leq max_{i} \times tanh(k_u(  u^x_i  - (b_{i,1} q_1 + ...  + b_{i,n} q_n) ))
$$

in which $k_l, k_u$ are positive gains.

[image]

Finally, for all the linear equality/ inequality joint limits, we can write it as following:

$$
\begin{bmatrix}
min_{1} \times tanh(k_u( (b_{1,1} q_1 + ...  + b_{1,n} q_n) -l^x_1 ))
\\\\
\vdots 
\\\\
min_{c'} \times tanh(k_u( (b_{c',1} q_1 + ...  + b_{c',n} q_n) -l^x_{c'} ))
\end{bmatrix}
 \leq B y \leq 
\begin{bmatrix}
max_{1} \times tanh(k_u(  u^x_1  - (b_{1,1} q_1 + ...  + b_{1,n} q_n) ))
\\\\
\vdots 
\\\\
max_{c'} \times tanh(k_u(  u^x_{c'}  - (b_{c',1} q_1 + ...  + b_{c',n} q_n) ))
\end{bmatrix}
$$

and $c'$ is the number of constraints. Finally, such constraint will be concatenated to the qp constraint matrix.

cc @DanielePucci  @lrapetti  @gabrielenava  @Yeshasvitvs

kouroshD, 2019-06-20:
The implementation of the qp with customized constraints is done with the commit [9f93730](https://github.com/robotology/human-dynamics-estimation/commit/9f9373041bbd3b1650519cbb1b82509a226d588a). It is running but has not been visualized to check if the results are correct.
Some of the shoulder constraints mentioned in [issue 182,  joint limit and tendons ](https://github.com/dic-iit/component_wholebody-teleoperation/issues/182) also have been added.

kouroshD, 2019-06-20:
With the commit [8019528](https://github.com/robotology/human-dynamics-estimation/commit/80195282c201e149e43b1349e844433fba56ac59), the results of qp with customized constraints complies with pseudo-inverse methods.

DanielePucci, 2019-06-21:
I have to get into the details, but if you followed the theory [here](https://github.com/dic-iit/element_qp-reactive-control/issues/51), it should be fine.

We really should **stop** using $q$ for joint angle representation. Please, change it to $s$, thanks @kouroshD

kouroshD, 2019-06-21:
> I have to get into the details, but if you followed the theory [here](https://github.com/dic-iit/element_qp-reactive-control/issues/51), it should be fine.

yes, as I mentioned in first comment, I have started from that, and generalize it for custom constraints.

> We really should **stop** using q for joint angle representation. Please, change it to s, thanks @kouroshD

done!

kouroshD, 2019-07-15:
I have implemented and tested the QP taking into account different joint constraints. The code is updated [here](https://github.com/robotology/human-dynamics-estimation/commit/3078b8dc54e3fffb82d045b51e8a07eafc492fea).

We have tested it together with @lrapetti and results seem fine. However, emulating the  robot's shoulder tendon limits was hard, we managed to produce situations where the constraints on the joint values (linear inequality problem) get activated.

However, there is one difference between the application of the method explained above in this issue and the one mentioned in [#51](https://github.com/dic-iit/element_qp-reactive-control/issues/51). In case of the robot, when we reach the limit,  [above figure](https://user-images.githubusercontent.com/17707730/59757687-9b4b7080-928c-11e9-9140-f4dfd7033a20.png) ±20, robot does not go beyond that limit. Therefore, joint velocity reaches `0` in case the limitations get activated. In case of retargeting, we do not have control on human motions, therefore joint values may go beyond the limits (according to the human measures, e.g., in the same figure linear combination of joint configurations reaches ±25), therefore we may produce velocity in contrary direction. In the worst case scenario, the linear inequality constraint limit, may become a linear equality constraint, however very hard to happen.
We have different options to follow:
- since it is very hard to happen such scenario, we leave the `tanh` function as it is.
- We shift the `tanh` function, so that, when we reach the limits and beyond it, the velocity keep being `0`.
- we use sigmoid function, similar to the second suggestion.

@DanielePucci  @lrapetti

kouroshD, 2019-07-30:
After discussing with @lrapetti , we think that the problem is coming from the sampling time of the discretization, and we may go out of hard limit we added in QP problem, even if it is not common.
So, by decreasing the $k_u$ and $k_l$ values in [this comment ](https://github.com/robotology/human-dynamics-estimation/issues/132#issuecomment-503501162), we will have lower slope in the central of the tanh function, so the limit also get activated sooner if the human goes close to the inequality linear constraint limit.

[image]

kouroshD, 2019-07-30:
Therefore, we proceed with the PR.
CC @DanielePucci

kouroshD, 2019-07-31:
The PR: https://github.com/robotology/human-dynamics-estimation/pull/137

claudia-lat, 2019-09-23:
Please @kouroshD drop a comment and eventually close the issue.

kouroshD, 2019-11-04:
The PR is merged, so I close the issue.

closed 2019-11-04 by kouroshD