gbionics/human-dynamics-estimation #138: Incorporating FTBox into rviz visualization

Yeshasvitvs, 2019-08-12:
Following the incorporation of FTBox in wearable repository https://github.com/robotology/wearables/issues/48, in this issue we track the details of using the FTBox inside Rviz visualization. Currently, we are using the [board sdf model](https://github.com/robotology/icub-gazebo-wholebody/blob/devel/models/board/board.sdf). For rviz visualization, we need to convert it to URDF model and make it discoverable by HDE.  The board model contains a base link and 4 handle links that are connected to the base link through fixed joints. The steps to visualize the board model in Rviz are:

- Starting a `yarprobotstatepublisher` for the board model. (I guess  this is not strictly needed)
- Running `HumanRobotPosePublisher` with the configuration file adapted to the board model. The `HumanRobotPosePublisher` is designed to set the base of the robot model with respect to any link of the human with a transformation given in the [configuration file](https://github.com/robotology/human-dynamics-estimation/blob/master/conf/xml/RobotPosePublisher.xml).

Currently, the naming of the parameters of `HumanRobotPosePublisher` are designed with the pHRI experiment using only  iCub and human together. We may need to update them to be reflective of its generality for more applications.

@lrapetti @claudia-lat @DanielePucci

lrapetti, 2019-08-13:
> . Currently, we are using the [board sdf model](https://github.com/robotology/icub-gazebo-wholebody/blob/devel/models/board/board.sdf). For rviz visualization, we need to convert it to URDF model and make it discoverable by HDE.

Probably @fiorisi may provide us with the URDF model of the real box at a certain point.

fiorisi, 2019-08-13:
Unfortunately creating the URDF with the [usual pipeline](http://wiki.icub.org/wiki/Creo_Mechanism_to_URDF) is a long process. As agreed I provide the transforms that identify the location of each FT.
Additionally, here you can find the `.stl` and `.obj` meshes of the box.
[FTbox_meshes.zip](https://github.com/robotology/human-dynamics-estimation/files/3496674/FTbox_meshes.zip)

[image]

[image]

[image]

[image]

traversaro, 2019-08-13:
@Yeshasvitvs @lrapetti  If you are willing to play a bit with some C++, I think @fiorisi can give you the STEP AP242 of the assembly, and you can process it to generate the URDF/SDF in the form that you prefer, without the need of blocking anyone from the mechanical side. I am doing something similar for an internal IIT's  project, and once you get the general idea it is quite easy to extract any kind of information you may want. If you are interested in this, feel free to ping me.

Yeshasvitvs, 2019-08-19:
@traversaro I am curious to check it. Let me know when you are available. Thanks.

lrapetti, 2023-03-22:
This is something we are no longer investigating, so I will close it

closed 2023-03-22 by lrapetti