gbionics/human-dynamics-estimation #140: Add feature to visualize berdy estimated wrench

Yeshasvitvs, 2019-08-20:
Currently, we give wrench as the feet of the human (obtained through FTShoes) as inputs to the measurement vector for berdy with a low covariance. We only retrieve the information of joint torque estimates and send it through [`IHumanDynamics` interface](https://github.com/robotology/human-dynamics-estimation/blob/master/interfaces/IHumanDynamics/IHumanDynamics.h).

 If I understood correctly, the estimated dynamic variables  vector from berdy will also contain the wrench estimates (@traversaro  please confirm this and point out which function to use for retrieving this information). The main goal of this demo is to display the estimate the wrenches  at the hands of the human when different weights are being handled. In this case, additional to the wrench at the feet with low covaariance, we need to include two wrenches at the hands to be present and set their covariance to be high. Being consistent with the architecture of HDEv2 I believe a clean way to consider this is to update `HumanWrenchProvider` device to take `dummy` wrench sources from the configuration files and then set the wrench values for these `dummy` sources to **zero**. Inside `HumanDynamicsEstimator` device we add the high variance for these `dummy` wrench sources. 

Furthermore, the current way of visualizing wrench in Rviz is through [`IAnalogSensor` interface implementation](https://github.com/robotology/human-dynamics-estimation/blob/40e949a4e09aad4a412b0f7af107d80d7831ce7b/devices/HumanWrenchProvider/HumanWrenchProvider.cpp#L829) inside `HumanWrenchProvider` and then [attaching it to `analogServer` device](https://github.com/robotology/human-dynamics-estimation/blob/master/conf/xml/Human.xml#L232). So, we need to update `HumanDynamicsEstimator` to have `IAnalogSenor` interfaces.

@traversaro @diegoferigo @lrapetti @claudia-lat @kouroshD @DanielePucci

traversaro, 2019-08-21:
> If I understood correctly, the estimated dynamic variables vector from berdy will also contain the wrench estimates (@traversaro please confirm this and point out which function to use for retrieving this information). 

The net external wrenches estimates are contained in the vector of estimated dynamic variables $d$ see for example Equation 14/15 in https://www.mdpi.com/1424-8220/19/12/2794/pdf . As in the existing estimator joint torques are extracted via the `BerdyHelper::extractJointTorquesFromDynamicVariables` method ( https://github.com/robotology/human-dynamics-estimation/blob/e68a6ab7f5387c3efcdcde7728ae1f843f637a66/devices/HumanDynamicsEstimator/HumanDynamicsEstimator.cpp#L1192 ), similarly you can extract net external 6D force/torque via the `BerdyHelper::extractLinkNetExternalWrenchesFromDynamicVariables` method ( https://github.com/robotology/idyntree/blob/c8bf721b771fa4b1e7c3a940632e121060719a19/src/estimation/include/iDynTree/Estimation/BerdyHelper.h#L662 ).

Yeshasvitvs, 2019-08-21:
@traversaro I see that this method gets the net wrench on [all the links of the model](https://github.com/robotology/idyntree/blob/c8bf721b771fa4b1e7c3a940632e121060719a19/src/estimation/src/BerdyHelper.cpp#L2115). I do not see any method that takes a specific link name as an argument and gets the net wrench on just that. It will be useful to have that too. What do you think ?

traversaro, 2019-08-21:
I think you could easily get that from the `LinkNetExternalWrenches` instance,  i.e. : 
~~~cxx
netWrenches[model.getLinkIndex("linkName");
~~~
if you want to also add a `extractLinkNetExternalWrenchFromDynamicVariables(const VectorDynSize& d, Wrench& netExtWrench, const LinkIndex lnkIndex)` that could also make sense.

Yeshasvitvs, 2019-08-21:
For the time being for the sprint I will proceed with the available method and and use the extract link specific wrench `netWrenches[model.getLinkIndex("linkName")`. 

We can add a new method with link index as an argument in berdy later

Yeshasvitvs, 2019-08-21:
The feature to visualize berdy estimated wrench is implemented in [feature/visualize-berdy-estimated-wrench](https://github.com/robotology/human-dynamics-estimation/tree/feature/visualize-berdy-estimated-wrench)

Yeshasvitvs, 2019-08-21:
The implementation details are:

- Extract all the links net external wrench estimates from berdy dynamic variables using [extractLinkNetExternalWrenchesFromDynamicVariables()](https://github.com/robotology/idyntree/blob/c8bf721b771fa4b1e7c3a940632e121060719a19/src/estimation/include/iDynTree/Estimation/BerdyHelper.h#L662) done in https://github.com/robotology/human-dynamics-estimation/commit/a32afa1158049ed8498ba9c100fdf48690050a26
- Implement `IAnalogSensor` interface methods inside `HumanDynamicsEstimator` and stream the external wrench estimates of the interested links as done in https://github.com/robotology/human-dynamics-estimation/commit/bf69f5ded2cd4e9e14d0643f5c876cf7006a0f7c

DanielePucci, 2019-08-22:
Great @Yeshasvitvs 
> We can proceed to test it online with a human handling objects of different weights with hands.

You can proceed with opening a new issue specifically on the testing with different weights

Yeshasvitvs, 2019-08-23:
@kouroshD and I did some tests yesterday. 

- The arrows in **yellow** color are **measured** wrench values
- The arrows in **blue** color are **estimated** wrench values from berdy

Here @kouroshD is lifting a heavy object (not certain about the weight but it is the toolbox from the iRonCub). The covariance for the dummy wrench sources at the hands is put at **100000**. What we see is that the measured wrench at the feet and the estimated wrench are the same. The measured wrench at the hands is set to **zero** as it is a dummy source and the estimated wrench is high owing to the high covariance. This is something I expected will be the case and I not sure how the estimated wrench at the hands will change according to the weight handled by the human. I went through the original paper [Simultaneous state and dynamics estimation in articulated structures](https://ieeexplore.ieee.org/document/7353848) to understand berdy and check the validation details. The paper only validated the joint kinematic quantities but not the link net external wrenches (@traversaro please correct me if I am wrong). On the other hand, @claudia-lat uses offline berdy i.e. MAPEst successfully to get reasonable wrench estimates at the hands in the case of exo-skeleton experiments. So, may be there is something I am missing in using c++ implementation of berdy! 

Another minor detail is related to the how the wrench at the hands should be represented i.e. w.r.t. which frame of reference. If I understood well, the berdy wrench estimates are expressed in the body frame (@traversaro please confirm this) 

[image]

[image]

We need to discuss this further and fix it @DanielePucci @traversaro  @claudia-lat @kouroshD @lrapetti

traversaro, 2019-08-23:
>  I went through the original paper Simultaneous state and dynamics estimation in articulated structures to understand berdy and check the validation details. The paper only validated the joint kinematic quantities but not the link net external wrenches (@traversaro please correct me if I am wrong).

Indeed, the only validation seems on kinematic quantities.

> If I understood well, the berdy wrench estimates are expressed in the body frame (@traversaro please confirm this)

As far as I remember, that is correct. 

> So, may be there is something I am missing in using c++ implementation of berdy!

Why you think you are missing something? However, all this doubts seems to be more realated to the theory, and not directly related to the feature of visualizing the estimated wrench.

> We need to discuss this further and fix it

What we need to fix?

Yeshasvitvs, 2019-08-23:
> However, all this doubts seems to be more related to the theory

I agree, I am not crystal clear at the moment about the theory. 

> What we need to fix?

The estimation of the wrench at the hands is supposed to reflect the weight of the objects handled by the human. This is towards AnDy Y2 Demo. By fix I mean, we need to show that the estimated wrench at the hands reflect the weight of the objects at the hands.

traversaro, 2019-08-23:
> By fix I mean, we need to show that the estimated wrench at the hands reflect the weight of the objects at the hands.

I suggest to have a cardboard  box, initially empty and in which you then add bottles of water (or in any case object of the same weight). As soon as you add bottles of water, the arrows should proportionally grow.

claudia-lat, 2019-08-23:
Preliminary implementation is done, so tuning and testing will be continued in #144

closed 2019-08-23 by claudia-lat

Yeshasvitvs, 2019-09-23:
@kouroshD @prashanthr05 this is the issue that follows https://github.com/robotology/human-dynamics-estimation/issues/145, where the berdy initialization changes are done in `y` measurement vector by considering the accelerometer values coming from `HumanStateProvider` through `IHumanState` interface.