gbionics/human-dynamics-estimation #144: Tune and test berdy estimated net wrench visualization in Rviz

Yeshasvitvs, 2019-08-23:
Following the updated of `HumanDynamicsEstimator` to extract the net link external wrenches https://github.com/robotology/human-dynamics-estimation/issues/140 and some initial testing https://github.com/robotology/human-dynamics-estimation/issues/140#issuecomment-524293991, in this issue we will tune the covariances and test the rviz visualization of berdy estimated net wrench as suggested in https://github.com/robotology/human-dynamics-estimation/issues/140#issuecomment-524314217 https://github.com/robotology/human-dynamics-estimation/issues/140#issuecomment-523894716

@claudia-lat @kouroshD @lrapetti @traversaro  @DanielePucci

Yeshasvitvs, 2019-08-27:
During the tuning we noticed that the values of the acceleration sensors inside the [measurement vector are set to zero](https://github.com/robotology/human-dynamics-estimation/blob/97fb28af0ff4e8737683727fd1b432249f5578ea/devices/HumanDynamicsEstimator/HumanDynamicsEstimator.cpp#L1112). This is clearly wrong and instead we need to consider the proper acceleration of the sensors attached to the human links for the berdy estimated wrenches to be reasonable. This problem will be tackled in a new issue

claudia-lat, 2019-09-18:
Tests are still ongoing. Moved this issue on the next iteration. @Yeshasvitvs

Yeshasvitvs, 2020-01-27:
Stack of tasks BERDY External net wrench tuning and visualization is Rviz is done. Closing

closed 2020-01-27 by Yeshasvitvs