gbionics/human-dynamics-estimation #145: Consider berdy accelerometer measuresment from xsens data

Yeshasvitvs, 2019-08-27:
Following the initial tuning done in https://github.com/robotology/human-dynamics-estimation/issues/144#issuecomment-525394323 for visualizing estimated wrench at the hands, we notice that the accelerometer measurements considered for the berdy measurements vector are set to zero. In this issue, we will track the details related to considering the proper acceleration of the links through the freebody acceleration measurements give by xsens.

Yeshasvitvs, 2019-08-28:
The SDK currently used with our xsens suits provides the user with __free body accelerations__ which is **the acceleration with gravity estimation subtracted expressed in the global frame.** The accelerometer sensor measurements consider in berdy estimation are **the acceleration with gravity, expressed in sensor frame.** 

From the previous investigation, the relation between the free body acceleration (fba) and the proper acceleration is given as:
 
<a href="https://www.codecogs.com/eqnedit.php?latex=^G\mathbf{a_{fba}}&space;=&space;^GR_S&space;*&space;(^S\mathbf{a}&space;&plus;&space;^SR_G&space;*\mathbf{g})\:&space;\:&space;\:&space;\:&space;\:&space;with\:\:&space;\mathbf{g}&space;=&space;[0,&space;0,&space;-9.81]" target="_blank"><img src="https://latex.codecogs.com/gif.latex?^G\mathbf{a_{fba}}&space;=&space;^GR_S&space;*&space;(^S\mathbf{a}&space;&plus;&space;^SR_G&space;*\mathbf{g})\:&space;\:&space;\:&space;\:&space;\:&space;with\:\:&space;\mathbf{g}&space;=&space;[0,&space;0,&space;-9.81]" title="^G\mathbf{a_{fba}} = ^GR_S * (^S\mathbf{a} + ^SR_G *\mathbf{g})\: \: \: \: \: with\:\: \mathbf{g} = [0, 0, -9.81]" /></a>

where <a href="https://www.codecogs.com/eqnedit.php?latex=^G\mathbf{a_{fba}}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?^G\mathbf{a_{fba}}" title="^G\mathbf{a_{fba}}" /></a> is the `sensorFreeAcceleration` of the new MVN2018, and <a href="https://www.codecogs.com/eqnedit.php?latex=^S\mathbf{a}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?^S\mathbf{a}" title="^S\mathbf{a}" /></a> is the old MVN4.2 `sensorAcceleration`.

Please note that while the `sensorAcceleration` was provided with respect to the sensor SoR, the `sensorFreeAcceleration` is expressed in the global SoR.

Yeshasvitvs, 2019-08-28:
In order to consider the free body acceleration coming from the xsens suit and correct it to proper acceleration to be used in the berdy measurements vector, the following are the key implementation details that are required:

- Update HumanStateProvider to accelerometerSensorMeasurementsOption parameter (that has default value of `none` for zero accelerometer values, `gravity` to consider only gravitational acceleration and `proper` to consider proper accelerations) and initialize accelerometer variables based on the input model - Done in https://github.com/robotology/human-dynamics-estimation/commit/8d598f6e876bd4fd50716e851d905e718d8af674 
-  Update HumanStateProvider to get freebodyacceleration sensors through werable input data  - Done in https://github.com/robotology/human-dynamics-estimation/commit/6e7293ec4988afed44d7172fadfcff1d49e69c66
- Add `MODEL_TO_DATA_ACCELEROMETER_PARENT_LINK_NAMES` parameter in the configuration file - Done in https://github.com/robotology/human-dynamics-estimation/commit/93be7d06d5ba43544d9fea74339b014e2c23cede
- Update the run method in HumanStateProvider with computation for proper acceleration - Done in https://github.com/robotology/human-dynamics-estimation/commit/06aabb4040c0419bedbe9af520ced45d78200025
- Update the IHumanState interface with accelerometer methods and implement them in HumanStateProvider - Done in https://github.com/robotology/human-dynamics-estimation/commit/b040c8b25f53873a7d58a36f69e140fd1f3a1dcf
- Update run of HumanDynamicsEstimator to read the accelerometer values coming from IHumanStateInterface and use them for berdy measurements vector - Done in https://github.com/robotology/human-dynamics-estimation/commit/04768bad750e54e66cf69be7555a0639da4be91a

@traversaro @diegoferigo @lrapetti @claudia-lat @DanielePucci please do a code review if possible. Thanks

PS: In this implementation I considered freebody accelerations coming in the iwear interface in HumanStateProvider to compute proper acceleration and pass it through IHumanState interface. A cleaner way is to consider freebody accelerations coming in the iwear interface directly in the HumanDynamicsEstimator and then use the computed proper accelerations in the berdy measurement vector.

Yeshasvitvs, 2019-08-28:
I did some tests on the acceleration values as @traversaro suggested. The test is done with static T-pose data set and one of the values related to the Pelvis link are shown below.

```

====================Accelerometer name :  Pelvis_accelerometer ================= 

Free body acceleration from xsens :  0.0128947 0.0315074 -0.0186999  

World_H_Base transform :  
Position : -0.135081 -0.177225   0.219933 
Rotation :  0.83787   0.544481  -0.0389044
           -0.544356  0.838726   0.0146756
            0.0406207 0.00888165 0.999135

In euler angles [-0.841514572867 -2.229607964 -33.0174034122] (degree) i.e. rotation about Z-axis 
 
Base_H_Sensor transform :  
Position :   0.083594 -0.00983071 -0.0916579 
Rotation :  -0.298603  0.0494737  -0.953094
            -0.158715 -0.987323   -0.00152535
            -0.941088  0.150815    0.30267

In euler angles [0.2 -72.381 -170.5924] (degree) i.e there is a rotation around y-axis and z-axis. [I double checked this with the urdf](https://github.com/robotology/human-gazebo/blob/master/humanSubject01/humanSubject01_66dof.urdf#L1500)
 
World_H_Sensor transform :  
Position :  -0.0668267 -0.232321  0.131663 
Rotation :  -0.299995  -0.501994 -0.811175
             0.0156166 -0.852812  0.521985
            -0.953813   0.143925  0.263679

In euler angles [-63.1995248 -54.210902 120.8628558] (degree) i.e there is a rotation around all the three axes

corrected acceleration = fbAcceleration - gravity 

Corrected accceleration :  0.0128947 0.0315074 9.7913
 
Computed Proper acceleraton :  -9.34244 1.37587 2.58774  (sent through IHumanState interface in HumanStateProvider)

Received Proper acceleration :  -9.34244   1.37587   2.58774 (received through IHumanState interface in HumanDynamicsEstimator)

Berdy classical acceleration estimates : -6.15775 -5.04919 10.1027  -26.9387 -0.157612 2.62974 

```

Upon double checking [MAPEst code for computing proper acceleration](https://github.com/claudia-lat/MAPest/blob/floatingWithIWear_featureAngAccSensor/Experiments/23links_human/main_floating.m#L72) I notice that the the [orientation considered](https://github.com/claudia-lat/MAPest/blob/437930be39b75d74518d21c02fef58438de7e82f/Experiments/23links_human/src/extractWearableDataFromIWear.m#L194) is obtained directly from Xsens data. This is what @traversaro was pointing this morning. So, I will follow the same and use the sensor orientation coming from xsens through wearable data to compute proper acceleration.

@traversaro @DanielePucci @claudia-lat @lrapetti @kouroshD

Yeshasvitvs, 2019-08-28:
> In which frame is this expressed? Can we rotate it in the world frame?

If I understood the math in https://github.com/robotology/human-dynamics-estimation/issues/145#issuecomment-525716713 correctly, this is expressed in sensor frame. If it is rotated wrt to world frame we get,

`Corrected accceleration :  0.0128947 0.0315074 9.7913`

>  Is this consistent or it quickly changes?

It is changing from iteration to iteration
```
[INFO]Acceleration for link Pelvis:-8.76714 -4.03546 1.09955  -29.5293 -10.2373 4.35225 
[INFO]Acceleration for link Pelvis:-4.12799 -0.811947 1.66756  -15.1175 -17.3661 19.2023 
[INFO]Acceleration for link Pelvis:-4.11463 -0.811448 1.67887  -15.1152 -17.3903 19.1833 
[INFO]Acceleration for link Pelvis:-2.21607 0.392443 1.8908  -6.59637 -15.356 16.6369 
```

Yeshasvitvs, 2019-08-28:
HumanStateProvider is updated to get sensor orientation from wearable data and the proper acceleration computation is updated using it - Done in https://github.com/robotology/human-dynamics-estimation/commit/b400850c965e1da8f03d66b6118256a2404d71d4 @traversaro please review it.
I did another offline test with the same T-pose data set,

```

====================Accelerometer name :  Pelvis_accelerometer ================= 

Free body acceleration from xsens :  -0.0265831 0.0291235 -0.0520811 

World_H_Base transform :  
Position : -0.134692 -0.177573   0.219953 
Rotation :  0.837364  0.545422  -0.036556
           -0.54533   0.838116   0.0133223
            0.0379045 0.00877945 0.999243

In euler angles [-0.841514572867 -2.229607964 -33.0174034122] (degree) i.e. rotation about Z-axis
 
Base_H_Sensor transform :  
Position :   0.083594 -0.00983071 -0.0916579 
Rotation :  -0.298603  0.0494737  -0.953094
            -0.158715 -0.987323   -0.00152535
            -0.941088  0.150815    0.30267

In euler angles [0.2 -72.381 -170.5924] (degree) i.e there is a rotation around y-axis and z-axis
 
World_H_Sensor transform :  
Position :  -0.0667046 -0.23262   0.131447
Rotation :  -0.302203  -0.502594 -0.809983
             0.0172778 -0.852461  0.522505
            -0.953087   0.143908  0.2663

Sensor orientaion :
Rotation : 0.214648 0.749846 -0.625825
          -0.33952  0.658085  0.672049
           0.91578  0.0682261 0.395844

corrected acceleration = fbAcceleration - gravity 

Corrected accceleration : -0.0265831 0.0291235 9.75792
 
Computed Proper acceleraton :   8.92051 0.664977 3.89883  (sent through IHumanState interface in HumanStateProvider)

Received Proper acceleration :  8.92051   0.664977   3.89883 (received through IHumanState interface in HumanDynamicsEstimator)

Berdy classical acceleration estimates : -8.75649 -4.02295 1.13533  -29.5153 -10.1618 4.32696 

```

traversaro, 2019-08-28:
> Received Proper acceleration :  -9.34244   1.37587   2.58774 (received through IHumanState interface in HumanDynamicsEstimator)

In which frame is this expressed? Can we rotate it in the world frame? 

> Berdy classical acceleration estimates : -6.15775 -5.04919 10.1027  -26.9387 -0.157612 2.62974 

This indeed seems completely wrong. Is this consistent or it quickly changes?

traversaro, 2019-08-29:
Can you rotate the `Received Proper acceleration` back in  the world frame? Otherwise it is difficult to check if it is ok or not, rotating it on the world we can compare it with `0.0 0.0 9.81` and verify that it is correct. For the same reason, it is convenient to rotate the linear part of the `Berdy classical acceleration estimates`. However, as rapidly changing, there seems to be some problem in any case on the correctness of the acceleration estimate. 
I suggest to: 
* If you are familiar with it, run at least once the estimator under `valgrind`, to make sure that there are no more error related to accessing memory that should not be accessed
* Verify that a minimal set of "trusty" variables are present for doing a correct estimation. If there is no such minimum set, the estimation process is ill-posed. A minimum set is for example the one used in the classical RNEA for deterministic computation of the proper sensor acceleration for each link: 
  * Proper linear sensor acceleration for at least one link (and we should be ok with at least one accelerometer)
  * Angular acceleration for at least one link (not sure if you have that)
  * Joint acceleration for all joints, even set to 0 at least temporary 
  * Measurements of all external forces except for one 
* If you have a minimal set of "trusty" variables, one thing you can do is to simplify the problem by setting to zero some "velocity" and "acceleration" sensors, for example set to zero joint velocity and joint acceleration
* If after all these checks the estimation is still extremely noisy, I suggest to decrease the covariance on the prior of dynamic variables: this would make sure that, if multiple solution exists to a problem, the one closed to zero is chosen consistently, even if the input data to the estimation varies slightly.

Yeshasvitvs, 2019-08-29:
> Can you rotate the Received Proper acceleration back in the world frame? Otherwise it is difficult to check if it is ok or not, rotating it on the world we can compare it with 0.0 0.0 9.81 and verify that it is correct. 

I double checked this in matlab and the resulting value is equal to the corrected acceleration.

> For the same reason, it is convenient to rotate the linear part of the Berdy classical acceleration estimates. However, as rapidly changing, there seems to be some problem in any case on the correctness of the acceleration estimate.

As discussed with @traversaro @claudia-lat @DanielePucci we need to put the angular acceleration measurement covariance to a low value. Currently it is set to `1e4` and this could be one of the reasons for wrong values in the link estimated angular accelerations. @claudia-lat added an angular accelerometer berdy sensor type in MAPEst and set the covariance to be low e.g `1e-4` to ensure reasonable values of the link angular acceleration estimates. This is not done in berdy back-end c++ code base. So, a quick way is to simple set the covariance low for the link angular acceleration measurement in the dynamics variable vector. 

Another important points discussed are:

- After the bug in acceleration handling https://github.com/robotology/human-dynamics-estimation/commit/8d598f6e876bd4fd50716e851d905e718d8af674 we need to use the [same covairances as used in MAPEst code.](https://github.com/claudia-lat/MAPest/blob/cd29bddab6d4bb1f5cefab6b76cbd3674b4a4d23/Experiments/23links_human/configureAndRunMAPest.m)
- Update the angular acceleration sensor measurements coviariance to a low value of `1e-4`. Eventually, we need to handle the angular acceleration as @claudia-lat does in MAPEst. The link [angular acceleration coming from wearable data](https://github.com/robotology/wearables/blob/ff6bd445a9f6d377f13286993824ad768dccb1ab/interfaces/IWear/include/Wearable/IWear/Sensors/IVirtualLinkKinSensor.h#L37) is [rotated in to the sensor frames and used as input in the measurement vector.](https://github.com/claudia-lat/MAPest/blob/floatingWithIWear_featureAngAccSensor/Experiments/23links_human/main_floating.m#L160). @traversaro please confirm if there is any berdy sensor for handling angular accelerations.
- Double check the link acceleration estimates from HumanDynamicsEstimator and MAPEst, given the same covariances and measurements are used.

@traversaro @DanielePucci  @claudia-lat @lrapetti @kouroshD

Yeshasvitvs, 2019-08-29:
Using the same [covariances as in MAPEst](https://github.com/claudia-lat/MAPest/blob/floatingWithIWear_featureAngAccSensor/Experiments/23links_human/configureAndRunMAPest.m) except for covariance of dynamics constraints (i.e model constraints) which is set to `1e-2` as used by @claudia-lat - Done in https://github.com/robotology/human-dynamics-estimation/commit/d8ac5db7c7e34254405d75d2155978204575ac87

The test with offline T-pose data set shows near zero estimated wrenches on the hands

[image]

I created a [gist with the accelerometer values](https://gist.github.com/Yeshasvitvs/122d2ec6e956e144e76f4264f3a577bf). The [berdy estimates of link acceleration values](https://gist.github.com/Yeshasvitvs/122d2ec6e956e144e76f4264f3a577bf#file-hde_acceleration_estimation_covariances_as_mapest-L1013) seem reasonable.

Yeshasvitvs, 2019-08-29:
The implementation details to consider also the angular acceleration are:

- Extract the link acceleration from the wearable data inside HumanStateProvider - Done in https://github.com/robotology/human-dynamics-estimation/commit/e78a32cb4a90935e64bd594b0f8d6a4a551b99b0
- Consider the link angular acceleration while computing the proper acceleration - Done in https://github.com/robotology/human-dynamics-estimation/commit/afbb8d7d2e1f8e03306cae955e5cb330e79f5280
- Add angular acceleration sensors for berdy in HumanDynamicsEstimator and consider the proper angular acceleration passed from HumanStateProvider through IHumanStateInterface. Also, set low covariance for angular accelerometer measurements - Done in https://github.com/robotology/human-dynamics-estimation/commit/99c14bd07b0891834e6131020fb507b3f69d0905

I did an offline test again with the same T-pose data set above and the wrench estimates stay the same as before https://github.com/robotology/human-dynamics-estimation/issues/145#issuecomment-526147281. 

The proper acceleration input to measurement vector and the berdy acceleration estimates are much cleaner now.

```

 ================ Proper acceleration input to berdy measurement vector=============== 
 
 Pelvis_accelerometer   8.98606   0.747295   3.89202   -0.115815   -0.154251   -0.247145 
 T8_accelerometer   8.3949   -2.76636   4.31518   -0.540112   -1.37011   -0.133384 
 Head_accelerometer   -1.61375   7.39182   -6.35315   -0.719664   -0.782844   0.384807 
 RightShoulder_accelerometer   0.876   -9.00732   3.87303   -1.12346   -1.08107   0.262004 
 RightUpperArm_accelerometer   1.80908   -6.79949   6.88783   2.77393   0.134517   0.129087 
 RightForeArm_accelerometer   -1.77137   -3.81307   9.1174   3.56776   -0.554723   -0.0958761 
 RightHand_accelerometer   -0.0816426   -0.971783   9.77799   8.05463   0.154156   -3.15276 
 LeftShoulder_accelerometer   2.24121   9.23796   2.19167   0.546564   -0.086622   -0.472285 
 LeftUpperArm_accelerometer   1.00302   9.18254   3.35583   0.562287   1.35362   0.399118 
 LeftForeArm_accelerometer   -2.15981   4.40401   8.76249   -1.95055   0.394953   -0.465483 
 LeftHand_accelerometer   1.18485   1.18204   9.93193   -4.47637   -0.0144446   0.630221 
 RightUpperLeg_accelerometer   9.7581   0.198159   -1.15702   -0.195016   0.187814   0.18045 
 RightLowerLeg_accelerometer   9.64246   -1.45988   -1.2154   -0.663705   -0.0322869   0.177207 
 RightFoot_accelerometer   5.43319   -2.75597   7.66629   0.035844   0.062717   -0.0755735 
 LeftUpperLeg_accelerometer   9.74963   0.822468   -0.593879   -0.523412   -0.0550324   0.0375711 
 LeftLowerLeg_accelerometer   9.6525   -1.00929   -1.21794   0.11832   0.190711   -0.503846 
 LeftFoot_accelerometer   5.55679   1.60327   7.89536   -0.136939   0.214656   -0.100911 

 ================ Berdy classical acceleration estimates size is  67  ============== 
Acceleration for link Pelvis:-4.52193 0.311736 -1.77796  0.280567 -0.314875 0.0550207 
Acceleration for link L5_f1:-1.55328 0.348305 5.22177  0.225497 -0.298134 0.0722518 
Acceleration for link L5_f2:0.982819 0.444893 12.3222  0.172936 -0.284507 0.0918266 
Acceleration for link L5:3.71194 0.590631 19.6062  0.128248 -0.273177 0.104914 
Acceleration for link L3_f1:3.02743 0.647757 15.7515  0.0869989 -0.351038 0.122648 
Acceleration for link L3_f2:2.20117 0.592026 12.0958  0.0469303 -0.438622 0.140561 
Acceleration for link L3:1.65099 0.550928 8.51378  0.0121408 -0.525119 0.159662 
Acceleration for link T12_f1:1.52082 0.4454 6.38074  -0.0338252 -0.580029 0.184859 
Acceleration for link T12_f2:1.37821 0.288508 4.33841  -0.0806283 -0.645969 0.207104 
Acceleration for link T12:1.35278 0.141632 2.31554  -0.121429 -0.714948 0.233644 
Acceleration for link T8_f1:2.43191 -0.894269 -0.723888  -0.221459 -0.82695 0.266751 
Acceleration for link T8_f2:3.63944 -1.96057 -3.72556  -0.32108 -0.956509 0.29504 
Acceleration for link T8:4.88667 -3.02724 -6.79425  -0.41635 -1.08923 0.331261 
Acceleration for link Neck_f1:4.92599 -2.14882 -5.78308  -0.445212 -1.0101 0.314733 
Acceleration for link Neck_f2:6.29493 -1.72767 -2.9396  -0.56332 -0.918427 0.205279 
Acceleration for link Neck:6.17272 -1.5872 -1.84472  -0.648923 -0.815105 0.246894 
Acceleration for link Head_f1:7.03111 -1.25231 -1.78073  -0.679651 -0.812614 0.21641 
Acceleration for link Head_f2:7.89906 -1.16414 -2.19139  -0.693849 -0.798552 0.31289 
Acceleration for link Head:8.91593 -1.61504 -2.1131  -0.781109 -0.749247 0.36755 
Acceleration for link RightShoulder_f1:1.19451 -3.81527 -6.04648  -0.439521 -0.461445 -0.0113222 
Acceleration for link RightShoulder_f2:-3.10353 -3.3087 -5.84168  -0.527668 0.103317 -0.519361 
Acceleration for link RightShoulder:-6.41016 -3.52225 -5.59909  -0.685351 0.565763 -1.11974 
Acceleration for link RightUpperArm_f1:-6.81094 -5.09869 -2.71773  -0.170302 -0.526865 -0.308734 
Acceleration for link RightUpperArm_f2:-5.8817 -5.22301 4.18734  0.667297 -1.30277 0.211466 
Acceleration for link RightUpperArm:-7.092 -2.27204 5.42127  0.490169 -2.60549 0.534059 
Acceleration for link RightForeArm_f1:-2.57558 -2.03685 5.30878  0.304058 -2.96542 -0.156164 
Acceleration for link RightForeArm_f2:-0.162456 -0.399525 6.51945  0.15356 -3.33481 -0.0448388 
Acceleration for link RightForeArm:3.57372 1.53223 8.77917  0.283736 -3.69746 -0.0136849 
Acceleration for link RightHand_f1:4.66745 2.17157 7.74564  0.204235 -5.18998 -0.246831 
[... cut ...]
Acceleration for link LeftFoot_f2:4.3486 4.47269 3.46121  0.864527 -0.548656 -1.38198 
Acceleration for link LeftFoot:6.80785 6.31284 6.15652  1.00426 -0.726279 -0.931229 
Acceleration for link LeftToe_f1:7.07437 6.60663 9.46665  0.974636 -0.92438 -0.842599 
Acceleration for link LeftToe_f2:9.59417 6.50576 11.8343  0.796732 -1.06115 -0.986683 
Acceleration for link LeftToe:10.7093 6.38552 15.7085  0.781111 -1.20722 -0.975499 

```

I think we can proceed to go for online testing and tuning #144 

@traversaro @DanielePucci @claudia-lat @kouroshD @lrapetti

claudia-lat, 2019-08-30:
Please @diegoferigo if you can review the code it would be perfect!

Yeshasvitvs, 2019-09-01:
A new `COM_ACCELEROMETER_SENSOR` is added to berdy helper class https://github.com/robotology/idyntree/pulls https://github.com/robotology/idyntree/issues/570 

- `HumanStateProvider` is updated to get COM acceleration and proper com accelearation (_abuse of notation_) i.e considering the gravity acceleration is computed. Also, `IHumanState` is updated with new methods to pass this com acceleration to `HumanDynamicsEstimator`- Done in https://github.com/robotology/human-dynamics-estimation/commit/678f66504eb0f15eaf4dab52c5743a547f821d16 
- `HumanDynamicsEstimator` is updated with the new `COM_ACCELEROMETER_SENSOR` berdy sensor and the measurements of proper com acceleration are considered in the run method - Done in https://github.com/robotology/human-dynamics-estimation/commit/99b8b49ee6f3a1d4109a9dce1b3ece42499a2061

@traversaro @DanielePucci @claudia-lat @kouroshD @lrapetti

claudia-lat, 2019-09-23:
This issue is related to the accelerometer measurements and it has been fixed [here](https://github.com/robotology/human-dynamics-estimation/issues/145#issuecomment-526229020).  The COM sensor is investigated in other issues.
Let's close this issue.

closed 2019-09-23 by claudia-lat

Yeshasvitvs, 2019-09-23:
FYI @prashanthr05  this is the issue related to considering accelerometer values inside `HumanStateProvider`.

Yeshasvitvs, 2019-09-24:
##### Issue Summary

Following https://github.com/robotology/human-dynamics-estimation/issues/145#issuecomment-525716713, the relation between the free body acceleration (fba) and the proper acceleration is given as: (Considering only the linear acceleration)
 
<a href="https://www.codecogs.com/eqnedit.php?latex=^G\mathbf{a_{fba}}&space;=&space;^GR_S&space;*&space;(^S\mathbf{a}&space;&plus;&space;^SR_G&space;*\mathbf{g})\:&space;\:&space;\:&space;\:&space;\:&space;with\:\:&space;\mathbf{g}&space;=&space;[0,&space;0,&space;-9.81]" target="_blank"><img src="https://latex.codecogs.com/gif.latex?^G\mathbf{a_{fba}}&space;=&space;^GR_S&space;*&space;(^S\mathbf{a}&space;&plus;&space;^SR_G&space;*\mathbf{g})\:&space;\:&space;\:&space;\:&space;\:&space;with\:\:&space;\mathbf{g}&space;=&space;[0,&space;0,&space;-9.81]" title="^G\mathbf{a_{fba}} = ^GR_S * (^S\mathbf{a} + ^SR_G *\mathbf{g})\: \: \: \: \: with\:\: \mathbf{g} = [0, 0, -9.81]" /></a>

what we get from the xsens IMU through wearable `fbAcceleration` is: <a href="https://www.codecogs.com/eqnedit.php?latex=^G\mathbf{a_{fba}}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?^G\mathbf{a_{fba}}" title="^G\mathbf{a_{fba}}" /></a>
the sensor proper acceleration is :  <a href="https://www.codecogs.com/eqnedit.php?latex=^S\mathbf{a}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?^S\mathbf{a}" title="^S\mathbf{a}" /></a>

The relation can be rearranged as: <a href="https://www.codecogs.com/eqnedit.php?latex=^S\mathbf{a}&space;\&space;=&space;\&space;^SR_G&space;\&space;(^G\mathbf{a_{fba}}&space;-&space;\mathbf{g})" target="_blank"><img src="https://latex.codecogs.com/gif.latex?^S\mathbf{a}&space;\&space;=&space;\&space;^SR_G&space;\&space;(^G\mathbf{a_{fba}}&space;-&space;\mathbf{g})" title="^S\mathbf{a} \ = \ ^SR_G \ (^G\mathbf{a_{fba}} - \mathbf{g})" /></a>

Here `G` is the xsens inertial frame of reference. The values that are passed to `HumanDynamicsEstimator`  through `IHumanState` interface are <a href="https://www.codecogs.com/eqnedit.php?latex=^S\mathbf{a}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?^S\mathbf{a}" title="^S\mathbf{a}" /></a> which are expressed in the accelerometer sensor frame 

At the implementation level, the details for handling **linear** acceleration can be summarized as:

- On each run, we obtain the `fbAcceleration` values [through wearable interface](https://github.com/robotology/human-dynamics-estimation/blob/feature/visualize-berdy-estimated-wrench/devices/HumanStateProvider/HumanStateProvider.cpp#L922) i.e. <a href="https://www.codecogs.com/eqnedit.php?latex=^G\mathbf{a_{fba}}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?^G\mathbf{a_{fba}}" title="^G\mathbf{a_{fba}}" /></a>

- <a href="https://www.codecogs.com/eqnedit.php?latex=^G\mathbf{a_{fba}}&space;-&space;\mathbf{g}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?^G\mathbf{a_{fba}}&space;-&space;\mathbf{g}" title="^G\mathbf{a_{fba}} - \mathbf{g}" /></a> [is computed](https://github.com/robotology/human-dynamics-estimation/blob/feature/visualize-berdy-estimated-wrench/devices/HumanStateProvider/HumanStateProvider.cpp#L1044)  

- Sensor orientation is obtained [through wearable interface](https://github.com/robotology/human-dynamics-estimation/blob/feature/visualize-berdy-estimated-wrench/devices/HumanStateProvider/HumanStateProvider.cpp#L929) i.e. <a href="https://www.codecogs.com/eqnedit.php?latex=^GR_S" target="_blank"><img src="https://latex.codecogs.com/gif.latex?^GR_S" title="^GR_S" /></a>

- Finally the sensor proper acceleration [is computed](https://github.com/robotology/human-dynamics-estimation/blob/feature/visualize-berdy-estimated-wrench/devices/HumanStateProvider/HumanStateProvider.cpp#L1090) <a href="https://www.codecogs.com/eqnedit.php?latex=^S\mathbf{a}&space;\&space;=&space;\&space;^SR_G&space;\&space;(^G\mathbf{a_{fba}}&space;-&space;\mathbf{g})" target="_blank"><img src="https://latex.codecogs.com/gif.latex?^S\mathbf{a}&space;\&space;=&space;\&space;^SR_G&space;\&space;(^G\mathbf{a_{fba}}&space;-&space;\mathbf{g})" title="^S\mathbf{a} \ = \ ^SR_G \ (^G\mathbf{a_{fba}} - \mathbf{g})" /></a> and it is expressed in the sensor frame

At the implementation level, the details for handling **angular** acceleration can be summarized as:

- On each run, we obtain the `linkAcceleration` values [through wearable interface](https://github.com/robotology/human-dynamics-estimation/blob/feature/visualize-berdy-estimated-wrench/devices/HumanStateProvider/HumanStateProvider.cpp#L889) and [use only the angular part](https://github.com/robotology/human-dynamics-estimation/blob/feature/visualize-berdy-estimated-wrench/devices/HumanStateProvider/HumanStateProvider.cpp#L1064) i.e. <a href="https://www.codecogs.com/eqnedit.php?latex=^G\mathbf{a_{link}}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?^G\mathbf{a_{link}}" title="^G\mathbf{a_{link}}" /></a>

- Finally, the sensor angular acceleration [is computed](https://github.com/robotology/human-dynamics-estimation/blob/feature/visualize-berdy-estimated-wrench/devices/HumanStateProvider/HumanStateProvider.cpp#L1090) i.e.  <a href="https://www.codecogs.com/eqnedit.php?latex=^S\mathbf{a_{link}}&space;=&space;\&space;^SR_G&space;\&space;^G\mathbf{a_{link}}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?^S\mathbf{a_{link}}&space;=&space;\&space;^SR_G&space;\&space;^G\mathbf{a_{link}}" title="^S\mathbf{a_{link}} = \ ^SR_G \ ^G\mathbf{a_{link}}" /></a>