gbionics/human-dynamics-estimation #147: Investigate external forces estimation from berdy

Yeshasvitvs, 2019-09-24:
Following #140 #145 we still do not see correct external force estimation from berdy. This issue will track all the details during the code verification and brainstorming for the sprint investigation 

@claudia-lat @lrapetti @prashanthr05 @kouroshD @traversaro @DanielePucci

Yeshasvitvs, 2019-09-25:
I went through the paper [Simultaneous Floating-Base Estimation of Human Kinematics and Joint Torques](https://www.mdpi.com/1424-8220/19/12/2794) and we have a couple of comparison tests done on

- Feet wrench (Measured and Estimated)
- Base linear acceleration (Measured and Estimated)

We can try to do the same validation tests with HDE after considering the accelerometer values inside the measurement vector `y`. These tests will help us to check if there are any mistakes at this level even before considering the rate of change of momentum constraint.

To perform the tests, we will get the following data sets with the wired suit placed on the table and a `Npose` calibration quality set to `poor`. This will provide data sets where the IMU accelerations are close to  zero.

- Suit on the table and human on the shoes
- Suit on the table and no human on the shoes

We dump the wearable data and run HDE with the dumped data. The following two variables are dumped to a text file

- Measurement vector `y` after feeding the data of accelerations and wrench 
- Dynamic variables vector `y` after berdy estimation

Yeshasvitvs, 2020-01-27:
This tasks is done and tested with stack of tasks berdy implementation. Closing.

cc @lrapetti  @claudia-lat @DanielePucci

closed 2020-01-27 by Yeshasvitvs

DanielePucci, 2020-01-28:
Probably, a pointer to your thesis and the code would help future reference for this @Yeshasvitvs