gbionics/human-dynamics-estimation #148: Add COM Computation for all the IK methods in human state provider

kouroshD, 2019-09-25:
Currently, we have the CoM information only for integration based IK. We should add it also for other IK methods.

claudia-lat, 2020-01-27:
@kouroshD is this issue something to do or do you think it's better to put in Icebox?

lrapetti, 2020-11-16:
@kouroshD I think CoM computation is now achieved, am I right?

kouroshD, 2020-11-16:
> @kouroshD I think CoM computation is now achieved, am I right?

We have this for integration based IK.
For other IK methods, not sure. We can check this, and if it is not implemented, we can do it.
This is also highly related to @Yeshasvitvs  work, related to SOT work. Maybe he has some points to add here.

Yeshasvitvs, 2020-11-16:
> We have this for integration based IK.

That is on the devel branch

> This is also highly related to @Yeshasvitvs work, related to SOT work. 

I am not using COM value in SOT HDE

We can close this issue.