gbionics/human-dynamics-estimation #149: Acquire test data set to check HDE and MAPEst implementation of BERDY

Yeshasvitvs, 2019-09-25:
Following the idea of https://github.com/robotology/human-dynamics-estimation/issues/147#issuecomment-535052995, in this issue we track the details related to the test data that will be used to do the tests for ensuring HDE and MAPEst implementation of berdy for the validation tests on

- Feet wrench (Measured and Estimated)
- Base linear acceleration (Measured and Estimated)

@claudia-lat @lrapetti @prashanthr05 @kouroshD  @DanielePucci

lrapetti, 2019-09-26:
I have collected the two datasets:
- dataset wearing the shoes
https://drive.google.com/open?id=1SGINH3w1_pvEMdm3swF4YL0DOmRBySF-
- data without the shoes
https://drive.google.com/open?id=1s8BJPWMuQanzxy4Re6kG_G4H-ijK9bdG

 @claudia-lat can you please check if you are able to MAPEst with those datasets?

Yeshasvitvs, 2019-09-26:
I ran HDE with humanSubject3_48dof model and with the following covariances

```
<group name="PRIORS">
            <param name="mu_dyn_variables">0.0001</param>
            <param name="cov_dyn_variables">10000.0</param>
            <param name="cov_dyn_constraints">0.0001</param>
            <param name="cov_measurements_ACCELEROMETER_SENSOR">(0.0011 0.0011 0.0011)</param>
            <param name="cov_measurements_THREE_AXIS_ANGULAR_ACCELEROMETER_SENSOR">(1e-3 1e-3 1e-3)</param>
            <param name="cov_measurements_DOF_ACCELERATION_SENSOR">0.666e-5</param>
            <group name="cov_measurements_NET_EXT_WRENCH_SENSOR">
                <param name="value">1e-6</param>
                <param name="specific_elements">(LeftFoot RightFoot)</param>
                <param name="LeftFoot">(0.0589998348 0.0589998348 0.0359999712 0.002250000225 0.002250000225 0.56e-3)</param>
                <param name="RightFoot">(0.0589998348 0.0589998348 0.0359999712 0.002250000225 0.002250000225 0.56e-3)</param>
            </group>
        </group>
```

- The output text files containing the measurement vector `y` and the dynamic variable estimates for **dataset wearing the shoes** is   [HDE_output-d-y-vectors_test_data_suit_on_table_human_on_shoes.zip](https://github.com/robotology/human-dynamics-estimation/files/3657586/HDE_output-d-y-vectors_test_data_suit_on_table_human_on_shoes.zip)

- The output text files containing the measurement vector `y` and the dynamic variable estimates for **dataset without the shoes** is [HDE_output-d-y-vectors_test_data_suit_on_table_human_out_of_shoes.zip](https://github.com/robotology/human-dynamics-estimation/files/3657596/HDE_output-d-y-vectors_test_data_suit_on_table_human_out_of_shoes.zip)

claudia-lat, 2019-09-26:
Closing #149

closed 2019-09-26 by claudia-lat