gbionics/human-dynamics-estimation #150: Add the joint limits to the constrained QP solver

kouroshD, 2019-09-26:
In this issue, we want to track the addition of the joint limits to the constrained QP problem.

kouroshD, 2019-09-26:
The implementation has been added in this commit.
[ the QP constraints for all the joint limits are added (found from urd… # e21b51e](https://github.com/robotology/human-dynamics-estimation/commit/e21b51e4d984cad817c33849305240f72dadf57e).
Since the customized constraint were added before, I have appended the joint limit matrices/vectors to the customized constraint matrices/vectors.
https://github.com/robotology/human-dynamics-estimation/commit/e21b51e4d984cad817c33849305240f72dadf57e#diff-f8d68d61a7d1cf3627cbd20c215bb5a8R705-R739

kouroshD, 2019-09-26:
This has to be tested with the previous implementation to check the responses are OK.

kouroshD, 2019-09-29:
Following are the results of the new implementation for considering the joint limits in the QP problem.  As I mentioned in the previous comments, for the safety reasons also the saturation is applied. 
In the Following figure, we can see the difference between the updated implementation of the QP with commit [ fix the bug for the joint limit 956f772b31](https://github.com/robotology/human-dynamics-estimation/commit/956f772b31a16d1f51533cd08686568fcc4bb1b8).
- [Blue color: we consider the joint limits in QP as expalnined in issue #132]()
- Red Color: we do not consider the joint limits in QP , and only the  saturation is applied.

[image]
[image]

kouroshD, 2019-09-30:
I have tested on the human model the changes, with the following details, IK takes in average **`5ms`** to solve, in macOS.
```
[INFO]QP initialization is done, more info: 
[INFO]constraint Matrix size: 204  ,  144 
[INFO]upper bound vector size: 204 
[INFO]lower bound vector size: 204 
[INFO]Hessian Matrix size: 144  ,  144 
[INFO]Gradient Vector size: 144 
```

kouroshD, 2019-09-30:
I should later check the results in rViz, then we can proceed with the PR merge. @lrapetti

DanielePucci, 2019-10-25:
@kouroshD how did the tests with rViz went?

kouroshD, 2019-10-25:
So, in the sprint https://github.com/dic-iit/component_wholebody-teleoperation/issues/177 , we have tested several times the branch before we merge it to `devel`. It worked fine, therefore merged the PR https://github.com/robotology/human-dynamics-estimation/pull/137 . We can close this issue.

closed 2019-10-25 by kouroshD

lrapetti, 2020-02-27:
I was checking the effect of the joint constraints used in the QP, trying to plot the velocity-configuration space for some joints, and I got some plots that might be interesting for you
[image]

This kind of plots I think might be very useful when considering tuning of those gains.

kouroshD, 2020-02-27:
@lrapetti  This is the phase space trajectory or phase portrait of the `l_elbow`, which is cool. (Applied Nonlinear Control, Slotine, chapter 3!)

A question, why the boundary condition does not seem linear since the constraints are a system of linear equations?

lrapetti, 2020-02-27:
> A question, why the boundary condition does not seem linear since the constraints are a system of linear equations?

If you refer to the velocity boundary condition is not linear since it depends on the joint configuration ($\dot{s}^{\text{upper}}=f^u(s)$ and $\dot{s}^{\text{lower}}=f^l(s)$ are non-linear, since they involve the arctangent function).

> btw, the fact that it goes out of upper and lower bound is related to `k_u` and `k_l`, but it should be OK, we can explain it, since it depends on the dynamics of the system (how fast it is) and the sampling rate.

yes, I agree with this observation

kouroshD, 2020-02-27:
Did you also consider this constraint?
 https://github.com/robotology/human-dynamics-estimation/blob/devel/conf/xml/RobotStateProvider_iCub.xml#L88-L89
I am not sure if it is possible this to visualize it since it is a function of `l_wrist_prosup` as well.

lrapetti, 2020-02-27:
> Did you also consider this constraint?
> https://github.com/robotology/human-dynamics-estimation/blob/devel/conf/xml/RobotStateProvider_iCub.xml#L88-L89
> I am not sure if it is possible this to visualize it since it is a function of `l_wrist_prosup` as well.

No, I didn't notice there was this coupled constraint for the elbow! Thanks for pointing it out

kouroshD, 2020-02-27:
btw, I highly suspect, the zones on the top-left and bottm-right is less cowded because of that constraint! maybe, you can double check it with the data of the `l_wrist_prosup`.