gbionics/human-dynamics-estimation #151: Add logger for the Human-Dynamics-Estimation devices

lrapetti, 2019-10-03:
Currently a logger for the HDE devices is not implemented. Adding a logger may help in dumping the data for offline processing and debugging.

One option is to use [`mat2logger`](https://github.com/ADVRHumanoids/MatLogger2) for which a very preliminary implementation can be found in [`lrapetti/human-dynamics-estimation/AddLogger`](https://github.com/lrapetti/human-dynamics-estimation/tree/AddLogger). It has the advantage of enabling to use directly the output data structure in  `matlab`, and is currently used in other project in the `robotology` universe.

lrapetti, 2019-10-03:
What need to be understood is if mat2logger handles only scalar, or if it is possible to work also with `string/char`? Because in our case it is important also to associate the measurement with the order of the joints used in the model, and we need to store the list of joints in a matlab file (this list should be saved only once, not at each timestamp).

lrapetti, 2019-10-03:
Is there any solution for string that you are aware of so that we can have everything dumped as `.mat`?
@traversaro @GiulioRomualdi @prashanthr05

traversaro, 2019-10-03:
> Is there any solution for string that you are aware of so that we can have everything dumped as .mat?
@traversaro @GiulioRomualdi @prashanthr05

I do not know with `matlogger2`, but with generic matio it is possible to read and write cell arrays of strings, see for example the following function: 
~~~cpp
    void writeVectorOfStringToMat(std::string name, const std::vector<std::string>& strings, mat_t* mat)
    {
        size_t dims[2] = {1, static_cast<size_t>(nj)};
        matvar_t* cell_array = Mat_VarCreate(name.c_str(),MAT_C_CELL,MAT_T_CELL,2,dims,NULL,0);
        RTF_ASSERT_ERROR_IF_FALSE(mat, Asserter::format("Failed call to Mat_VarCreate"));
        for(int i=0; i < strings.size(); i++)
        {
            std::string stringToWrite = strings[i];
            // TODO(traversaro): verify if there are encoding problems
            size_t stringDim[2] = { 1, stringToWrite.size() };
            matvar_t *cell_element = Mat_VarCreate("dummy", MAT_C_CHAR, MAT_T_UTF8, 2, stringDim, const_cast<char *>(stringToWrite.c_str()), 0);
            Mat_VarSetCell(cell_array, i, cell_element);
        }
        Mat_VarWrite(mat, cell_array, MAT_COMPRESSION_NONE);
        Mat_VarFree(cell_array);
    }
~~~

claudia-lat, 2019-12-10:
I'll move the issue on the Iteration.  If you @lrapetti  have new comments on this, please add it.

lrapetti, 2019-12-11:
Probably we should consider implementing the logger with `matio` so that we are able to dump string.

Yeshasvitvs, 2020-02-03:
In view of https://github.com/robotology/human-dynamics-estimation/issues/175#issuecomment-581395222, @diegoferigo and I discussed and decided that it is better to stick with the design choice of having a single outputPort from all the wrapper devices. 

Towards having a clean way for logging and real-time visualization purposes, an ideal choice as @diegoferigo suggested is to have a new device say `HDEDataCollector` which collects all the data through different interfaces like `IHumanState` and `IHumanDynamics` into a single place. Furthermore, the data can be streamed through relevant yarp ports that can be used for real-time yarpscope visualization, or implement `matio` logging at a single place. Also, for the future, we can add capability/interfaces to use visualization tools such as [Processing](https://processing.org/) or [Grafana](https://grafana.com/) or [TensorBoard](https://www.tensorflow.org/tensorboard) 

@diegoferigo @traversaro @lrapetti @kouroshD @claudia-lat @DanielePucci

Yeshasvitvs, 2020-02-05:
@traversaro is this the [matio library](https://github.com/tbeu/matio) you were referring to in  https://github.com/robotology/human-dynamics-estimation/issues/151#issuecomment-537966403 ?

diegoferigo, 2020-02-06:
@Yeshasvitvs It is, and you might even try to check the CMake integration of https://github.com/tbeu/matio/pull/107 to import the library. It's there since forever, not sure when is going to get merged but it should be working.

traversaro, 2020-02-06:
> 
> 
> @Yeshasvitvs It is, and you might even try to check the CMake integration of [tbeu/matio#107](https://github.com/tbeu/matio/pull/107) to import the library. It's there since forever, not sure when is going to get merged but it should be working.

`matio` is available in apt, homebrew and vcpkg, so I suggest to install it via the system package managers and avoid compiling from sources, especially from a branch with a not clear status.

diegoferigo, 2020-02-06:
Ow right, good catch, I forgot it was available in the repos.

prashanthr05, 2020-02-08:
MATIO is nice but it is C based and one needs to worry a lot about memory management. So code carefully :)

kouroshD, 2020-02-10:
> What need to be understood is if mat2logger handles only scalar, or if it is possible to work also with `string/char`? Because in our case it is important also to associate the measurement with the [...]

It would be strange if it does not allow logging string/char! If you want, we can do a quick test to see if it allows or not!

lrapetti, 2020-02-10:
> It would be strange if it does not allow logging string/char! If you want, we can do a quick test to see if it allows or not!

I tried in the past and it seemed not be allowed for what I understood

DanielePucci, 2020-02-15:
Hi @lrapetti  missing history points to this issue

CC
@claudia-lat

lrapetti, 2020-02-17:
This task is now addressed using `HumanDataCollector` (https://github.com/robotology/human-dynamics-estimation/issues/179) for what concerns `HumanStateProvider` and `HumanDynamicsEstimation` data. What will be missing then is a logger for `wearables` data.

Yeshasvitvs, 2020-02-17:
In view of https://github.com/robotology/human-dynamics-estimation/issues/179 where the feature of adding matio logging is discussed, this issue is redundant and not necessary. Closing

@lrapetti

closed 2020-02-17 by Yeshasvitvs